INTI-CMNB/KiBot #489: [BUG]

Informaticore, 2023-09-01:
**Describe the bug**
Render Blender PNG is missing. In my yaml file I am using:

```
outputs:
  - name: 'Blender 3D Image'
    type: 'blender_export'
    dir: 'Generated/render'
    options:
      point_of_view:
        rotate_x: 45
        rotate_y: 45
        rotate_z: 75
      pcb3d:
        download: true
      render_options:
        samples: 100
        transparent_background: true
      outputs:
        type: render
```

the container I am using is: `container: ghcr.io/inti-cmnb/kicad7_auto_full:latest`

I have not touched the config since a while and with my commit some time ago it was generating that PNG

Informaticore, 2023-09-01:
@set-soft what else do you need? the output which is supposed to generate a PNG is not generating a PNG anymore.
The project is there, what else can I tell you? 

this is the yaml: https://github.com/open-thngs/so-dimm-esp32-cm/blob/main/esp-cm3.kibot.yaml
this is the action: https://github.com/open-thngs/so-dimm-esp32-cm/blob/main/.github/workflows/kibot.yml

set-soft, 2023-09-01:
I'll try to look at your files later, you can provide the debug output, if Blender is failing you'll see the Blender error there.

set-soft, 2023-09-01:
Hi @Informaticore !
In order to get the _debug_ information you must use `-v`, or `-vvvv` for even more details.
I think I found the problem, while I find a solution you can try changing `latest` by `dev` (1.6.4 development) or `1.6.3-1_k7.0.5.1_d11.6_b3.4` (1.6.3 release using KiCad 7.0.5 and Blender 3.4, assuming your project can be loaded by KiCad 7.0.5).
The problem seems to be related to the pcb2blender upgrade.

Informaticore, 2023-09-01:
> Hi @Informaticore ! In order to get the _debug_ information you must use `-v`, or `-vvvv` for even more details. I think I found the problem, while I find a solution you can try changing `latest` by [...]

KiCad 7.0.5 should work just fine, I am using 7.0.6 currently. I changed it to dev and it took a while (14min compared to 2) but I guess thats fine and might be related to actually render the things. Now it created the PNG

set-soft, 2023-09-01:
Yes, the render for your board is slow. Most probably because of the ESP32 module.