INTI-CMNB/KiBot #496: [BUG] `TypeError` with KiKit-annotation style multiboard project (but internal tool)

burnpanck, 2023-09-16:
Fill as much as you can, remove the rest

**Describe the bug**
I have a PCB project containing multiple boards in a single file/project.
I use the KiKit annotation method to separate the boards.
KiKit is able to handle the board correctly (the one in the KiBot docker image).
Running KiBot with a `kibom` variant using the `sub_pcbs` key (no `tool` key - default is `internal`) however results in an error in any output related to the pcbs
(observed so far with `print_pcb` and `ibom`, but not with `pdf_sch_print`). 
The traceback is
```
Traceback (most recent call last):
  File "/usr/bin/kibot", line 33, in <module>
    sys.exit(load_entry_point('kibot==1.6.3', 'console_scripts', 'kibot')())
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/kibot/__main__.py", line 520, in main
    generate_outputs(outputs, args.target, args.invert_sel, args.skip_pre, args.cli_order, args.no_priority,
  File "/usr/lib/python3/dist-packages/kibot/kiplot.py", line 527, in generate_outputs
    _generate_outputs(outputs, targets, invert, skip_pre, cli_order, no_priority, dont_stop)
  File "/usr/lib/python3/dist-packages/kibot/kiplot.py", line 517, in _generate_outputs
    run_output(out, dont_stop)
  File "/usr/lib/python3/dist-packages/kibot/kiplot.py", line 418, in run_output
    out.run(get_output_dir(out.dir, out))
  File "/usr/lib/python3/dist-packages/kibot/out_base.py", line 197, in run
    self.options.run(self.expand_filename(output_dir, output))
  File "/usr/lib/python3/dist-packages/kibot/out_ibom.py", line 187, in run
    self.filter_pcb_components()
  File "/usr/lib/python3/dist-packages/kibot/out_base.py", line 750, in filter_pcb_components
    self._sub_pcb.apply(self.comps_hash)
  File "/usr/lib/python3/dist-packages/kibot/var_base.py", line 313, in apply
    self.board_rect = self.search_reference_rect(self.reference)
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/kibot/var_base.py", line 271, in search_reference_rect
    edges = self.get_pcb_edges()
            ^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/kibot/var_base.py", line 216, in get_pcb_edges
    edges.append(Edge(edge))
                 ^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/kibot/var_base.py", line 41, in __init__
    self.start = GS.get_start_point(shape)
                 ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/kibot/gs.py", line 612, in get_start_point
    return g.GetStart()+pcbnew.wxPoint(g.GetRadius(), 0)
           ~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
TypeError: unsupported operand type(s) for +: 'VECTOR2I' and 'wxPoint'
```
The outline contains Arcs and an internal circle. The issue does not arise when using `tool: kikit` to separate the boards.

**Environment (please complete the following information):**
Where are you running KiBot: Manually in docker, using `kicad7_auto:1.6.3`

set-soft, 2023-10-02:
Hi @burnpanck !

The above patch should fix the problem (and also add support for KiCad 5).

You can check it using the _kicad7_auto_full:dev_ image

closed 2023-10-02 by set-soft

burnpanck, 2023-10-03:
Indeed, seems to work now as expected - thanks! That said, we have migrated to separate board files in the meantime. The multiboard project was a good intermediate ground while moving from rigid-flex to two separate boards.

commit c216d4bf (2023-10-02): [Sub PCBs][Fixed] Using annotation method

- For some edeges and KiCad versions

Closes #496