INTI-CMNB/KiBot #497: [BUG] boardCommentN not passed to Kikit

Unichord, 2023-09-18:
I'm automating panelization on Windows 10 WSL, it works well but some parameters are not passed to Kikit (or there is some other problem), as boardTitle, boardRevision and boardDate are passed correctly and they appear on the board. Comments, on the other hand, don't appear on the board, if I add some other text before or after {boardCommentN} in the yaml, it appears on the panel.

This is the yaml:
```
text3:
          type: simple
          hoffset: -25
          voffset: 4
          width: 4
          height: 4
          thickness: 0.4
          hjustify: right
          text: {boardComment2}
          anchor: tr
```

This is the kicad_pcb:
```
(paper "A4")
  (title_block
    (title "GFP1")
    (date "2023-09-18")
    (rev "1.0")
    (comment 2 "23.4480")
  )
```

By the way, is there a way to copy exactly the schematic Page Settings to the pcb Page Settings (board title, comments, company, etc)? I usually fill in the schematic only, and it's a bit of a hassle to copy all information in the pcb just for this to work. Otherwise, would it be possible to just pass external variables the same way it's done for other yaml? Like this:
```
text3:
          type: simple
          hoffset: -25
          voffset: 4
          width: 4
          height: 4
          thickness: 0.4
          hjustify: right
          text: ${my_var}
          anchor: tr
```

set-soft, 2023-09-18:
Hi @Unichord !

Can you post a small example where it can be reproduced? The *text* should be passed to KiKit without any changes, you can check it using **-vvvv**.

About the copy of fields, I think using variables is the way to go. It should work for the panel. Just use text variables in the title block and then set them in KiCad and/or using set_text_variables.

About them in the panelize texts: it should also work because they are passed to KiKit and they should get in the PCB, so then printing it should expand the variables.

Unichord, 2023-09-21:
Hello @set-soft, I sent you an email with the whole project and more info, have you by chance received it?

set-soft, 2023-09-24:
Hi @Unichord !

I got the mail, I had to copy: *panel_std.kibot.yaml* and *pre.kibot.yaml* to the *Doc/Settings* folder.

I noticed you are running the panelize output twice, I think you need to use something like:

```diff
--- .panel.kibot.yaml~	2023-09-18 14:26:52.000000000 -0300
+++ panel.kibot.yaml	2023-09-24 16:21:58.307643732 -0300
@@ -11,6 +11,7 @@
   comment: Panelize board
   type: panelize
   dir: ${panel}
+  disable_run_by_default: panel_std
   options:
     configs:
       - name: panel
```

Enabling debug information you can see what is passed to KiKit, I did it and got:

```json
    "text2": {
        "height": "4mm",
        "text": "{boardTitle}",
        "thickness": "0.4mm",
        "type": "simple",
        "voffset": "4mm",
        "width": "4mm"
    },
    "text3": {
        "anchor": "tr",
        "height": "4mm",
        "hjustify": "right",
        "hoffset": "-25mm",
        "text": "{boardComment2}",
        "thickness": "0.4mm",
        "type": "simple",
        "voffset": "4mm",
        "width": "4mm"
    },
```

So *boardComment2* is passed to KiKit. Why it doesn't work ... not sure, ask @yaqwsx

I'm adding an option to expand ${} and %X markers before sending it to KiKit, so you'll be able to use text variables (defined by you) and/or environment variables. I think KiCad will also expand its internal vars.

Unichord, 2023-09-26:
Hi @set-soft, external variables is the optimal solution I guess, I'll wait for a new version. Thanks.

closed 2023-09-28 by set-soft