INTI-CMNB/KiBot #500: [FEATURE] Gerber X3 Support

schneider-engineering, 2023-09-29:
**Is your feature request related to a problem? Please describe.**
Gerber X3 Position/Component export does not seem to be supported. (Or I could not find out how to enable it).

**Describe the solution you'd like**
We would like to be able to export Gerber X3 Position files.

**Additional context**
See below screenshots for how to export gerber X3 files. Despite having the term "gerber" in the name, in KiCad it is placed in "Component Placement". 

[image]
[image]

Any help on this would be appreciated.

schneider-engineering, 2023-09-29:
A colleague just noticed that kicad-cli actually supports x3 export already:
`kicad-cli pcb export pos --format gerber --use-drill-file-origin --side front`
`kicad-cli pcb export pos --format gerber --use-drill-file-origin --side back`

Would it be possible to include this in KiBot?

set-soft, 2023-10-02:
Hi @schneider-engineering !

I'm adding some experimental support using the *kicad-cli* interface.
It is available on the _dev_ branch, you can try the *kicad_auto_full:dev* docker image.
Is really experimental: new to KiBot and KiCad itself says is experimental.
The command line help is wrong, your example helped to clarify it! Reading the help it says:

```
 --side                 	Valid options: front,back,both. Gerber format only supports "both". [default: "both"]
```

But for gerber you *must* specify a side, *both* isn't supported. Then it says:

```
--use-drill-file-origin	Use drill/place file origin (ascii or csv only)
```

But is also supported for gerbers, as you show in your example, and this option is the default KiCad behavior for other position files.

Can you try this functionality and let me know if the generated files are correct?
BTW: are you using them for some particular manufacturer or for some particular machine?

commit d0c489e2 (2023-10-02): [Position][Added] Experimental support for gerber position files

Closes #500
closed 2023-10-02 by set-soft