INTI-CMNB/KiBot #501: [HOW] Clear or disable cached models

N0cynym, 2023-10-06:
**What do you want to achieve?**

I have removed my SOT-89B-1.wrl 3D model from the pcb file because this model is broken.
My problem is that kibot seems to cache this model and thinks it's still used but it's no more part of my pcb file!

```
DEBUG:- Forcing WRL /root/.cache/kibot/3d/Package_TO_SOT_SMD.3dshapes/SOT-89B-1.wrl (kibot - out_base_3d.py:189)
WARNING:(W098)  3 3D models downloaded or cached (kibot - out_base_3d.py:522)
```

N0cynym, 2023-10-13:
I was just wondering why kibot is forcing to the broken wrl file which I disabeled and later removed from the pcb file content.
It is correct that this file was availible (but broken) and for some reason kibot seems to cache the lib download/3d models and forces to them.

I fixed the wrl now but just want to ask if I could force kibot to use what I have defined in my pcb file and not to try to use wrl...

set-soft, 2023-10-13:
KiCad needs BOTH, some operations can be performed without a WRL (or you a get a very poor result).

KiBot caches things downloaded, not everything. You can always clean the cache in your workflow if you don't want it.

closed 2023-10-13 by set-soft