INTI-CMNB/KiBot #502: [HOW] How do I install missing dependencies in Github Actions

mundodisco8, 2023-10-06:
**What do you want to achieve?**
I pushed some changes to project for the first time in over a year. When I pushed to Github, it failed because one of the packages required in my workflow is no longer installed (mistune, required in Populate3D).

My question is how can I run `apt-get install python3-mistune` in the Github Actions context?

```
/usr/bin/docker run --name b81[18](https://github.com/mundodisco8/Snowflake/actions/runs/6431124766/job/17463386892#step:4:19)86a2b4a348d0a8951a1a01c21adc_9b833e --label 94351b --workdir /github/workspace --rm -e "INPUT_CONFIG" -e "INPUT_DIR" -e "INPUT_SCHEMA" -e "INPUT_BOARD" -e "INPUT_QUICKSTART" -e "INPUT_SKIP" -e "INPUT_TARGETS" -e "INPUT_VARIANT" -e "INPUT_INSTALL3D" -e "INPUT_VERBOSE" -e "INPUT_ADDITIONAL_ARGS" -e "INPUT_CACHE3D" -e "HOME" -e "GITHUB_JOB" -e "GITHUB_REF" -e "GITHUB_SHA" -e "GITHUB_REPOSITORY" -e "GITHUB_REPOSITORY_OWNER" -e "GITHUB_REPOSITORY_OWNER_ID" -e "GITHUB_RUN_ID" -e "GITHUB_RUN_NUMBER" -e "GITHUB_RETENTION_DAYS" -e "GITHUB_RUN_ATTEMPT" -e "GITHUB_REPOSITORY_ID" -e "GITHUB_ACTOR_ID" -e "GITHUB_ACTOR" -e "GITHUB_TRIGGERING_ACTOR" -e "GITHUB_WORKFLOW" -e "GITHUB_HEAD_REF" -e "GITHUB_BASE_REF" -e "GITHUB_EVENT_NAME" -e "GITHUB_SERVER_URL" -e "GITHUB_API_URL" -e "GITHUB_GRAPHQL_URL" -e "GITHUB_REF_NAME" -e "GITHUB_REF_PROTECTED" -e "GITHUB_REF_TYPE" -e "GITHUB_WORKFLOW_REF" -e "GITHUB_WORKFLOW_SHA" -e "GITHUB_WORKSPACE" -e "GITHUB_ACTION" -e "GITHUB_EVENT_PATH" -e "GITHUB_ACTION_REPOSITORY" -e "GITHUB_ACTION_REF" -e "GITHUB_PATH" -e "GITHUB_ENV" -e "GITHUB_STEP_SUMMARY" -e "GITHUB_STATE" -e "GITHUB_OUTPUT" -e "RUNNER_OS" -e "RUNNER_ARCH" -e "RUNNER_NAME" -e "RUNNER_ENVIRONMENT" -e "RUNNER_TOOL_CACHE" -e "RUNNER_TEMP" -e "RUNNER_WORKSPACE" -e "ACTIONS_RUNTIME_URL" -e "ACTIONS_RUNTIME_TOKEN" -e "ACTIONS_CACHE_URL" -e GITHUB_ACTIONS=true -e CI=true -v "/var/run/docker.sock":"/var/run/docker.sock" -v "/home/runner/work/_temp/_github_home":"/github/home" -v "/home/runner/work/_temp/_github_workflow":"/github/workflow" -v "/home/runner/work/_temp/_runner_file_commands":"/github/file_commands" -v "/home/runner/work/Snowflake/Snowflake":"/github/workspace" 94351b:811886a2b4a348d0a8951a1a01c21adc  "-c ArtifactCreation.kibot.yaml" "-C NO" "-d Artifacts" "-b hardware/Snowflake.kicad_pcb" "-e hardware/Snowflake.kicad_sch" "-i NO" "-q NO" "-s __NONE__" "-t __ALL__" "-v 0" "-V __NONE__" "-x "
*****************************************************************************************
*****************************************************************************************

KKKKKKKKK    KKKKKKK  iiii  BBBBBBBBBBBBBBBBB                             tttt
K:::::::K    K:::::K i::::i B::::::::::::::::B                         ttt:::t
K:::::::K    K:::::K  iiii  B::::::BBBBBB:::::B                        t:::::t
K:::::::K   K::::::K        BB:::::B     B:::::B                       t:::::t
KK::::::K  K:::::KKKiiiiiii   B::::B     B:::::B   ooooooooooo   ttttttt:::::ttttttt
  K:::::K K:::::K   i:::::i   B::::B     B:::::B oo:::::::::::oo t:::::::::::::::::t
  K::::::K:::::K     i::::i   B::::BBBBBB:::::B o:::::::::::::::ot:::::::::::::::::t
  K:::::::::::K      i::::i   B:::::::::::::BB  o:::::ooooo:::::otttttt:::::::tttttt
  K:::::::::::K      i::::i   B::::BBBBBB:::::B o::::o     o::::o      t:::::t
  K::::::K:::::K     i::::i   B::::B     B:::::Bo::::o     o::::o      t:::::t
  K:::::K K:::::K    i::::i   B::::B     B:::::Bo::::o     o::::o      t:::::t
KK::::::K  K:::::KKK i::::i   B::::B     B:::::Bo::::o     o::::o      t:::::t    tttttt
K:::::::K   K::::::Ki::::::iBB:::::BBBBBB::::::Bo:::::ooooo:::::o      t::::::tttt:::::t
K:::::::K    K:::::Ki::::::iB:::::::::::::::::B o:::::::::::::::o      tt::::::::::::::t
K:::::::K    K:::::Ki::::::iB::::::::::::::::B   oo:::::::::::oo         tt:::::::::::tt
KKKKKKKKK    KKKKKKKiiiiiiiiBBBBBBBBBBBBBBBBB      ooooooooooo             ttttttttttt

🤖 KiBot GitHub Action v2 🚀

KiBot: 1.6.3
KiCad: 7.0.7~ubuntu23.04.1
Debian: 12.1
KiAuto: 2.2.7
KiCost 1.1.18
iBoM: 2.7.0

*****************************************************************************************
*****************************************************************************************
Options: -c 'ArtifactCreation.kibot.yaml' -d 'Artifacts' -b 'hardware/Snowflake.kicad_pcb' -e 'hardware/Snowflake.kicad_sch'
WARNING:(W008) Unable to find KiCad configuration file (/github/home/.config/kicad/7.0/kicad_common.json) (kibot - config.py:[19](https://github.com/mundodisco8/Snowflake/actions/runs/6431124766/job/17463386892#step:4:20)7)
WARNING:(W010) Unable to find KiCad user templates (kibot - config.py:431)
- Running the DRC
WARNING:(W058) Missing KiCad main config file /github/home/.config/kicad/7.0/kicad_common.json (pcbnew_do - misc.py:[20](https://github.com/mundodisco8/Snowflake/actions/runs/6431124766/job/17463386892#step:4:21)0) (kibot - kiplot.py:1[32](https://github.com/mundodisco8/Snowflake/actions/runs/6431124766/job/17463386892#step:4:33))
- Running the ERC
- Updating BoM in XML format
- 'Used to generate the BoM in CSV, HTML, TSV, TXT, XML or XLSX format using the internal BoM.' (BoM_In_CSV) [bom]
- 'Used to generate the BoM in CSV, HTML, TSV, TXT, XML or XLSX format using the internal BoM.' (BoM_In_XSLX) [bom]
- 'This is the information for the drilling machine in gerber format.' (Gerber_Drills) [gerb_drill]
- 'This is the main fabrication format for the PCB.' (Gerber_Generation) [gerber]
- 'Generates an interactive web page useful to identify the position of the components in the PCB.' (ibom_example) [ibom]
- 'Exports the PCB to the most common exchange format. Suitable for printing.' (pdf_sch_print_example) [pdf_sch_print]
- 'Creates a markdown file explaining how to assembly a PCB using KiCad 3D rendered.' (Populate3D) [populate]
- Trying to install mistune (from PyPi)
ERROR:Missing `mistune` python module (mistune), install it (kibot - dep_downloader.py:797)
ERROR:Debian package: python3-mistune (kibot - dep_downloader.py:797)
ERROR:Arch package: python-mistune (kibot - dep_downloader.py:797)
ERROR:Output that needs it: populate (kibot - dep_downloader.py:797)
ERROR:Try running the installation checker: kibot-check (kibot - dep_downloader.py:797)
```

**Do you have some PCB/Schematic to use as example?**
The project is [this one, on the CICD branch](https://github.com/mundodisco8/Snowflake/tree/CICDTest)

**Do you have some configuration file (.kibot.yaml) that you are using?**
You can attach it or paste the content in the following section:

Pasted in a comment

**Environment (please complete the following information):**
Where are you running KiBot:
- In a CI/CD workflow? `GitHub`
- In a GitHub action? Which tag are you using? `v2_k7`

mundodisco8, 2023-10-06:
```yaml
# ATTENTION! THIS ISN'T A FULLY FUNCTIONAL EXAMPLE.
# You should take portions of this example and edit the options to make
# them suitable for your use.
# This file is useful to know all the available options.
kibot:
  version: 1

global:
  # Acknoledge some of the WARNINGS thrown in the process
  # Errors in other areas, when filtered, will appear as warnings, and can be also be filtered here too
  filters:
    # When running KiBot in Github, it will complain that we have no kicad_common.json
    - filter: "No kicad_common"
      number: 8
      regex: 'Unable to find KiCad configuration file'
    # ... and again when doing the DRC
    - filter: "No kicad_common while DRC"
      number: 58
      regex: 'Missing KiCad main config file'
    # ... or user templates...
    - filter: "No user templates"
      number: 9
      regex: 'Unable to find KiCad user templates'
    # Even when passing a kicad_common.json, it has no environmental vars
    - filter: "No Environmental Vars in Kicad Config"
      number: 9
      regex: 'KiCad config without environment.vars section'
    # I "think" the image has no 3D models to reduce size...
    - filter: 'No 3D Models'
      number: 10
      regex: 'Unable to find KiCad 3D models'
    # S1 is the silkscreen with the text in the back (turn this...). It doesn't have a symbol associated in the schematic
    # - filter: 'S1 in board, but not in schematic'
    #   number: 13
    #   regex: 'S1'
    - filter: "Battery Internal connection"
      number: 58
      regex: 'BT1 on B.Cu\n.+BT1 on B.Cu'
    # Complaint about the resistor of the potentiometer being `0-50k`
    - filter: 'Pot value'
      number: 20
      regex: "Malformed value: `0-50k`"
  solder_mask_color: white
  silk_screen_color: black
  pcb_finish: ENIG

#### Filters for the Populate3D step
filters:
    - name: clock
      comment: "Clock Components: U1, RV1, C1, R1-5, BT1, SW1"
      type: generic
      include_only:
        - field: Reference
          regex: '\bR[1-5]\b'
        - field: Reference
          regex: 'U1|RV1|C1'
    - name: currentResistors
      comment: "Remaining current resistors: R7-10, R13-18"
      type: generic
      include_only:
        - field: Reference
          regex: 'R[7-9]|R1[0|3-8]'
    - name: LEDs
      type: generic
      include_only:
        - field: Reference
          regex: 'D[2-9]|D1[0-3]'

    - name: all_back
[... 1176 lines cut ...]
      # [boolean=true] Clip silkscreen at solder mask edges (KiCad 6)
      subtract_mask_from_silk: true
      # [boolean=false] When enabled the image will be post-processed to make the background transparent.
      # In this mode the `background1` and `background2` colors are ignored
      transparent_background: false
      # [string='#00ff00'] Color used for the chroma key. Adjust it if some regions of the board becomes transparent
      transparent_background_color: '#00ff00'
      # [number=15] [0,100] Chroma key tolerance (percent). Bigger values will remove more pixels
      transparent_background_fuzz: 15
      # [string=''] Board variant to apply
      # variant: ''
      # [string='top'] [top,bottom,front,rear,right,left,z,Z,y,Y,x,X] Point of view
      # view: 'top'
      # `wait_ray_tracing` is an alias for `wait_render`
      # [number=-600] How many seconds we must wait before capturing the render (ray tracing or normal).
      # Lamentably KiCad can save an unfinished image. Enlarge it if your image looks partially rendered.
      # Use negative values to enable the auto-detect using CPU load.
      # In this case the value is interpreted as a time-out.
      # wait_render: -60
      # [number=1280] Image width (aprox.)
      width: 1280
      # [number=0] Zoom steps. Use positive to enlarge, get closer, and negative to reduce.
      # Same result as using the mouse wheel in the 3D viewer
      zoom: 4
```

set-soft, 2023-10-06:
Hi @mundodisco8 !
mistune should be there, this is a fault in the docker image, I'm moving this issue to the docker images project.