INTI-CMNB/KiBot #503: [FEATURE] Multiple configuration files

gcormier, 2023-10-08:
KiKit can support [multiple configuration files](https://yaqwsx.github.io/KiKit/v1.4/panelization/cli/#configurations), and takes priority from the last file specified to override previous options. The CLI takes highest priority.

It would be great if KiBot could do the same. That is,

`kibot -c kibot-generic-panel.yaml -c kibot-productspecific-panel.yaml`

In this use case, perhaps `kibot-generic-panel.yaml` is all the default options and settings used for many projects ( eg tooling, fiducials, text placements), and `kibot-product-specific-panel.yaml` might just be as simple as having `rows` and `cols` set for this specific project's panel.

set-soft, 2023-10-09:
Hi @gcormier !

I think this is already addressed by the import mechanism, read more [here](https://kibot.readthedocs.io/en/latest/configuration/imports.html)

closed 2023-10-09 by set-soft

gcormier, 2023-10-09:
Thanks, I think you are right and this will cover my use-case.

Would it be possible to have external templates defined over HTTP as a future feature? So one could point to a static URL (eg a static github.com url)?

I suppose right now it could be done locally with git submodules.

set-soft, 2023-10-11:
Hi @gcormier !

About the http: if you really think this is helpful please add a feature request.

And yes, this is usually solved using submodules, so you can share the common stuff between projects.

schneider-engineering, 2023-10-11:
@set-soft: Is there any way to dynamically specify the filename on the command line?

This does not work:
```
definitions:
  VAR_LIST_FILE: default_variants.yaml

import:
  - file: @VAR_LIST_FILE@
    outputs: none
    preflights: none
    filters: none
    variants: all
    global: none
    groups: none
```

It seems the import is run before the substitiution is done? Error looks like this:
```
ERROR:Trace stack: (kibot.gs - gs.py:737)
  File "/usr/local/lib/python3.11/dist-packages/kibot/__main__.py", line 507, in main
    outputs = cr.read(cf_file)
              ^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/kibot/config_reader.py", line 680, in read
    data = self.load_yaml(fstream, collected_definitions)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/kibot/config_reader.py", line 660, in load_yaml
    raise KiPlotConfigurationError("Error loading YAML "+str(e))
ERROR:Error loading YAML while scanning for the next token
found character '@' that cannot start any token
```

schneider-engineering, 2023-10-11:
Hey @set-soft,

thank you. All I was missing was the `...`. The YAML parser complaining about `@` lead me down the wrong path.

set-soft, 2023-10-11:
Note that the error is on purpose, the @ is invalid for YAML. If the parser complains it means something went wrong and the variables weren't replaced.