INTI-CMNB/KiBot #504: [HOW] Named all gerber layers

andersruneson, 2023-10-08:
**What do you want to achieve?**
So we have a generic kibot.yaml file for all our projects and we do not need to configure the config file for each board. That works well.

We just recently started migrating to KiCad7 and KiBot7 and found an "issue" that we previously had a work around for.

We would like the kibot.yaml to include generating all gerber layers, so that we do not need to change the config per board, but we also would like to name the output file, hence we cannot use the simple variant of just generating all with:
```
  layers:
  - 'copper'
```

But instead we do :
```
  layers:
  - layer: F.Cu
    suffix: 'Cu Top'
    description: Front copper
  - layer: In1.Cu
    suffix: 'CuInner1'
    description: Inner1 copper
  - layer: In2.Cu
    suffix: 'CuInner2'
    description: Inner2 copper
  - layer: In3.Cu
    suffix: 'CuInner3'
    description: Inner3 copper
  - layer: In4.Cu
    suffix: 'CuInner4'
    description: Inner4 copper
  - layer: B.Cu
    suffix: 'Cu Bottom'
    description: Bottom copper
```
But since not all of the inner layers always exist (as we use the same yaml for all our board, sometimes we only have 2 layers but most of the time 4 layers) we got a warning (49, Layer Inner copper... is'nt used) running KiBot5 which we handled with the filter warning/error function, as a work around.

But now with KiBot7 we get this error when the yaml has layers that don't exist in the layout:
"ERROR:In output `'Gerbers in modern format, recommended by the standard' (gerber_modern) [gerber]`: Inner layer `In3.Cu (3 'Inner3 copper' CuInner3)` is not valid for this board (kibot - kiplot.py:421)"
and I don't think we can filter away that one.

Do you have a suggestion other than configuring the yaml file per board or using the "all layers" function?

Thanks!

andersruneson, 2023-10-09:
Hi, thanks for fast reply and fix! Is this in the docker images already? 

Is below how it's meant to be used?
```
global:
  layer_defaults:
  - layer: In1.Cu
    suffix: 'CuInner1'
    description: Inner1 copper
  - layer: In2.Cu
    suffix: 'CuInner2'
    description: Inner2 copper
  - layer: In3.Cu
    suffix: 'CuInner3'
    description: Inner3 copper
  - layer: In4.Cu
    suffix: 'CuInner4'
    description: Inner4 copper
```

And then in the gerber_modern output I just have this to generate all according to the defaults?
```
  layers:
  - 'copper'
  - layer: B.Paste
    suffix: 'Paste Bottom'
    description: Bottom solder paste
  - layer: F.Paste
    suffix: 'Paste Top'
    description: Front solder paste

...
```

set-soft, 2023-10-11:
> Hi, thanks for fast reply and fix! Is this in the docker images already?

Yes, in the *dev* images. Was included in this action: https://github.com/INTI-CMNB/KiBot/actions/runs/6457421834

> Is below how it's meant to be used?

Yes, just defining the layers here. Be careful with renamed layers.

> And then in the gerber_modern output I just have this to generate all according to the defaults?

Yes, a generic use here, the defaults will be applied.

andersruneson, 2023-10-12:
Just tested my example above and it works well! Many thanks!

commit 1a5b52da (2023-10-09): [Global options][Added] `layer_defaults`

- To specify the default suffix and description.

Closes #504