INTI-CMNB/KiBot #506: [HOW] Questions about Blender Rendering

mundodisco8, 2023-10-09:
**What do you want to achieve?**

My goal is to use Blender to render some images of my board that I can generate automatically on every commit, so my README.MD always shows the latest render of my board.

I am not familiar to Blender, and I don't know how much can be done "from" KiBot, as a KiBot action or if on the other hand I should only use KiBot to generate the Blender files and then use Blender's own automation to generate those renders.

My questions are (if you prefer me to open one ticket for each, tell me and I will do it)
1. In my renders, I always get green soldermask. My project has the colours correctly set (KiCad would render it as white Soldermask). I opened the Blender project and noticed the pcb2blender shader, where I can set the colours to what I want. Is there a way to get it to work on the renders?
2. I have a decorative element, which has the ENIG plating exposed, but it's not meant to be soldered on. The render, obviously, treats it as a solderpad, and covers it in paste. I haven't seen an option to enable/disable elements like the solder paste.
3. I'm trying to render four images. Top, bottom, and then something slanted like your default view for top and bottom. I tried to use the `point of view: view` option but using top or bottom doesn't change anything. I suspect that I have to set "top" and "bottom" views on Blender before I get anything, but I can't see how to do it "before" I trigger the generation with KiBot. I also tried to move the point of view, with the `rotate_Axis` options, but because Blender uses the top left of the board as origin of coordinates, when I rotate the board, it gets out of the camera view. I haven't seen an option to "move" the point of view.

If you think this could be better done by using blender's automation options, don't lose any time on it, I just wanted to ask before I go down Blender's rabbit hole :D

**Do you have some PCB/Schematic to use as example?**
I'm messing around with [this project](https://github.com/mundodisco8/Snowflake) but it doesn't have anything special to it.

**Do you have some configuration file (.kibot.yaml) that you are using?**
You can attach it or paste the content in the following section:
```yaml
Paste YAML example here
```

**Environment (please complete the following information):**
Where are you running KiBot:
- In a CI/CD workflow? (Ex: `GitHub`/`GitLab`)
- In a GitHub action? Which tag are you using? (Ex `v1`, `v1_k6`, `v2`, `v2_d`, `v2_dk6`, `v2_k6`)
- Using a docker image? (Ex: `kicad_auto:ki6`, `kicad_auto:latest`, `kicad_auto:dev`, `kicad_auto:dev_k6`, a custom one? can you provide the `Dockerfile`)
- Using a desktop system? Which OS, distro and version?

**Additional context**
Add any other context about your setup/project.

mundodisco8, 2023-10-10:
> pcb2blender 2.7 supports it. But we currently use 2.6. Will be supported in the future.

Fantastic, can't wait for that!

> Does this component have solder paste?
No it doesn't. Copper and Mask. It might be an artifact, though: it might be that it defaults to HASL finish? I think that the new pcb2blender might help with that, as it will pick the ENIG from the file, I imagine. Check it out, it's only 1 pass to speed things up.

[image]

> Did you try using the current dev images?

> Also take a look at the new internal templates: 3DRender_top, 3DRender_top_straight, 3DRender_bottom and 3DRender_bottom_straight. They are created to simplify it.
> More [here](https://kibot.readthedocs.io/en/latest/configuration/imports.html#importing-internal-templates)

That's exactly what I needed. It works as a charm in dev. After trying the same config in latest (and commenting some of the options, that don't seem to be supported on latest), my PCB is still off. Commenting the "camera" group of options and leaving it to take the defaults worked, so I'll stop pretending I'm Spielberg and comment that "camera" out. Happy to move to dev too.

closed 2023-10-10 by mundodisco8

mundodisco8, 2023-10-12:
Hey! I just read this! I noticed that something was off when I run my pipeline and got some new "errors", I checked the repo and noticed the jump to 2.7, I came to say that I noticed the change 👍 

Check that out! Wonderful! Thanks for the work you're are putting into this!

BTW, the errors weren't errors, but debug messages

```
ERROR:6 (0, 0, 0) (kibot - out_pcb2blender_tools.py:223)
```

set-soft, 2023-10-18:
Thanks for reporting the debug message that I forgot to remove.

Is related to problems when using custom colors for the solder/silk mask.