INTI-CMNB/KiBot #508: How to replace variables in the schematic header block

andersruneson, 2023-10-13:
This may be a really stupid question, but I'm trying to use set_text_variables to replace the schematic header block variables and have no success. So is this possible?

This is what my header block looks like in the Drawing Sheet Editor:
[image]

And I've got a yaml that looks like this:
```
preflight:
  set_text_variables:
    - name: 'COMMENT1'
      text: 'COMMENT1'
    - name: 'Comment1'
      text: 'Comment1'
    - name: 'comment1'
      text: 'comment1'
    - name: 'Comment 1'
      text: 'Comment 1'
    - name: 'comment 1'
      text: 'comment 1'
    - name: 'title'
      text: 'TESTtitle'
    - name: 'TITLE'
      command: 'echo HEJ TITEL'
    - name: 'ISSUE_DATE'
      text: '12345'
    - name: 'REVISION'
      text: 'EA12345-AB EB...'

```

But nothing in the header is replaced when running the basic_pdf_sch_print output.

andersruneson, 2023-10-13:
[testproject.zip](https://github.com/INTI-CMNB/KiBot/files/12898148/testproject.zip)
Attached is a testproject, in the small chance of me actually doing it correct.

set-soft, 2023-10-13:
Hi @andersruneson !

You can't change the variables that are internally defined by KiCad. They are values you enter in the *Page Settings* dialog.

Also: KiCad interprets them in a case sensitive way. So you can actually use:

[image]

And get:

[image]

Using:

```yaml
kibot:
  version: 1

preflight:
  set_text_variables:
    - name: 'COMMENT1'
      text: 'COMMENT1'
    - name: 'Comment1'
      text: 'Comment1'
    - name: 'comment1'
      text: 'comment1'
    - name: 'Comment 1'
      text: 'Comment 1'
    - name: 'comment 1'
      text: 'comment 1'
    - name: 'title'
      text: 'TESTtitle'
    - name: 'TITLE'
      command: 'echo HEJ TITEL'
    - name: 'issue_date'
      text: '12345'
    - name: 'revision'
      text: 'EA12345-AB EB...'

outputs:
  - name: 'print_sch'
    comment: "Print schematic (PDF)"
    type: pdf_sch_print
    dir: .
    options:
      output: Schematic.pdf
```

closed 2023-10-13 by set-soft

andersruneson, 2023-10-14:
Ahaaaaaa! Thanks so much!