INTI-CMNB/KiBot #509: [FEATURE] Provide `arm64` versions of the Docker images

dmohns, 2023-10-15:
**Is your feature request related to a problem? Please describe.**
Currently, all KiBot docker images are provided for `amd64` platform. For users on `arm64` platforms this requires to run the docker images using emulation, i.e. QEMU or Rosetta2 (in case of MacOS). This does work most of the time, but is not very performant and occasionally run into strange errors related to architecture, which are hard to debug for the average developer and user of KiBot.

I understand that this is not very relevant for the CI/CD use-case _right now_, as almost all runners are on `amd64` platform. However, `arm64` is on the rise for Desktop environments and KiBot helps tremendously to reproducibly generate documentation and production files in projects with multiple contributors (using different OS'es).

**Describe the solution you'd like**
All Docker images are provided in `amd64` as well as `arm64` platform. When the Docker images are run on `arm64` no emulation is needed and native performance can be utilised.

**Describe alternatives you've considered**
Running KiBot with emulation. See above

**Additional context**

- There is already a similar comment here: https://github.com/INTI-CMNB/kicad_auto/issues/14 But I felt it would be better suited as a discussion in this repo. Please feel free to close my issue as duplicated, if the linked issue is enough to track the feature.
- I already did some experiments of building the Docker image chain on `arm64`. With a few modifications to the Docker files I was able to run everything expect for Blender. I can share results if this is insightful.

set-soft, 2023-10-18:
Hi @dmohns !

At first sight it looks like too much work is needed. So I'm not sure about it.

The first step should be to add KiAuto packages for ARM64 arch. This means we must create Debian packages of KiAuto for both architectures. Without this KiAuto won't use the interposer and the KiCad GUI manipulation will be unreliable.

With the current KiAuto package I think the best will be to add a cross-compiled *libinterposer.so*. But again, a lot of work to just bust the speed on a minority of machines.

Even trying to make a "draft", that just uses the old GUI manipulation mechanism, it implies doubling the storage, bandwidth, etc. of the images build. Not to mention that buildx likes to need a lot more resources and complicate the setup a lot. This is because it uses docker images for things that the standard build just uses the OS installed tools.

If you are patient we could try to explore it and see the results. Did you build the "base_os" images for ARM?

set-soft, 2023-10-19:
> > If you are patient we could try to explore it and see the results. Did you build the "base_os" images for ARM?
> 
> Yes, with a few modification the base images built. Essentially I needed remove/change all Debian packages that don't provide `arm64`
> 
> * Changed `rar` -> `unrar` [here](https://github.com/INTI-CMNB/kicad_debian/blob/master/base_os/Dockerfile#L12)

Ok, we should see a mechanism to just skip it for ARM, *unrar* only uncompress, and what we need is to compress. But is completely optional.

> * The Ubuntu PPA for KiCad doesn't provide `arm64` either. However, I noticed that Debian [Bookworm-Backport](https://packages.debian.org/bookworm-backports/kicad) repo has `arm64` version, so I changed to use that. Changing `kicad` to `kicad/backport` [here](https://github.com/INTI-CMNB/kicad_debian/blob/master/ki7/Dockerfile#L14)

Interesting looks like Debian is catching-up.

> I struggled to get a working (up-to-date) version of Blender. Here is what I tried
> 
> * Using the Debian Bookworm version (3.4). Seems to outdated to proper function with some of the KiCad scripts
> * Installing the Provided version by blender. Well... it's an `amd64` so obviously it didn't work
> * I even changed the entire build chain to use Debian-Sid instead of Debian-Bookworm. This did make KiCad scripts work, but it fails at a later step with some opague "Build without OpenDenoiser" warnings, which I figured was beyond my knowledge to debug 😉
> 
> I then postponed the attempt of getting Blender to run 🙈

From what I understand the Debian packages are only usable with hardware acceleration, otherwise they are useless. 

This is because the Debian maintainer has a strict understanding of Debian guidelines and refuses to include the Intel OpenDenoiser lib. I think he's wrong. The problem is like this: Intel people needed some functionality found in another lib, so they took this lib (free software) and heavily modified it for their own needs. The modifications can't be incorporated to the original lib because they are too specific to OpenDenoiser. So Intel people is including the code in the OpenDenoiser code base. As they don't plan to incorporate the changes to the upstream the Debian policy forbids the incorporation of OpenDenoiser. But the point is that the changes are changing the third party lib in a way that the upstream will never accept them. End result: OpenDenoiser can't be packaged according to the Blender Debian package maintainer. I see this ridiculous.

You may ask: is OpenDenoiser really important? Well just try to render something using the CPU, you'll need a lot of time per render pass. In 10 passes you'll get a very noisy image, you'll need something like 100 passes to get rid of the ray tracing noise. In a test I did with a simple board it took 40 minutes. But using OpenDenoiser you can just do 10 passes and then apply OpenDenoiser, you get a good approximation for the 100 passes result. Suddenly you went from 40 to 4 minutes. So yes, OpenDenoiser is critical if you don't have access to a hardware accelerator, which is the normal case for a regular CI/CD server.

This is why I'm not using the Debian package in the docker images, is a pity because I'm wasting around 640 MiB (from a 3.3 GiB total) because I'm forced to use the official Blender tarball.

All of this makes me wonder: What's the OS you are using? Are you using macOS? Perhaps we don't really need docker images for ARM64, but better native support for macOS. KiCad and Blender are available for macOS, and Darwin is a UNIX BSD system, so I don't see why KiBot couldn't run natively on macOS.

burnpanck, 2025-03-13:
Maybe with the new KiCad 9 IPC API, the `libinterposer.so` issues could be avoided (hopefully no more need for GUI manipulation). Nowadays, GitHub Actions offers arm64 runners which are both faster and cheaper than the amd64 equivalent. Also, macOS nowadays is essentially exclusively arm64, so, while emulation still works, it does get somewhat slow.

set-soft, 2025-03-13:
I hope the remaining GUI interaction gets solved by `kicad-cli` but I saw too much regressions and slow advances in a lot of `kicad-cli` fronts, so I can't be sure.