INTI-CMNB/KiBot #510: [HOW] Using symbol field for specify component rotation

radiolyc, 2023-10-22:
### Intro
Hi! I tried to use  filter for pre-rotation of components and it works.
### Issue
But the problem is different components can have different rotation even if the footprint (package) is the same.
### Question
So, Is it possible to use symbol field to specify rotation correction angle for specific component?
### Example
This mechanism is used in **Fabrication Toolkit** extension: https://github.com/bennymeg/JLC-Plugin-for-KiCad#-offset-component-rotation
[image]

set-soft, 2023-11-08:
Hi @radiolyc !

I'm trying to implement the *Fabrication Toolkit* mechanism, but during the validation process I found that KiBot is apllying wrong rotations on the bottom side components. I guess most people only uses the top side for components, and the bottom side is used only in extreme cases, usually not for cheap boards manufactured on JLCPCB. But I want to make it work. I also noticed that recent versions of *Fabrication Toolkit* just disabled the rotations based on the footprint! The tool seems to currently rely on rotations manually specified by the user. I guess this is something to do with the complexity found to match the bottom side rotations. I think this must be some subtle detail, but in my first tries the correction for the bottom side wasn't consistent ...

radiolyc, 2023-11-13:
@set-soft thanks for answer.

Yeah, now mostly I'm using it only for top side component boards, but double side boards should be also useful.

Also Fabrication Tool has position offset correction field, which is also super useful, because many **KiCad** footprints have just different from **EasyEDA** center and offset corrections (rotation and shifting) give a chance to get correct parts placement after uploading manufacturing files.

set-soft, 2023-11-14:
Hi @radiolyc !
I think I found what's going on with the bottom layer, KiCad is mirroring the components, while JLCPCB isn't. I hope to have a fix for KiBot today or tomorrow. 

About the position feature: Looking at the plugin you mention it applies a rotation to the offset using the original KiCad angle. This is strange to me. Do you have a small example where this adjust is needed? So I can verify the result is correct.

A small example with a couple of components will be enough, just the schematic with the components, no need for connections, and the PCB with the footprints and board edge. The example should include the fields in question with at least one rotation and one offset. This will help to verify the implementation.

set-soft, 2023-11-15:
Hi @radiolyc !
The current *dev* branch should implement it. I just need to verify it works as expected

radiolyc, 2023-11-16:
@set-soft Here is small example of _Fabrication Tool_.
[image]
[fabrication_tool_test_prohect.zip](https://github.com/INTI-CMNB/KiBot/files/13382280/fabrication_tool_test_prohect.zip)

set-soft, 2023-11-17:
Hi @radiolyc !

Ok, now you'll get the same results using KiBot. Looking at your example I found the offsets uses the opposite sign. 
Note that you could get different results for bottom components, but this is because the *Fabrication Toolkit* is applying the bottom rotations wrongly. JLCPCB changed at least three times the way bottom rotation must be applied.

set-soft, 2023-11-17:
I'm closing this issue because is now implemented. Any errors in the implementation should be opened as a new issue.

closed 2023-11-17 by set-soft