INTI-CMNB/KiBot #512: [HOW] How can I add my 3D components library to render_3d

Kampi, 2023-10-31:
**What do you want to achieve?**
3D rendering with all components. I have a KiCad design with a couple of components with 3D models attached to them. The path of the 3D model is added with an environment variable, i. e.

```sh
KICAD_LIBRARY:3D/ST/VFQFN68-8x8mm_STM.stp
```

where `KICAD_LIBRARY` is the path to the KiCad library dir from me:

[image]

These footprints are missing in the rendering at the moment and KiBot is printing

```sh
daniel@PC-Daniel:~/SoilMoistureSensor/project$ kibot
Using config file: soilmoisturesensor.kibot.yaml
Using SCH file: SoilMoistureSensor.kicad_sch
- '3D rendering' (3D rendering) [render_3d]
WARNING:(W009) KiCad config without environment.vars section (kibot - config.py:356)
WARNING:(W047) Missing 3D model for IC1: `/home/daniel/Projects/SoilMoistureSensor/project/:KICAD_LIBRARY:3D/ST/MLPF-WB55-01E3_STM.stp` (kibot - out_base_3d.py:503)
WARNING:(W047) Missing 3D model for IC2: `/home/daniel/Projects/SoilMoistureSensor/project/:KICAD_LIBRARY:3D/ST/VFQFN68-8x8mm_STM.stp` (kibot - out_base_3d.py:503)
WARNING:(W047) Missing 3D model for Y2: `/home/daniel/Projects/SoilMoistureSensor/project/:KICAD_LIBRARY:3D/NDK/XTAL_NX2012SA_NDK.stp` (kibot - out_base_3d.py:503)
WARNING:(W047) Missing 3D model for IC4: `/home/daniel/Projects/SoilMoistureSensor/project/:KICAD_LIBRARY:3D/Broadcom/APDS-9306-065.stp` (kibot - out_base_3d.py:503)
WARNING:(W047) Missing 3D model for Y1: `/home/daniel/Projects/SoilMoistureSensor/project/:KICAD_LIBRARY:3D/NDK/NX2016SA_NDK.stp` (kibot - out_base_3d.py:503)
WARNING:(W098)  9 3D models downloaded or cached (kibot - out_base_3d.py:516)
Found 7 unique warning/s (7 total)
```

[image]

**Do you have some PCB/Schematic to use as example?**

[SoilMoistureSensor.kicad_pcb.txt](https://github.com/INTI-CMNB/KiBot/files/13218714/SoilMoistureSensor.kicad_pcb.txt)

**Do you have some configuration file (.kibot.yaml) that you are using?**

```yaml
kibot:
  version: 1

global:
  out_dir: "../production/Export-%D"
  environment:
    models_3d: /home/daniel/KiCad

outputs:
  - name: "3D rendering"
    comment: "3D rendering"
    type: render_3d
    dir: docs/images
    options:
      ray_tracing: true
      rotate_x: 3
      rotate_z: -2
      output: "PCB_Rendering.%x"e
```

**Environment (please complete the following information):**
Where are you running KiBot:
- Ubuntu 22.04 in WSL2

set-soft, 2023-10-31:
If you are really using an environment variable you should use ${KICAD_LIBRARY}

But I'll suggest using a path relative to your project and then use the KIPRJMOD variable defined by KiCad. 

Note that you can also use Linux/Windows symlinks to make a link to another folder in your filesystem.

I'm not sure what exactly are you using, the syntax looks like an alias (KiCad 6, but deprecated in 7), but you also have a : before the name ...

Kampi, 2023-11-01:
Hi,

I´m using KiCad 7. Exporting an environment variable doesn´t work:

```sh
daniel@PC-Daniel:~/Projects/SoilMoistureSensor/project$ export KICAD_LIBRARY=/home/daniel/KiCad
daniel@PC-Daniel:~/Projects/SoilMoistureSensor/project$ echo ${KICAD_LIBRARY}
/home/daniel/KiCad
daniel@PC-Daniel:~/Projects/SoilMoistureSensor/project$ kibot
Using config file: soilmoisturesensor.kibot.yaml
Using SCH file: SoilMoistureSensor.kicad_sch
- '3D rendering' (3D rendering) [render_3d]
WARNING:(W009) KiCad config without environment.vars section (kibot - config.py:356)
WARNING:(W047) Missing 3D model for IC1: `/home/daniel/Projects/SoilMoistureSensor/project/:KICAD_LIBRARY:3D/ST/MLPF-WB55-01E3_STM.stp` (kibot - out_base_3d.py:503)
WARNING:(W047) Missing 3D model for IC2: `/home/daniel/Projects/SoilMoistureSensor/project/:KICAD_LIBRARY:3D/ST/VFQFN68-8x8mm_STM.stp` (kibot - out_base_3d.py:503)
WARNING:(W047) Missing 3D model for Y2: `/home/daniel/Projects/SoilMoistureSensor/project/:KICAD_LIBRARY:3D/NDK/XTAL_NX2012SA_NDK.stp` (kibot - out_base_3d.py:503)
WARNING:(W047) Missing 3D model for IC4: `/home/daniel/Projects/SoilMoistureSensor/project/:KICAD_LIBRARY:3D/Broadcom/APDS-9306-065.stp` (kibot - out_base_3d.py:503)
WARNING:(W047) Missing 3D model for Y1: `/home/daniel/Projects/SoilMoistureSensor/project/:KICAD_LIBRARY:3D/NDK/NX2016SA_NDK.stp` (kibot - out_base_3d.py:503)
WARNING:(W098)  9 3D models downloaded or cached (kibot - out_base_3d.py:516)
Found 7 unique warning/s (7 total)
daniel@PC-Daniel:~/Projects/SoilMoistureSensor/project$
```

The footprints are still missing and it looks that the path isn´t correct, because the path is expanded with the root dir `/home/daniel/Projects/SoilMoistureSensor/project/`.

Do I understand it right that using `KIPRJMOD` requires a change of all pathes inside the KiCad PCB?

Kampi, 2023-11-05:
@set-soft 

I guess I found the issue. It looks like that KiBot doesn´t handle `KICAD_LIBRARY:3D/` not correctly. After changing it to `${KICAD_LIBRARY}/3D/` it works with

```
kibot:
  version: 1

global:
  out_dir: "../production/Export-%D"
  date_format: "%Y-%m-%d"
  environment:
    extra_os:
      - name: KICAD_LIBRARY
        value: "/home/daniel/KiCad"

outputs:
  - name: "3D rendering"
    comment: "3D rendering"
    type: render_3d
    dir: docs/images
    options:
      ray_tracing: true
      rotate_x: 3
      rotate_z: -2
      output: "PCB_Rendering.%x"
```

Kampi, 2023-11-06:
Hi @set-soft,

yes it worked so far. But I changed the path for the 3D model now in all of my libraries, so there is no issue anymore :)