INTI-CMNB/KiBot #515: [BUG] EasyEDA Can cause pipeline failures

gcormier, 2023-11-07:
I was wondering why my pipeline was failing. I now see that it's probably due to API errors.

I have run this same pipeline now over and over for testing - some will fail, some will succeed.

[image]

https://github.com/gcormier/megadesk_companion/blob/main/.github/workflows/fabrication.yml

1. In the least, perhaps a better error can be displayed? (eg. "Could not download 3D model from EasyEDA API. Please try again later.")
2. I assume this is about downloading 3D models - it would be nice to be able to disable this. (I realize in this case, there is a render step in the pipeline, so it's probably that the download failure *should* actually cause the pipeline to fail, as otherwise we would render it with a missing component.)

Output from one of the failed:
```
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/requests/models.py", line 9[71](https://github.com/gcormier/megadesk_companion/actions/runs/6778143461/job/18423101077#step:9:72), in json
    return complexjson.loads(self.text, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/json/__init__.py", line 346, in loads
    return _default_decoder.decode(s)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/json/decoder.py", line 337, in decode
    obj, end = self.raw_decode(s, idx=_w(s, 0).end())
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/json/decoder.py", line 355, in raw_decode
    raise JSONDecodeError("Expecting value", s, err.value) from None
json.decoder.JSONDecodeError: Expecting value: line 1 column 1 (char 0)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/bin/kibot", line 33, in <module>
    sys.exit(load_entry_point('kibot==1.6.3', 'console_scripts', 'kibot')())
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/kibot/__main__.py", line 520, in main
    generate_outputs(outputs, args.target, args.invert_sel, args.skip_pre, args.cli_order, args.no_priority,
  File "/usr/lib/python3/dist-packages/kibot/kiplot.py", line 527, in generate_outputs
    _generate_outputs(outputs, targets, invert, skip_pre, cli_order, no_priority, dont_stop)
  File "/usr/lib/python3/dist-packages/kibot/kiplot.py", line 517, in _generate_outputs
    run_output(out, dont_stop)
  File "/usr/lib/python3/dist-packages/kibot/kiplot.py", line 418, in run_output
    out.run(get_output_dir(out.dir, out))
  File "/usr/lib/python3/dist-packages/kibot/out_base.py", line 197, in run
    self.options.run(self.expand_filename(output_dir, output))
  File "/usr/lib/python3/dist-packages/kibot/out_step.py", line 84, in run
    board_name = self.filter_components()
                 ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/kibot/out_base_3d.py", line 550, in filter_components
    if self.download_models(force_wrl=force_wrl, all_comps=all_comps) or dnp_removed:
       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/kibot/out_base_3d.py", line 498, in download_models
    replace = self.try_download_easyeda(m3d.m_Filename, full_name, downloaded, sch_comp, lcsc_field)
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/kibot/out_base_3d.py", line 230, in try_download_easyeda
    replace = download_easyeda_3d_model(lcsc_id, self._tmp_dir, fname)
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/kibot/EasyEDA/easyeda_3d.py", line 317, in download_easyeda_3d_model
    data = api.get_cad_data_of_component(lcsc_id)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/kibot/EasyEDA/easyeda_3d.py", line 52, in get_cad_data_of_component
    cp_cad_info = self.get_info_from_easyeda_api(lcsc_id=lcsc_id)
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/kibot/EasyEDA/easyeda_3d.py", line 43, in get_info_from_easyeda_api
    api_response = r.json()
                   ^^^^^^^^
  File "/usr/lib/python3/dist-packages/requests/models.py", line 9[75](https://github.com/gcormier/megadesk_companion/actions/runs/6778143461/job/18423101077#step:9:76), in json
    raise RequestsJSONDecodeError(e.msg, e.doc, e.pos)
requests.exceptions.JSONDecodeError: Expecting value: line 1 column 1 (char 0)
```

set-soft, 2023-11-07:
Hi @gcormier !

Fails to download a 3D model should be non-critical errors.

Do you know the LCSC id of the component? Is this a temporal error? (the ID is printed in the debug output).

The a98b46195c4f0652905df4a22c794e2428b027c7 patch should fix it. Should be available in `v2_dk7` action.

closed 2023-11-07 by set-soft

gcormier, 2023-11-07:
Thanks for fixing!

I can try to get the ID later, although I suspect if LCSC has fixed the API then it's unlikely I can reproduce this now.

commit a98b4619 (2023-11-07): [3D models][Download] Catch Easy EDA errors

- Print an error and continue
Fixes #515