INTI-CMNB/KiBot #518: Treat DRC errors as warnings

andersruneson, 2023-11-15:
I think this might be related to issue #374 .

We run a CI/CD pipeline where we set '--skip-pre run_drc' when running a commit (so that the pipeline even when we are not done with the layout) but we run the drc for tagged builds (a release). 
The issue then comes when we have a few "approved" drc-errors that we need to filter with regex; to be able to write a good regex we need to know the drc-errors first. So we need to tag a release just to get the drc-errors. 

It would be really nice if we could have all drc-errors as warnings when running a commit-build so they are available in the build-log.

set-soft, 2023-11-16:
Hi @andersruneson !

Yes, #374 mentions the fact that we need to allow a better control on what is reported by the DRC. But you can just always run KiBot for DRC in your regular workflow and ignore the error level, just to log the current errors. For this you must run KiBot directly, not using the GitHub Action.

You can ignore a shell command error adding `|| true` see [this](https://stackoverflow.com/questions/11231937/bash-ignoring-error-for-a-particular-command)

You can also:
- Run KiBot locally. Using WSL or docker for Windows.
- Run the DRC manually and save the DRC report, then open the DRC report and see which errors do you want to filter.
- Use DRC exclusions and enable `drc_exclusions_workaround`, then you don't need to add filters, just choose the excluded errors from KiCad

andersruneson, 2023-11-16:
Right, just ignoring the result with || true might work. We won't get email from gitlab about actual errors anymore, but that might be fine.
Thanks!

set-soft, 2023-11-17:
> We won't get email from gitlab about actual errors

Yes, but just run it in a job dedicated to collect the errors you might want to filter, not in the main job. Use other jobs to generate all the needed stuff and let them inform about any error.

set-soft, 2024-04-05:
Today I added two new preflights: `erc` and `drc`. They replace `run_erc` and `run_drc`.
The new preflights are for KiCad 8 only, and also experimental.
But both of them have options to just report the issues and continue (`dont_stop`).

closed 2024-04-05 by set-soft