INTI-CMNB/KiBot #521: [HOW] Exclude components from BOM

asereq, 2023-11-28:
Hi!

First of all, thanks so much for the work! This tool is awesome!

**What do you want to achieve?**
I would like to create a BOM excluding some components. 

**Additional context**
I am using KiCAD 7 over linux.
I was trying two approaches without success:
1. Using the "Do not fit" checkbox on KiCad 7. The components in the BOM appears as DNP but there are listed. I want to exclude them, I tried with the 'exclude_filter' but I don't found how to create the right filter. For example, I saw the "Footprint Populate" column (and I use it) but it appears to not be present on the filter engine (I used the -vv debug flag).
2. Another approach was to create a variant (great work, again, the idea is great!) using the kibom style (+VARIANT0, -VARIANT1, etc). And, again, I can set that a component could be listed as DNP but I can't exclude it from the BOM.

So, how can I proceed to exclude the DNP components from the BOM? I would like to avoid the work of tagging as 'DNP' on the Config column. If I am going to add a Config column I would like to use all the power of variants (not only the DNP tag), but that approach is not working for me.

Thanks again!

set-soft, 2023-11-28:
Hi @asereq !

Are you using the internal BoM? (output type *BoM*) Did you try the *ignore_dnf* option?

Note that generic filters support the *exclude_not_on_board* to filter components.

If you still having problems to apply a correct filter please post a small test case (schematic, PCB and KiBot config) so I can see what's wrong.

asereq, 2023-11-28:
Oooohhhh!!!! Yeah!!!!!
the option "ignore_dnf" is the key! I was interpreting it as "ignore the dnf flag and put all the components in the bom" but the right thing is "ignore the components marked as dnf", great! (hours and hours of work in that, I was thinking that the problem was on the kicad 7 ported but not).
Thanks, the problem was mine, the documentation is well "Exclude DNF (do not fit) components".
However maybe the option could be called "exclude_dnf" to match the other exclude options.

Thanks again and sorry for the confusion.

closed 2023-11-28 by asereq