INTI-CMNB/KiBot #526: [HOW] How to produce panelized gerbers and other recursive actions

CraigBurden, 2023-12-04:
Hi there!

I have the `panelize` output working very nicely, generating a panelized PCB and placing the design in a sub folder, I'd really like to generate gerbers for this panel.

Is this possible? And more generally is it possible to run outputs on other outputs like this? I think this is likely one of the only examples of this need though

Thanks in advance!

CraigBurden, 2023-12-05:
Hi @set-soft,

I am not sure that I agree that it would be problematic, although of course you would know better than I do. What is your reason for saying that? Also I don't see how it would be slower than running them separately? 

My reasoning for wanting this is that I design a single PCB, which is then panelized and ultimately the panelized gerbers are the output I need as that is what I produce. Yes of course I can just add an additional KiBot config file somewhere and run that afterward in a CI pipeline, but I also use this outside of CI, so running two things isn't ideal.

To generalise the functionality, it would be nice to be able to define an input file/project for each output type. Assuming that the outputs are run in the sequence defined by the config file, this small change would allow a multitude of similar things like generating gerbers, drills, renders, 3d models etc on panelized boards but also multi-board projects etc, all from the comfort of a single config file.

set-soft, 2023-12-05:
> I am not sure that I agree that it would be problematic, although of course you would know better than I do. 

This is because you don't know how KiCad works ;-)

> What is your reason for saying that? 

Even when the KiCad API uses classes doesn't really support more than one project loaded at the same time. A lot of settings and values are static and global. Switching from a PCB and its variants usually works because they are almost the same thing. But with the panel ... is much more delicated.

> Also I don't see how it would be slower than running them separately?

Because KiCad will be loading and unloading the panel more than once. The panel is many times bigger than the PCB. But if you run a separate KiBot for the panel, the panel will be un/loaded a few times, or perhaps never unloaded.

> so running two things isn't ideal.

This problem has many solutions: makefiles, shell scripts, etc.

> To generalise the functionality, it would be nice to be able to define an input file/project for each output type. 

I have a better idea, but implementing it is a lot of work. You define "inputs", the normal project, some variants, a panel, etc. KiBot generates them. And you define which outputs are applied to each "input". Nice, but needs too much work.

> Assuming that the outputs are run in the sequence defined by the config file,

This is an option, but usually you don't do it. Outputs has priorities. 

> this small change would allow a multitude of similar things like generating gerbers, drills, renders, 3d models etc on panelized boards but also multi-board projects etc, all from the comfort of a single config file.

Remmember configuration files are modular, you can import stuff and also pass parameters to the imported configs. So you don't need to repeat stuff.

memen45, 2024-08-27:
I know this issue is closed for a while now, but I actually have the same question. If you'd prefer to create a new issue, just let me know.

Would you advise to create different kibot.yaml files for the panel? Is it even possible to run kibot when there is no schematic file present?

set-soft, 2024-08-27:
> Would you advise to create different kibot.yaml files for the panel? 

Yes, this is the best option.

> Is it even possible to run kibot when there is no schematic file present?

Yes, you can do it, unless you need filters and/or variants. In the worst case you can use the schematic for the original project.

memen45, 2024-08-27:
I actually do use variants. And filters for rotate_footprint when obtaining the position files.

In the main KiBot.yaml:
```
- outputs: 
  - type: panelize
    dir: 'panel'
```
(Without using a variant)

In the panel directory, KiBot.yaml:
```
- filters:
  <filters>
- outputs:
  - type: gerber
    variant: '@VARIANT@'
  - etc. etc.
```
(Using variants, and filters, outputting pcb_print, bom, position, etc.)

Is this the workflow you'd propose here?

set-soft, 2024-08-27:
> Is this the workflow you'd propose here?

Yes

memen45, 2024-08-27:
Thanks a lot for your fast responses and help! It works (almost) perfectly now. Thanks also for creating and maintaining this super useful tool!