INTI-CMNB/KiBot #528: [BUG] Docker container image will always fail with a W009 error

oliv3r, 2023-12-14:
When running kibot from the kicad7-full docker container, it will obviously not find the users `kicad_common.json`. Instead, kicad will generate a basic `kicad_common.json` within the container, but the environment section will be non-existant. As such, kibot will fail with a
`WARNING:(W009) KiCad config without environment.vars section` error.

While one way to deal with this, is to patch kicad to always create the environment.vars section, alternatively, kibot should treat the fact that the section is missing the same as if the section was null, instead of warning that there was no such section. See https://github.com/INTI-CMNB/KiBot/blob/8bc7ea6324a97b73d489ac850cad87e6110f4635/kibot/kicad/config.py#L350 for the function in question.

Not sure why the kicad6 environment loader is being used on a kicad7 project though ;)

set-soft, 2023-12-14:
> When running kibot from the kicad7-full docker container, it will obviously not find the users `kicad_common.json`. 

Not unless you mount a folder containing it ...

> Instead, kicad will generate a basic `kicad_common.json` within the container, but the environment section will be non-existant. As such, kibot will fail with a `WARNING:(W009) KiCad config without environment.vars section` error.

Nope, KiBot won't fail, this is just a warning so you know that no KiCad environment variables are available. Note that older KiCad versions stored its default variables in this section. KiBot supports KiCad 5, 6 and 7.

> While one way to deal with this, is to patch kicad to always create the environment.vars section,

Which will restore the old behavior ...

> alternatively, kibot should treat the fact that the section is missing the same as if the section was null, instead of warning that there was no such section. 

Why? If we do this you'll never know. You are assuming KiBot is always used from the docker image AND the user never mounts a dir containing the desired KiCad config. The most common case for GitHub Actions, but not the only use.

> See
> 
> https://github.com/INTI-CMNB/KiBot/blob/8bc7ea6324a97b73d489ac850cad87e6110f4635/kibot/kicad/config.py#L350
> 
> for the function in question.
> Not sure why the kicad6 environment loader is being used on a kicad7 project though ;)

Simply: because they didn't change the format. Thanks god they aren't changing all file formats on each major release. Lamentably they didn't choose one format for all the files (already discussed), IMHO they should use something like JSON or YAML for *all* the files (project, schematic, PCB, config files, lib tables, etc.). KiCad 6 release was a big opportunity to achieve it, and drop the ridiculous lisp like format ...

closed 2023-12-14 by set-soft

oliv3r, 2023-12-14:
Thanks @set-soft,

I've updated my post above to have all information in the same spot, but will reply inline anyway :)

> > When running kibot from the kicad7-full docker container, it will obviously not find the users `kicad_common.json`.
> 
> Not unless you mount a folder containing it ...
But when running in a CI/CD environment, which is one of the main KiBot scenario's, this is not an option is it? Can't ask each user to upload their config file for each pipeline run :)

> Nope, KiBot won't fail, this is just a warning so you know that no KiCad environment variables are available. Note that older KiCad versions stored its default variables in this section. KiBot [...]

So this warning is only applicable to kicad 5 (and maybe 6?)

But that's also how you setup your pipeline, If you want to fail (exit -1) kibot on any warning or error, (or grep the output for warnings to go into an 'allow failure' mode, you still end up with an orange or red pipeline.

And then, if the only solution is to have a kibot filter
```yaml
global:
  filters:
    - number: 9
      regex: 'KiCad config without environment.vars section'
```
that's not that 'clean' either.

> Which will restore the old behavior ...
>
>
> Why? If we do this you'll never know. You are assuming KiBot is always used from the docker image AND the user never mounts a dir containing the desired [...]

Exactly, which results in always producing the warning?

> Simply: because they didn't change the format. Thanks god they aren't changing all file formats on each major release. Lamentably they didn't choose one format for all the files (already discussed), [...]
yeah ;) so I suppose the function name could be more generic, or less kicad6 specific to avoid confusion there :p