INTI-CMNB/KiBot #529: [BUG] KiBot and ERC on non-existant output directory runs forever

oliv3r, 2023-12-14:
Using the docker kicad7 full image, and defining `global.output: 'something/'`, causes kibot to try to save to a non-existant folder. Reviewing the video, we see that the save dialog of the erc pops up, tries to save the report, but then comes back to the result page waiting ...

Since we launch the erc with eeschema_do and the correct path, we can also ensure the path exists before launching eeschema_do, and maybe even create the folder in python.

set-soft, 2023-12-14:
Hi @oliv3r !
You are misusing the options. The `output` is for the file name, the path should go to `dir`.

set-soft, 2023-12-14:
I found the real problem: KiCad 7 is applying a nasty Windows behavior, assuming the user is stupid, and adding extensions to file names when you ask otherwise. So KiCad 7 can't create an ERC log file named "something". This is a bug in KiCad, I'll look for a workaround.

set-soft, 2023-12-14:
The above patch implements a woraround so you can choose any name, even when KiCad 7  can't create report files with arbitrary names

closed 2023-12-14 by set-soft

oliv3r, 2023-12-14:
Ah, i miss-undertsood the option. It was under 'global' so a single filename is a bit weird. As it would mean a single filename for each operation? I'll look into the 'path' option, but `global.output` to me, reads like 'all generated output goes here'...

Thanks for resolving the bug though :)

oliv3r, 2023-12-25:
@set-soft I'm not sure the issue is completely resolved. I pushed by basic stuff into a temporary git repository, to get the pipeline going [0]. But I see 10 minute timeouts on erc. I pulled the video from the pipeline, and it tries to save the erc report using a `#` as a filename. After hitting the 'save' button, it just sits there indefinitely, well till it gets killed. (Why can't we run the erc via the cli/API yet ... *sigh*).

[image]

Here's my kibot yaml file, which is very basic, as I want only the erc to run.
```yaml
kibot:
  version: 1

global:
  dir: 'reports'
  restore_project: true

preflight:
  erc_warnings: true
  run_drc: false
  run_erc: true
  set_text_variables:
    - name: 'VCS_SCH_TIMEDATE'
      command: 'date -d @$(git log -1 --format="%at" "${KIBOT_SCH_NAME}") -Iseconds'
    - name: 'VCS_SCH_HASH'
      command: 'git log -1 --format="%H" "${KIBOT_SCH_NAME}"'
      before: 'Git <'
      after: '>'
    - name: 'VCS_TAG'
      command: 'git describe --always --dirty'
```

[0]: https://gitlab.com/olliver/lapod/

set-soft, 2023-12-26:
Without log files is impossible to tell something. Note that this actually works in GitHub CI/CD and locally, so I can't discard this is a problem with GitLab CI/CD

oliv3r, 2023-12-31:
Fair enough, and your fix seems to cover it for the most part. Since the issue is resolved for the time being, I'm not seeing that problem anymore, its fine for now I suppose. But as I wrote in comment XX, I think this is a broader generic issue that happens more often.

Having it working locally and in github CI isn't really a good measure. I think it depends a lot on how the task is setup to be run, which variables are set. The github action works with a very tight boundary, e.g. 'you run the pipeline this way, and only this way'. Locally, I run only in docker containers, so _should_ be the same as in the pipeline, but I also there found some discrepancies, weird stuff for the most part though. But I'll keep opening bug reports and hopefully we can improve the quality :D Though I think some bugs simply relate to the fact that we are automating the GUI version, rather having a reliable API :(

oliv3r, 2024-01-04:
I think this is the same as #544 as well.

commit 4745bacc (2023-12-14): [ERC][KiCad 7][Fixed] Problems when creating a report without ext

- Workaround for KiCad 7 explicitly creating a different file

Fixes #529