INTI-CMNB/KiBot #530: [BUG] Schematic DNP and diff do not work great together in the output as both are red.

oliv3r, 2023-12-16:
When generating a pdf of the schematic, in red and green mode, it's hard to judge if a DNP part is red because something got removed, or red because it's marked as DNP.

Since everything else is black and white, the DNP cross should be as well, because the cross wasn't added or removed.

See for example:
[image]

set-soft, 2023-12-18:
Hi @oliv3r !
Please be more specific, an example could also help.
I guess this is KiCad 7, and you are using the *diff* output, if this is the case it might be a KiCad 7 issue failing to honor the monochrome option.

oliv3r, 2023-12-19:
@set-soft apologies, I am indeed running the kicad 7 full docker container, and am generating a pdf diff of the schematic. How do you print? If I print using file -> print and then print to pdf file, the black/white flag from the UI is honored.

Here's the schematic and the pdf
[lapod.zip](https://github.com/INTI-CMNB/KiBot/files/13720866/lapod.zip)

set-soft, 2023-12-20:
> @set-soft apologies, I am indeed running the kicad 7 full docker container, and am generating a pdf diff of the schematic. How do you print? If I print using file -> print and then print to pdf file, the black/white flag from the UI is honored.

But I need information about the options you are using. Try to create a small example where you can reproduce it. You should use files, git commits, unless you also include a small repo. Note that you can create a repo at GitHub with the files and just link it here. 

Giving the fact that the crosses are red it can be a diff. The 6c336371bd889ac920ef682cf756c6f130ca714a patch addressed a bug that could potentially generate this problem, but I can't be sure without an example.

> Here's the schematic and the pdf [lapod.zip](https://github.com/INTI-CMNB/KiBot/files/13720866/lapod.zip)
> 
> Btw, I have responded to a few tickets you already closed. Is that okay, or do you have other means to be reached?

Yes, is ok. The best mechanism is using the issues, so everybody can track it.

oliv3r, 2023-12-25:
> But I need information about the options you are using. Try to create a small example where you can reproduce it. You should use files, git commits, unless you also include a small repo. Note that [...]

Right, I'm 90% certain that this is not a diff thing, as that part has 'always' been a DNP part. even before I even added the file to my local repo. I'll zip up my worktree and put it here, once I get my other pipeline issues sorted. Give me a few :)

set-soft, 2024-01-09:
Hi @oliv3r ! Any news about it?

oliv3r, 2024-01-15:
> Hi @oliv3r ! Any news about it?

No! I have been so focused on everything else related to the pipeline, this has been left behind for a bit, Also, this was only triggered for me outside of the pipeline "current state vs HEAD", which requires a different yaml file; but now that I'm getting the final touches into my pipeline; I'll look at this again soon!

set-soft, 2024-02-01:
Ok, I'll assume that 6c336371bd889ac920ef682cf756c6f130ca714a fixed it.

If you find a simple test case where this problem is visible please open a new issue.

closed 2024-02-01 by set-soft