INTI-CMNB/KiBot #531: [BUG] Raytracing in kicad 7 full docker container produces images without traces, poor quality

oliv3r, 2023-12-17:
Using the following config:

```yaml
kibot:
  version: 1

global:
  environment:
    models_3d: '${KIPRJMOD}/packages3D'
  filters:
    - number: 9
      regex: 'KiCad config without environment.vars section'

outputs:
  - name: 'render_3d_front'
    comment: "Exports the image generated by KiCad's 3D viewer."
    type: 'render_3d'
    dir: 'output/render_3d/'
    options:
      auto_crop: true
      background1: '#66667f'
      background2: '#cccce5'
      board: '#332b16'
      clip_silk_on_via_annulus: true
      copper: '#8b898c'
      dnf_filter: '_none'
      download: false
      height: 1440
      highlight: []
      move_x: 0
      move_y: -2
      no_smd: false
      no_tht: false
      no_virtual: false
      orthographic: false
      pre_transform: '_none'
      ray_tracing: true
      realistic: true
      rotate_x: -5
      rotate_y: -1
      rotate_z: -3
      show_adhesive: false
      show_board_body: true
      show_comments: false
      show_components: all
      show_eco: false
      show_silkscreen: true
      show_soldermask: true
      show_solderpaste: true
      show_zones: true
      silk: '#d5dce4'
      solder_mask: '#208b47'
      solder_paste: '#808080'
      subtract_mask_from_silk: true
      transparent_background: false
      transparent_background_color: '#00ff00'
      transparent_background_fuzz: 15
      variant: ''
      view: 'front'
      wait_render: -600
      width: 1440
      zoom: 4
```

I get an image that does not contain any PCB traces. Doing it in my local kicad setup, the PCB renders perfectly.

[image]
[image]

If i turn of both realistic and raytracing, i get traces, but the soldermask is red/pink
[image]

and if I turn on realistic only, i get a green PCB, but without traces.

set-soft, 2023-12-18:
The lack of traces is because you are using an opaque solder mask, so you can't see the traces. Specify a correct alpha channel to avoid this.

The quality difference is simply because the KiCad result depends on the hardware accelerator. Isn't rendered by KiCad itself.

And yes, if you are looking for quality forget about using KiCad for rendering.

closed 2023-12-18 by set-soft

oliv3r, 2023-12-18:
> The lack of traces is because you are using an opaque solder mask, so you can't see the traces. Specify a correct alpha channel to avoid this.
>
> The quality difference is simply because the KiCad [...]

Thanks @set-soft while not intuitive (and I got it straight from an example) what about the odd colors?

set-soft, 2023-12-20:
> Thanks @set-soft while not intuitive (and I got it straight from an example)

Tell me which example, so I can fix it. This is how KiCad currently works, I think KiCad 6 introduced transparency. The example is most probably a KiCad 5 example.

> what about the odd colors?

Do you mean the odd colors in the "non realistic" mode? I think this is what KiCad does when realistic is disabled! (uses the internal layer colors).

oliv3r, 2024-01-17:
So it is deffinatly related to 'realistic mode', but the colors are weird, cyan and purple. Though I suppose that could happen if colorr layers where mixed, but doesn't make sense at all.

I haven't even seen the 'realistic' option in the 3D preview GUI, I think it's enabled by default.