INTI-CMNB/KiBot #532: [BUG] Custom worksheet causes warnings

oliv3r, 2023-12-17:
With the kicad 7 full docker image, using my own worksheet, kibot complains about SHEETPATH being unknown and the version being incompatible. The version in the wks was made using kicad 7 and is dated feb 2022, so not that new.

```
WARNING:(W086) Unsupported worksheet version, loading could fail (kibot - worksheet.py:498)
WARNING:(W082) Unknown text variable `SHEETPATH` (kibot.gs - gs.py:407)
```

The outputs section I got from an kibot example, which looks like:
```yaml
  - name: 'pdf_pcb_print'
    comment: 'Exports the PCB to the most common exchange format. Suitable for printing.'
    type: 'pcb_print'
    dir: 'output/pdf'
    options:
      force_edge_cuts: true
      keep_temporal_files: false
      sheet_reference_layout: '../git.kicad_wks'
      scaling: 2.0
      pages:
        - layers: [ 'F.Paste', 'F.Adhes', 'Dwgs.User', 'F.Fab' ]
          sheet: 'Fabrication layers'
        - layers:
            - layer: 'F.Cu'
            - layer: 'F.Mask'
              color: '#14332440'
            - layer: 'F.SilkS'
            - layer: 'Dwgs.User'
          sheet: 'Top layer'
        - layers: [ 'In1.Cu', 'Dwgs.User' ]
          sheet: 'Inner1 plane'
        - layers: [ 'In2.Cu', 'Dwgs.User' ]
          sheet: 'Inner2 plane'
        - layers:
            - layer: 'B.Cu'
            - layer: 'B.Mask'
              color: '#14332440'
            - layer: 'B.SilkS'
            - layer: 'Dwgs.User'
          sheet: 'Bottom layer'
          mirror: true
```

I haven't really tweaked or modified it yet, but stumbled upon the error as I was going ...

[git.kicad_wks.zip](https://github.com/INTI-CMNB/KiBot/files/13697213/git.kicad_wks.zip)

set-soft, 2023-12-18:
Hi @oliv3r !
You still confuse *warning* and *error*. This is not an error, and this is why *somewhat work*. KiBot is just explaining you that it wasn't tested using v20220228 and hence you could find things like a newly defined variable, *SHEETPATH*. Which, yes, is irrelevant for a PCB.

> Which I think kicad sets to '/'.

In fact KiCad sets it to empty because this variable has meaning only for schematics.

Note that this format is poorly documented, official docs are [here](https://dev-docs.kicad.org/en/file-formats/sexpr-worksheet/), which says *Last Modified 2023-04-13*, but if you take a look at KiBot code you'll find comments like *Not documented 2022/04/15*, and it still not documented.

Now about the first empty page: This seems to be a bug in KiCad, take a look at the SVGs (*keep_temporal_files: false*), you'll see the scaled outputs are drawn out of the page. This is somehow fixed when the first SVG in the merged output is correctly inside the page. I have to figure out how to workaround it.  For your case: just manually include the *Edge.Cuts* layer (not forced).

set-soft, 2023-12-18:
The above patch solves the empty first page. Is a fix for KiBot, but doesn't solve the errors in computed coordinates seen in KiCad side, in a test using paste in almost all the PCB the coordinates made the PCB "invisible", your problem, without a real reason.

closed 2023-12-18 by set-soft

oliv3r, 2023-12-19:
@set-soft I'm not confusing errors and warnings, but I do set '-Werror' if you will, e.g. a pipeline run for a tagged release, is expected to be warning free. So warnings are 'just as bad'.

I suppose you could argue, that you confuse the severity of a warning, which could be 'INFO' :)

Anyway, ignoring the warning, I noticed one mistake I made, was that I thought I added my wks to my PCB, I did not, I only added it to my SCH. Removing the variable that sets the wks in the yaml file, and properly setting it in the pcb file, actually works, and the SHEETPATH variable IS used for the PCB, as the 'SHEET' variable gets put there. Perfectly fine for me! Strange though that it fails with an error if I supply it via kibot ...
Also removed the warning about the incorrect version ... more strangeness.

The first page thing, it was only empty if I had a single layer there. Having two layers 'removes' the problem. But using the manual edge cuts, if it works, is also fine by me. I suppose I triggered that bug by accident anyway, as I've decided to want something different on my first page now anyway :)

oliv3r, 2023-12-19:
@set-soft I've also created https://gitlab.com/kicad/services/kicad-dev-docs/-/issues/26 to track this upstream.

oliv3r, 2023-12-21:
> > @set-soft I'm not confusing errors and warnings, but I do set '-Werror' if you will, e.g. a pipeline run for a tagged release, is 
> I apply the following criteria: if I have to inform something that is quite possible to generate a problem then I name it warning. If you try to expand a variable that doesn't exist then this is a warning, not INFO. And I'm not talking about the severities, I'm saying you call _errors_ to messages reported as _warnings_.

I apply the following criteria; when something is run from the pipeline, there shall be no warnings or errors allowed. In very certain scenario's, a warning could be a `allow_failure: true` in gitlab pipeline terminology, resulting in an orange pipeline. But this should not be structural, we want green pipelines, not orange or reds.

In terms of a developer, I follow the criteria 'for development, warnings can be allowed, for production builds, treat all warnings as errors (`-Werror`).

You say though, warnings _could_ be a problem, but can be ignored. So that's where we differ, and that's fine; but now I need ways to deal with this :) For example, Warnings do not cause non-zero exit codes etc.

Though I suppose that discussion warrants its own bug report to talk about maybe? :)

set-soft, 2023-12-21:
> For example, Warnings do not cause non-zero exit codes etc.

And this is the case, warnings doesn't change the exit code.

> Though I suppose that discussion warrants its own bug report to talk about maybe? :)

If you want to propose a *-Werror* equivalent feature, then yes, open a separated issue.

oliv3r, 2024-01-04:
@set-soft Btw, considering the title 'support worksheets version 20220228, in combination with https://github.com/INTI-CMNB/KiBot/issues/532#issuecomment-1863555137, should we keep this issue open until this is fixed upstream?

set-soft, 2024-01-04:
I don't think they added anything important that isn't currently supported by KiBot. I think is safe to assume the "new" format is fully supported. If not people can open issues, this is quite common due to the lack of documentation.

set-soft, 2024-01-04:
v1.6.4 doesn't print a warning, assumes 20220228 is fully supported

commit 0972914f (2023-12-18): [PCB Print][Fixed] Missing edge cuts in the list of visible layers

- When forced
- This is needed and helps to avoid hitting a bug in KiCad

Fixes #532