INTI-CMNB/KiBot #533: [BUG] Front silkscreen cannot be pdf printed by itself

oliv3r, 2023-12-17:
Using the kicad full 7 docker image, rendering a PDF with just the silk screen, results in white output. Adding any other layer 'fixes' this.
The bottom silkscreen does not suffer from this problem.

[breakout_hdmi-assembly.pdf](https://github.com/INTI-CMNB/KiBot/files/13697400/breakout_hdmi-assembly.pdf)

```yaml
kibot:
  version: 1

global:
  environment:
    models_3d: '${KIPRJMOD}/packages3D'
  filters:
    - number: 9
      regex: 'KiCad config without environment.vars section'

preflight:
  run_drc: false
  run_erc: false
  set_text_variables:
    - name: 'VCS_PCB_DATE'
      command: 'date -d @$(git log -1 --format="%at" "${KIBOT_PCB_NAME}") -Idate'
    - name: 'VCS_PCB_TIMEDATE'
      command: 'date -d @$(git log -1 --format="%at" "${KIBOT_PCB_NAME}") -Iseconds'
    - name: 'VCS_PCB_HASH'
      command: 'git log -1 --format="%H" "${KIBOT_PCB_NAME}"'
      before: 'Git <'
      after: '>'
    - name: 'VCS_SCH_TIMEDATE'
      command: 'date -d @$(git log -1 --format="%at" "${KIBOT_SCH_NAME}") -Iseconds'
    - name: 'VCS_SCH_HASH'
      command: 'git log -1 --format="%H" "${KIBOT_SCH_NAME}"'
      before: 'Git <'
      after: '>'
    - name: 'VCS_TAG'
      command: '${CI_COMMIT_REF_NAME:-git describe --always --dirty}'
# restore_project: true

outputs:
  - name: 'pdf_sch_print'
    comment: 'Exports the schematic to the most common exchange format. Suitable for printing.'
    type: 'pdf_sch_print'
    dir: 'output/pdf'
    options:
      all_pages: true
      background_color: false
      frame: true
      monochrome: false
  - name: 'pdf_pcb_print'
    comment: 'Exports the PCB to the most common exchange format. Suitable for printing.'
    type: 'pcb_print'
    dir: 'output/pdf'
    options:
      force_edge_cuts: true
      keep_temporal_files: false
      sheet_reference_layout: '../git.kicad_wks'
      scaling: 1.5
      pages:
        - layers:
          - layer: 'F.Paste'
          - layer: 'F.Adhes'
          - layer: 'User.Comments'
          - layer: 'F.Fab'
          sheet: 'Fabrication layers'
        - layers:
          - layer: 'F.Cu'
          - layer: 'F.Mask'
            color: '#14332440'
          - layer: 'F.SilkS'
            color: '#f2eda1ff'
          - layer: 'User.Comments'
          sheet: 'Top layer'
        - layers:
          - layer: 'F.SilkS'
            color: '#00000000'
          - layer: 'User.Comments'
          sheet: 'Top silkscreen'
        - layers:
          - layer: 'F.Cu'
            color: '#c83434ff'
          - layer: 'User.Comments'
          sheet: 'Top copper'
        - layers:
          - layer: 'In1.Cu'
            color: '#7fc87fff'
          - layer: 'User.Comments'
          sheet: 'Inner1 copper'
        - layers:
          - layer: 'In2.Cu'
            color: '#ce7d2cff'
          - layer: 'User.Comments'
          sheet: 'Inner2 copper'
        - layers:
          - layer: 'B.Cu'
            color: '#4d7fc4ff'
          - layer: 'User.Comments'
          sheet: 'Bottom copper'
        - layers:
          - layer: 'B.SilkS'
            color: '#00000000'
          - layer: 'User.Comments'
          sheet: 'Bottom silkscreen'
          mirror: true
        - layers:
          - layer: 'B.Cu'
          - layer: 'B.Mask'
            color: '#14332440'
          - layer: 'B.SilkS'
            color: '#e8b2a7ff'
          - layer: 'User.Comments'
          sheet: 'Bottom layer'
          mirror: true
```
is my kibot yaml file.

set-soft, 2023-12-18:
This is the KiCad issue I mentioned in #532 

Note that KiBot from *dev* will work with the above configuration because you are forcing the PCB edge. But if you disable it you'll see the first page is empty. If you keep the temporals you'll see KiCad printed the important things outside the page.

One possible workaround is to always make the board edge visible, but this shouldn't be a solution.

set-soft, 2023-12-20:
Ok, I filled a KiCad [bug report](https://gitlab.com/kicad/code/kicad/-/issues/16418) lets see what the maintainers think about it. The *.Cu, *.SilkS and Edge.Cuts layers seems to work as expected, but the *.Fab, *.Paste, *.Adhes and user layers are scaled, but not centered. This moves the drawings outside the page for small boards, so you get an empty page.

set-soft, 2024-03-31:
Update: This bug is also present on KiCad 8.0.1 and the developers didn't even decide its priority.

oliv3r, 2024-09-27:
Hmm, I think i was using kicad 8.x for a while already when still on v1.7.0 and it was working 'fine' back then? I suppose for now, I'll just enable the 'always show edge' as a work-around, so I can stick to a tagged v1.8.0 ... especially if this is a KiCad bug :(