INTI-CMNB/KiBot #534: [BUG] Objects on multiple pages are not aligned identically.

oliv3r, 2023-12-17:
Using the kicad full 7 docker image, rendering a PDF with individual layers, causes some layers to be located differently then others.

[breakout_hdmi-assembly.pdf](https://github.com/INTI-CMNB/KiBot/files/13697400/breakout_hdmi-assembly.pdf)

```yaml
kibot:
  version: 1

global:
  environment:
    models_3d: '${KIPRJMOD}/packages3D'
  filters:
    - number: 9
      regex: 'KiCad config without environment.vars section'

preflight:
  run_drc: false
  run_erc: false
  set_text_variables:
    - name: 'VCS_PCB_DATE'
      command: 'date -d @$(git log -1 --format="%at" "${KIBOT_PCB_NAME}") -Idate'
    - name: 'VCS_PCB_TIMEDATE'
      command: 'date -d @$(git log -1 --format="%at" "${KIBOT_PCB_NAME}") -Iseconds'
    - name: 'VCS_PCB_HASH'
      command: 'git log -1 --format="%H" "${KIBOT_PCB_NAME}"'
      before: 'Git <'
      after: '>'
    - name: 'VCS_SCH_TIMEDATE'
      command: 'date -d @$(git log -1 --format="%at" "${KIBOT_SCH_NAME}") -Iseconds'
    - name: 'VCS_SCH_HASH'
      command: 'git log -1 --format="%H" "${KIBOT_SCH_NAME}"'
      before: 'Git <'
      after: '>'
    - name: 'VCS_TAG'
      command: '${CI_COMMIT_REF_NAME:-git describe --always --dirty}'
# restore_project: true

outputs:
  - name: 'pdf_sch_print'
    comment: 'Exports the schematic to the most common exchange format. Suitable for printing.'
    type: 'pdf_sch_print'
    dir: 'output/pdf'
    options:
      all_pages: true
      background_color: false
      frame: true
      monochrome: false
  - name: 'pdf_pcb_print'
    comment: 'Exports the PCB to the most common exchange format. Suitable for printing.'
    type: 'pcb_print'
    dir: 'output/pdf'
    options:
      force_edge_cuts: true
      keep_temporal_files: false
      sheet_reference_layout: '../git.kicad_wks'
      scaling: 1.5
      pages:
        - layers:
          - layer: 'F.Paste'
          - layer: 'F.Adhes'
          - layer: 'User.Comments'
          - layer: 'F.Fab'
          sheet: 'Fabrication layers'
        - layers:
          - layer: 'F.Cu'
          - layer: 'F.Mask'
            color: '#14332440'
          - layer: 'F.SilkS'
            color: '#f2eda1ff'
          - layer: 'User.Comments'
          sheet: 'Top layer'
        - layers:
          - layer: 'F.SilkS'
            color: '#00000000'
          - layer: 'User.Comments'
          sheet: 'Top silkscreen'
        - layers:
          - layer: 'F.Cu'
            color: '#c83434ff'
          - layer: 'User.Comments'
          sheet: 'Top copper'
        - layers:
          - layer: 'In1.Cu'
            color: '#7fc87fff'
          - layer: 'User.Comments'
          sheet: 'Inner1 copper'
        - layers:
          - layer: 'In2.Cu'
            color: '#ce7d2cff'
          - layer: 'User.Comments'
          sheet: 'Inner2 copper'
        - layers:
          - layer: 'B.Cu'
            color: '#4d7fc4ff'
          - layer: 'User.Comments'
          sheet: 'Bottom copper'
        - layers:
          - layer: 'B.SilkS'
            color: '#00000000'
          - layer: 'User.Comments'
          sheet: 'Bottom silkscreen'
          mirror: true
        - layers:
          - layer: 'B.Cu'
          - layer: 'B.Mask'
            color: '#14332440'
          - layer: 'B.SilkS'
            color: '#e8b2a7ff'
          - layer: 'User.Comments'
          sheet: 'Bottom layer'
          mirror: true
```
is my kibot yaml file.

set-soft, 2023-12-18:
The *individual_page_scaling* option says:

**Tell KiCad to apply the scaling for each page as a separated entity.
Disabling it the pages are coherent and can be superposed**

And this is what you want.

closed 2023-12-18 by set-soft

oliv3r, 2023-12-19:
@set-soft right, but why is this not the default when the option is omitted? It wasn't obvious to me from looking at the result that it was differently scaled, just differently aligned.

I know arguing about what makes most sense as default, to me, I would expect `individual_page_scaling` to be false as a default. It will give you consistent and predictable results. Especially since the scaling parameter is set to a fixed number, which would be applied to all pages. I'm not even sure what `individual_page_scaling` does with the fixed scaling anyway, so as a default, it makes more sense to set it to `false` imho.

set-soft, 2023-12-20:
> @set-soft right, but why is this not the default when the option is omitted? It wasn't obvious to me from looking at the result that it was differently scaled, just differently aligned.

Because the scaling is applied to each page individually, so you can get the best possible scale. 

> Especially since the scaling parameter is set to a fixed number, which would be applied to all pages.

Wrong. At output level the docs clearly states *Default scale factor* and you have a *scaling* option for each page.

> I'm not even sure what individual_page_scaling does with the fixed scaling anyway

Please read the docs, I think is quite clear.

> So we could re-use this issue to address that?

Nope. I give much more priority to compatibility. If you can propose a better name *individual_page_scaling* I can add an alias for it. And it