INTI-CMNB/KiBot #537: [FEATURE] compressed fab outputs should not keep source dirs

oliv3r, 2023-12-18:
While useful to retain the source files when compression of fab outputs is used, it is not needed in regular cases, also we can just unzip the compressed archives. As such, an option just as `keep_temporal_files` would be useful to be able to set to remove temporary files, having a cleaner output result.

```yaml
import:
- file: 'JLCPCB'
  definitions:
    _KIBOT_MANF_DIR: 'output/fab/JLCPCB'
    _KIBOT_POS_PRE_TRANSFORM: '[''_kicost_rename'', ''_rot_footprint'']'
```

set-soft, 2023-12-18:
The above patch moves the files.
Note that now the BoM and Position aren't part of the ZIP file.  This is because you must upload them separately.

closed 2023-12-18 by set-soft

oliv3r, 2023-12-19:
@set-soft makes sense, aren't the BOM and Position files used by the assembly service only anyway? I suppose an 'assembly' include would be nice for that at some point...

set-soft, 2023-12-20:
When you use JLCPCB you must upload the generated .ZIP first and then the BOM and CPL files. You need them separated. I recently found it while doing some adjusts to the rotation filters.

Also note that the other manufacturer outputs doesn't generate BoM and position. So you currently get the same for all, the ZIP contains the gerbers and drill outputs. In the JLCPCB case they offer a "free" PCBA service, so this is really popular and we generate the files, but again separately because they are uploaded separrately.

commit b432a647 (2023-12-18): [Fabrication Templates] Move files to the archive

- Is the most common case, so now we move (instead of copy)
- Can be configured using _KIBOT_COMPRESS_MOVE
- Left the JLCPCB BoM and Position outside, they must be uploaded
  separately

Closes #537