INTI-CMNB/KiBot #539: [HOW] SSE4.1 support in blender

oliv3r, 2023-12-28:
Not all CI runners will support SSE4.1, I can imagine virtual machines won't, but my personal runner (an old Phenom II X6) certainly doesn't, causing blender to fail with:

```
OpenImageDenoiser is not supported on this CPU: missing SSE 4.1 support disable sse4.1 in blender
```

Not sure if we actually need it, if we don't, we should just disable it, but otherwise, we may have to detect this during the running of the pipeline.

In my gitlab pipeline, I've added a simple sed line to do this based on sse4 detection.

```sh
cycles_openimagedenoise="$(grep -q 'sse4_1' '/proc/cpuinfo' && echo 'true' || echo 'false')"
export BLENDER_USER_CONFIG='/tmp/blender.cfg'
printf '[addons]\n\tcycles_openimagedenoise = %s\n' "${cycles_openimagedenoise}" > "${BLENDER_USER_CONFIG}"
```

Though probably nicer is to integrate this into the python side, with the same check, but then using `--disable-addons cycles_openimagedenoise` instead?

oliv3r, 2023-12-28:
actually, I couldn't even get that to work, so I had to resort to patching kibot instead. This can of course be done much nicer within kibot by checking cpu flags, but there's a bunch of libs, and not sure which one you'd use ...

```diff
diff --git a/kibot/blender_scripts/blender_export.py b/kibot/blender_scripts/blender_export.py
index d37cd105..df2c293d 100644
--- a/kibot/blender_scripts/blender_export.py
+++ b/kibot/blender_scripts/blender_export.py
@@ -344,6 +344,9 @@ def main():
         print("Please use -f and -o the same amount of times")
         sys.exit(2)
 
+    print("OpenImageDenoiser is not supported on this CPU: missing SSE 4.1 support, disabeling")
+    addon_utils.disable("cycles_openimagedenoise")
+
     print("Importing PCB3D file ...")
     PCB3D_file = os.path.abspath(args.PCB3D_file)
     # Start with fresh settings
```

With the following gitlab CI
```yaml
  before_script:
    - kicad-cli --version
    - |
      if ! grep -q 'sse4_1' '/proc/cpuinfo'; then (cd /usr/lib/python3/dist-packages/ && patch -p1 < "${CI_PROJECT_DIR}/sse4.patch"); fi
```

(still waiting on the slow pipeline to see if it actually worked though :p)

set-soft, 2023-12-29:
Hi @oliv3r !

If this issue is something you can fix changing your Blender configuration I'll suggest using this approach.

I don't understand what do you mean with "this works for 50% ... but to render it to a png", if you can get a *.blender* file, but you can't get a *.png* then this doesn't work at all. In order to create the PNG you need to render the image, and this is when the Cycles renderer is used, and hence we need the denoiser.

Trying to render on a machine that can't run the denoiser isn't a good idea. You'll need 10 times more passes to get a good image. This means that a slow machine that needs 20 seconds to do a pass needs 200 seconds for  10 passes, but to achieve the same quality without the denoiser will need 2000 seconds. 

I'll suggest trying: `bpy.context.scene.cycles.use_denoising = False` to see if this fixes your problem. If this works we can add options to **manually** disable it. This sentence is what Blender GUI says is doing when you disable the denoiser.

oliv3r, 2023-12-29:
> If this issue is something you can fix changing your Blender configuration I'll suggest using this approach.

As I'm using this in a CI, with the docker container, and on different hosts with different CPU configurations, this is obviously not an option. I have to do CPU detection, and apply a patch to kibot from the pipeline. The much better approach of course, is to do the CPU check within kibot itself of course. If you can suggest a library you prefer to do the test, I don't mind submitting my patch. But i already found 3 libraries that get the CPU flags/features.

> I don't understand what do you mean with "this works for 50% ... but to render it to a png", if you can get a _.blender_ file, but you can't get a _.png_ then this doesn't work at all. In order to [...]

Fair nuff, 50% being, in that I ask it to generate the `.blend` file and to render a `.png` file :)

So disabling the denoiser, generates the .blend file. but then blender just quits when generating the PNG file, appearantly due to the missing denoiser. Since SSE4.1 is missing on older AMD processors (intel had it for a much longer time afaik), and is certainly not present in virtual machines (e.g. CI runners in VM's), you basically are saying we cannot use blender to generate PNG files in CI setups reliably.

I'll try to do some more research if we can't generate PNG's from blender without the denoiser,  but I wonder how blender did this 5 years ago, where SSE4.1 wasn't so common? This sounds weird to me tbh, especially as it's only a plugin. No Plugin no png?!

set-soft, 2023-12-29:
> > If this issue is something you can fix changing your Blender configuration I'll suggest using this approach.
> 
> As I'm using this in a CI, with the docker container, and on different hosts with different CPU configurations, this is obviously not an option. 

Yes, it is. You can add layers to docker images, you can run commands to change the config, etc.

> I have to do CPU detection, and apply a patch to kibot from the pipeline. The much better approach of course, is to do the CPU check within kibot itself of course.

I don't agree, this is a very specific Blender detail for a very specific situation.

>  If you can suggest a library you prefer to do the test, I don't mind submitting my patch. But i already found 3 libraries that get the CPU flags/features.

The test is irrelevant the denoiser is optional in Blender, Debian packages doesn't have it, and the requirements for it are unknown and could change. 

> Fair nuff, 50% being, in that I ask it to generate the `.blend` file and to render a `.png` file :)
>
> So disabling the denoiser, generates the .blend file. but then blender just quits when generating [...]

Is this something you tried or just a guess?

> you basically are saying we cannot use blender to generate PNG files in CI setups reliably.

Nope, I say you must configure Blender if you want to run it in particular hardware. I.e. hardware not supported by Blender without tweaking.

> I'll try to do some more research if we can't generate PNG's from blender without the denoiser, but I wonder how blender did this 5 years ago, where SSE4.1 wasn't so common? This sounds weird to me tbh, especially as it's only a plugin. No Plugin no png?!

Don't be ridiculous, years ago, when the Intel AI denoiser didn't even exist, you needed a night for a good render. Now you can do it in real time using a good 3D accelerator. Nothing related to PNGs or KiBot. The point is that you are running Blender on an environment that needs tweaking to get it working. BTW: Cycles is not the only renderer in Blender, is the one needed by pcb2blender.

> Digging a little further, or rather, grepping in the blender 4.0 scripts/addons/cycles folder on my desktop, I notice that in the header files, there certainly are checks for SSE41. Could it be that the python wheels installed in the container, are binary, compiled against the host CPU architecture, not doing runtime detection? This would mean the container is setting the requirement of sort, rather then the host. If runtime detection is not possible (as this is something the blender add-in would have to do), we'd need to install the plugin post container creation, so that it the right CPU features are compiled in.

The installed Blender is the official distribution.

> Or not, searching further, I found this comment: https://projects.blender.org/blender/blender/issues/96651 which indicates this is certainly possible still using SSE2 only. They say 'just disable denoise' but I thought that was exactly what I was trying to do :( to be continued.

Did you even tried to run the command I suggested? (`bpy.context.scene.cycles.use_denoising = False`)

set-soft, 2024-01-02:
Try using the new render option, like this:

```yaml
      render_options:
        transparent_background: true
        samples: 10
        no_denoiser: true
```

But you'll need around 100 samples to compensate the lack of a denoiser.

closed 2024-01-02 by set-soft

oliv3r, 2024-01-04:
> Yes, it is. You can add layers to docker images, you can run commands to change the config, etc.
Sure I could, but I want to be able to 'just include' a gitlab yaml file, without heavy processing on the back. If we have to generate a new container for various configurations, then it shouldn't be a downstream solution obviously :p

> 
> > I have to do CPU detection, and apply a patch to kibot from the pipeline. The much better approach of course, is to do the CPU check within kibot itself of course.
> 
> I don't agree, this is a very specific Blender detail for a very specific situation.

It is specific in that blender doesn't function properly without SSE 4.1 (or as you found out without a flag :p). Ideally, blender should solve this imo of course (which it did as you found :p, my blender skills are poor).

> The test is irrelevant the denoiser is optional in Blender, Debian packages doesn't have it, and the requirements for it are unknown and could change.
>
>
> Is this something you tried or just a guess?

My runners do not have SSE 4.1, but simple VM's also emulate some generic old intel CPU. For example, this is what I get at one of my VPS's:
```
cat /proc/cpuinfo 
processor       : 0
vendor_id       : GenuineIntel
cpu family      : 6
model           : 13
model name      : QEMU Virtual CPU version 2.5+
...
flags           : fpu de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pse36 clflush mmx fxsr sse sse2 syscall nx lm rep_good nopl xtopology cpuid tsc_known_freq pni cx16 x2apic hypervisor lahf_lm pti
bugs            : cpu_meltdown spectre_v1 spectre_v2 spec_store_bypass l1tf mds swapgs itlb_multihit
```

> 
> > you basically are saying we cannot use blender to generate PNG files in CI setups reliably.
> 
> Nope, I say you must configure Blender if you want to run it in particular hardware. I.e. hardware not supported by Blender without tweaking.

That's what I tried :( But I know nothing of blender, and was relying on AI and docs to help me figure things out :) But sadly, the AI often lies. Or sends me down dumb rabbit holes. Like you can use a --disable flag to blender, but after an hour, it turns out it doesn't work in batch mode.

> Don't be ridiculous, years ago, when the Intel AI denoiser didn't even exist, you needed a night for a good render. Now you can do it in real time using a good 3D accelerator. Nothing related to [...]

Absolutly, which was my point! I simply know nothing of blender, how to tweak and configure any of this, which is my fault and problem of course.

> The installed Blender is the official distribution.
>
>
> Did you even tried to run the command I suggested? (`bpy.context.scene.cycles.use_denoising = False`)

shamefully, no; I hadn't come around it yet. I was trying to disable the plugin entirely, which works for blender file generation, but not for png generation :)

I'll test your new option/patches first now!! Thank you for that!!

oliv3r, 2024-01-04:
Hey @set-soft 

"It is what it is" :p, It's not like I can magically add SSE 4.1 :p 

I tried 100 samples at some point in a test run, and indeed, it takes forever :( I'll try to upgrade my hardware :D at hopefully sometime this year. When zen5 comes out, i'll get a nice zen4 :) 

anyway, I wanted to make this optional, depending on hardware functionality available; but I can't get the variable substitution to work :(

```yaml
      render_options:
        transparent_background: true
        no_denoiser: ${BLENDER_NO_DENOISER}
        samples: ${BLENDER_SAMPLES}
```

complains that `no_denoiser` must be a boolean, not a string. I'm not sure if variable substitution is not working (I couldn't get it to work for the diff one, where I wanted to get the branch name from a variable, so do a substation with sed), or if shell expantion always makes my 'true' into a variable.

I'm trying various combinations right now, but is this a new bug?

set-soft, 2024-01-04:
> but I can't get the variable substitution to work :(

https://kibot.readthedocs.io/en/latest/configuration/substitution.html#index-0

oliv3r, 2024-01-04:
@set-soft I have to report success! Indeed, denoising is definitely needed, the quality is pretty poor, but there's result, and the pipeline only ran for 3 minutes. So dedicating 20 minutes on the render on my poor hardware should be okay :p 

https://gitlab.com/olliver/lapod/-/jobs/5862607509

So thanks again for fixing this!

commit a48d0145 (2024-01-02): [Blender Export][Adde] Option to disable the denoiser

- Enables the use of old hardware and Debian packages
- You must be extremelly patient

Closes #539