INTI-CMNB/KiBot #540: Do not crash on missing files

oliv3r, 2023-12-29:
Running a (git) diff on files, where only one is available, the other file is usually/always treated as a '/dev/null' file. KiDiff should do the same, to avoid it crashing with the following error:

```
  File "/usr/lib/python3/dist-packages/kibot/out_diff.py", line 184, in cache_sch
    raise KiPlotConfigurationError('Missing file to compare: `{}`'.format(name))

ERROR:In section 'diffsch' (diff): Missing file to compare: `my_first.kicad_sch` (kibot.gs - gs.py:733)
```

E.g. this happens with the following kibot config:

```yaml
kibot:
  version: 1

global:
  environment:
    models_3d: '${KIPRJMOD}/packages3D'
  filters:
    - filter: 'KiCad 7 does not have environment.vars anymore. kibot#528'
      number: 9
      regex: 'KiCad config without environment.vars section'

outputs:
  - name: 'diffsch'
    comment: 'Generates a PDF with the differences between schematic.'
    type: 'diff'
    dir: 'outputs/diff'
    options: &diff_options
      add_link_id: false
      always_fail_if_missing: true
      cache_dir: '/tmp/pdfdiff'
      copy_instead_of_link: false
      diff_mode: 'red_green'
      force_checkout: false
      fuzz: 5
      new: 'HEAD'
      new_type: 'git'
      old: 'master'
      old_type: 'git'
      only_different: true
      only_first_sch_page: false
      threshold: 0
      use_file_id: false
      pcb: false
    layers: 'all'
  - name: 'diffpcb'
    comment: 'Generates a PDF with the differences between PCBs.'
    type: 'diff'
    dir: 'output/diff'
    options:
      <<: *diff_options
      pcb: true
    layers: 'all'
```

where master does not yet contain any file to diff against.

Though I just realize, this may be only a bug for kibot and I put it in the wrong spot, or maybe it's applicable to both ;)

oliv3r, 2023-12-29:
So I found the error of my ways from the code. I had enabled `always_fail_if_missing` which I shouldn't have set. I do still think that we shouldn't crash, but exit gracefully, but still, this being my fault, is not a bug of course.

set-soft, 2023-12-29:
> ```yaml
> always_fail_if_missing: true
> ```

You are explicitly asking to fail if the file is missing.

closed 2023-12-29 by set-soft

set-soft, 2023-12-29:
The code doesn't crash. If you get "ERROR:In section 'diffsch' (diff): Missing file to compare: `my_first.kicad_sch` (kibot.gs - gs.py:733)" the code is exiting gracefully.

oliv3r, 2024-01-04:
@set-soft Hmm, the traceback throws me off; Usually that indicates a crash. So if you are saying you are catching the failure, and the error indicates that things where handled gracefully, sure.

Since I'm running with -vvv to get logs to help us debug things, the following may not be relevant, but maybe we should only show the traceback when running with debugging?

More importantly, maybe add some more text before/after it?

```
Error: There has been a problem with X Y Z. Below find the traceback to help debug this issue, please attach this when reporting an issue
===== cut here 8<-------
[traceback]
------- cut here 8<====
Kibot exiting now.
```

for example :)

set-soft, 2024-01-04:
> Since I'm running with -vvv to get logs to help us debug things, the following may not be relevant, but maybe we should only show the traceback when running with debugging?

This is the current behavior