INTI-CMNB/KiBot #541: [FEATURE] Specify blender_export outputs in more detail

oliv3r, 2023-12-29:
Currently, the `blender_export` type combines quite a few outputs, that are not blender outputs, but can be generated and consumed by blender.

For example, we have pcb3d file output, which is an intermediate output, but also the various formats blender can generate, next to its native format.

As a user, I'd like to group the various outputs in directories, as such, I have `outputs/blender`, `outputs/pcb3d` or `outputs/ibom`. This is currently not possible as the `blender_export` type lacks `dir` options for `pcb3d` as well as the various other outputs (`x3d` for example).

set-soft, 2024-01-02:
> Currently, the `blender_export` type combines quite a few outputs, that are not blender outputs, but can be generated and consumed by blender.
> 
> For example, we have pcb3d file output, which is an intermediate output, but also the various formats blender can generate, next to its native format.

Note this can be generated by other output, and then consumed by this one.
 
> As a user, I'd like to group the various outputs in directories, as such, I have `outputs/blender`, `outputs/pcb3d` or `outputs/ibom`. This is currently not possible as the `blender_export` type lacks `dir` options for `pcb3d` as well as the various other outputs (`x3d` for example).

As mentioned above the `pcb3d` vs *blender generated* separation already exist.

set-soft, 2024-01-02:
Please test the above patch, it passed a very naive test.

closed 2024-01-02 by set-soft

oliv3r, 2024-01-04:
Excellent! I will give it a go for sure!

oliv3r, 2024-01-05:
@set-soft I've done a run with debugging enabled, https://gitlab.com/olliver/lapod/-/jobs/5867391514 and I still get the '@' symbol appended to my filenames. I'll try to run the pcb3d job as its own identity next, and see if I can 'inherit' the pcb3d file, without the blender job re-creating it.

set-soft, 2024-01-05:
Hi @oliv3r !

The above message is because Blender doesn't create multiple subdirs, so we must create them.

The d5fa776dbffa7cb5b7595ebdc27ce20c993a6c9d patch fixes it, I found it while fixing #546 

The tests I did asked for 1 missing dir and Blender created it, so I assumed it was able to create subdirs.

Note that the *dir* option will create names relative to the directory where all the other files of this output are handled. So you don't have to use *output/render3d*, use *../render3d* if the output itself uses *output/pcb3d* as dir (so you get *output/pcb3d/../render3d*. This is the same mechanism used by other outputs.

set-soft, 2024-01-05:
BTW: the docs says *Subdirectory for this output*, not *Directory for this output*

oliv3r, 2024-01-05:
> BTW: the docs says _Subdirectory for this output_, not _Directory for this output_

I just realized :) I would have maybe expected 'subdir' as argument, but it's fine. Since I couldn't specify the output for pcb3d (it doesn't have an dir component yet?) I 'hacked' the blender output dir to be `outputs/pcb3d`, and the other ones to be `outputs/blend` :p I'll use `../` instead to see if that works.

I briefly tried to use the `pcb2blender_tools` output, but it just generates a whole lot of files, not the pcb3d file that blender generates.

set-soft, 2024-01-05:
> I just realized :) I would have maybe expected 'subdir' as argument, but it's fine. 

Is the same concept using for the `dir` option of the output, relative to the global one.

> Since I couldn't specify the output for pcb3d (it doesn't have an dir component yet?) I 'hacked' the blender output dir to be `outputs/pcb3d`, and the other ones to be `outputs/blend` :p I'll use [...]

Look the internal template [PCB2Blender_2_7](https://kibot.readthedocs.io/en/latest/configuration/imports.html#importing-internal-templates)

commit d074294e (2024-01-02): [Blender Export][Added] Subdirectory for each output

Closes #541