INTI-CMNB/KiBot #542: [BUG] Render3D job keeps ogv, even on success

oliv3r, 2023-12-29:
Using the latest kibot 7 full docker container,

While it's nice in case of an error to keep the video file to identify errors, this wastes a lot of resources and space (especially on CI systems that keep files for each run).

```
> DEBUG:>>Interposer<<:IO:close:/tmp/pcbnew (deleted) (@18229.846 D 2.791) (pcbnew_do - interposer.py:159)
> DEBUG:exit_kicad_i got >>exit<< (pcbnew_do - interposer.py:598)
> DEBUG:Closing pipe with 1001 (pcbnew_do.kiauto.ui_automation - ui_automation.py:48)
> DEBUG:Terminating the session recorder (pcbnew_do.kiauto.ui_automation - ui_automation.py:172)
> DEBUG:Closing pipe with 1000 (pcbnew_do.kiauto.ui_automation - ui_automation.py:48)
> DEBUG:Restoring old KiCad common config (pcbnew_do.kiauto.file_util - file_util.py:188)
> DEBUG:Restoring old user hotkeys config (pcbnew_do.kiauto.file_util - file_util.py:188)
> DEBUG:Restoring old pcbnew config (pcbnew_do.kiauto.file_util - file_util.py:188)
> DEBUG:Restoring old colors config (pcbnew_do.kiauto.file_util - file_util.py:188)
> DEBUG:Restoring old 3D viewer config (pcbnew_do.kiauto.file_util - file_util.py:188)
> DEBUG:Checking if /workspace/breakout_hdmi/tmpauva7r7w.kicad_pro was modified (pcbnew_do.kiauto.file_util - file_util.py:338)
> DEBUG:Removing interposer dialog (/workspace/breakout_hdmi/outputs/render3d/pcbnew_interposer.log) (pcbnew_do - pcbnew_do:1737)
>  (kibot - kiplot.py:194)
DEBUG:Removing temporal files (kibot - out_base.py:949)
DEBUG:- File `/workspace/breakout_hdmi/tmpauva7r7w.kicad_pcb` (kibot - out_base.py:952)
DEBUG:- File `/workspace/breakout_hdmi/tmpauva7r7w.kicad_pro` (kibot - out_base.py:952)
DEBUG:- File `/workspace/breakout_hdmi/tmpauva7r7w.kicad_prl` (kibot - out_base.py:952)
DEBUG:- Executing: /usr/bin/convert /workspace/breakout_hdmi/outputs/render3d/breakout_hdmi-3D_top.png -trim +repage -trim +repage /workspace/breakout_hdmi/outputs/render3d/breakout_hdmi-3D_top.png (kibot - out_render_3d.py:29)
Found 1 unique warning/s (4 total, 1 filtered)
```

with the following yaml
```yaml
kibot:
  version: 1

global:
  environment:
    models_3d: '${KIPRJMOD}/packages3D'
  filters:
    - filter: 'KiCad 7 does not have environment.vars anymore. kibot#528'
      number: 9
      regex: 'KiCad config without environment.vars section'

preflight:
  run_drc: false
  run_erc: false

outputs:
  - name: 'render_3d_bottom'
    comment: "Exports the image generated by KiCad's 3D viewer."
    type: 'render_3d'
    dir: 'outputs/render3d'
    options: &render_options
      auto_crop: true
      background1: '#66667f'
      background2: '#cccce5'
      board: '#332b16'
      clip_silk_on_via_annulus: true
      copper: '#8b898c'
      dnf_filter: '_none'
      download: false
      height: 1024
      highlight: []
      move_x: 0
      move_y: -3
      move_z: 0
      no_smd: false
      no_tht: false
      no_virtual: false
      orthographic: false
      pre_transform: '_none'
      ray_tracing: true
      realistic: true
      rotate_x: 0
      rotate_y: 0
      rotate_z: 0
      show_adhesive: false
      show_board_body: true
      show_comments: false
      show_components: all
      show_eco: false
      show_silkscreen: true
      show_soldermask: true
      show_solderpaste: true
      show_zones: true
      silk: '#d5dce4'
      solder_mask: '#208b4780'
      solder_paste: '#808080'
      subtract_mask_from_silk: true
      transparent_background: false
      transparent_background_color: '#00ff00'
      transparent_background_fuzz: 15
      variant: ''
      view: 'bottom'
      wait_render: -600
      width: 1440
      zoom: 6
  - name: 'render_3d_front'
    comment: "Exports the image generated by KiCad's 3D viewer."
    type: 'render_3d'
    dir: 'outputs/render3d'
    options:
      <<: *render_options
      move_x: 0
      move_y: -2
      move_z: 0
      rotate_x: -5
      rotate_y: 0
      rotate_z: -2
      view: 'front'
      zoom: 4
  - name: 'render_3d_top'
    comment: "Exports the image generated by KiCad's 3D viewer."
    type: 'render_3d'
    dir: 'outputs/render3d'
    options:
      <<: *render_options
      move_x: 0
      move_y: -3
      move_z: 0
      rotate_x: 0
      rotate_y: 0
      rotate_z: 0
      view: 'top'
```

set-soft, 2024-01-02:
If you ask for debug then you get the video. If you want debug logs, but not the video try using `-L` or just remove the video manually.

closed 2024-01-02 by set-soft

oliv3r, 2024-01-04:
@set-soft I apologize! I think I fell into this trap before :p I turned on debugging because we wanted more logs to debug, but I failed to make the link with the video's (again).