INTI-CMNB/KiBot #543: [BUG] render3d can sometimes get confused on what filename to use, causes timeouts.

oliv3r, 2023-12-29:
When running the following yaml file in the latest kibot 7 full container in a CI, kicad gets confused and fails to save the file (see video in artifacts). Because of that (filename is a # only) kibot keeps waiting for the timeout, and then kills stuff.

Instead, kibot could check the intended filename, and see if it is valid at all.

I've noticed this issue in more flows, so I wonder, would it not be wiser, to save the file as a temporary, mktmpdir in python, check if all went well, store file with a random name there, and then when finished mv the tmp file to the intended destination? This would remove quite a few of these errors, in that the GUI kicad will have a super reliable way to generate the needed file ...

```yaml
kibot:
  version: 1

global:
  environment:
    models_3d: '${KIPRJMOD}/packages3D'
  filters:
    - filter: 'KiCad 7 does not have environment.vars anymore. kibot#528'
      number: 9
      regex: 'KiCad config without environment.vars section'

preflight:
  run_drc: false
  run_erc: false

outputs:
  - name: 'render_3d_bottom'
    comment: "Exports the image generated by KiCad's 3D viewer."
    type: 'render_3d'
    dir: 'outputs/render3d'
    options: &render_options
      auto_crop: true
      background1: '#66667f'
      background2: '#cccce5'
      board: '#332b16'
      clip_silk_on_via_annulus: true
      copper: '#8b898c'
      dnf_filter: '_none'
      download: false
      height: 1024
      highlight: []
      move_x: 0
      move_y: -3
      move_z: 0
      no_smd: false
      no_tht: false
      no_virtual: false
      orthographic: false
      pre_transform: '_none'
      ray_tracing: true
      realistic: true
      rotate_x: 0
      rotate_y: 0
      rotate_z: 0
      show_adhesive: false
      show_board_body: true
      show_comments: false
      show_components: all
      show_eco: false
      show_silkscreen: true
      show_soldermask: true
      show_solderpaste: true
      show_zones: true
      silk: '#d5dce4'
      solder_mask: '#208b4780'
      solder_paste: '#808080'
      subtract_mask_from_silk: true
      transparent_background: false
      transparent_background_color: '#00ff00'
      transparent_background_fuzz: 15
      variant: ''
      view: 'bottom'
      wait_render: -600
      width: 1440
      zoom: 6
  - name: 'render_3d_front'
    comment: "Exports the image generated by KiCad's 3D viewer."
    type: 'render_3d'
    dir: 'outputs/render3d'
    options:
      <<: *render_options
      move_x: 0
      move_y: -2
      move_z: 0
      rotate_x: -5
      rotate_y: 0
      rotate_z: -2
      view: 'front'
      zoom: 4
  - name: 'render_3d_top'
    comment: "Exports the image generated by KiCad's 3D viewer."
    type: 'render_3d'
    dir: 'outputs/render3d'
    options:
      <<: *render_options
      move_x: 0
      move_y: -3
      move_z: 0
      rotate_x: 0
      rotate_y: 0
      rotate_z: 0
      view: 'top'
```

See https://gitlab.com/olliver/lapod/-/jobs/5836088538

oliv3r, 2023-12-31:
Here'st he interposer log from the pipeline, probably best to attach it to the issue so it don't get lost.

[pcbnew_interposer.log](https://github.com/INTI-CMNB/KiBot/files/13801288/pcbnew_interposer.log)

set-soft, 2024-01-02:
This is the same as #544

closed 2024-01-02 by set-soft