INTI-CMNB/KiBot #544: [BUG] VRML job hangs even after successful file creation

oliv3r, 2023-12-29:
Using the latest kibot 7 full container, generating a VRML, never closes kicad.

Looking at the video, we see that the file is generated and saved properly, but kicad does not exit, causing a timeout error to fail everything.

The used yaml file:

```yaml
kibot:
  version: 1

global:
  environment:
    models_3d: '${KIPRJMOD}/packages3D'
  filters:
    - filter: 'KiCad 7 does not have environment.vars anymore. kibot#528'
      number: 9
      regex: 'KiCad config without environment.vars section'

preflight:
  run_drc: false
  run_erc: false

outputs:
  - name: 'vrml'
    comment: 'PCB in VRML format'
    type: 'vrml'
    dir: 'outputs/vrml'
    options:
      download: false
      dir_models: ''
```

See https://gitlab.com/olliver/lapod/-/jobs/5836088539 and https://gitlab.com/olliver/lapod/-/pipelines/1122395553

set-soft, 2024-01-02:
Hi @oliv3r !

This is a very bizarre situation, looking at the interposer logs I see:

```
IO:open:/builds/olliver/lapod/breakout_hdmi/outputs/vrml/breakout_hdmi-vrml.wrl
IO:fopen64:w
IO:close:/builds/olliver/lapod/outputs/breakout_hdmi/vrml/breakout_hdmi-vrml.wrl
```

Take a closer look, you are creating a file named `/builds/olliver/lapod/breakout_hdmi/outputs/vrml/breakout_hdmi-vrml.wrl`, this is the name asked in the call:

```
DEBUG:Executing: /usr/bin/pcbnew_do -r -v export_vrml --output_name breakout_hdmi-vrml.wrl -U millimeters /builds/olliver/lapod/breakout_hdmi/tmpjbaos2qa.kicad_pcb /builds/olliver/lapod/breakout_hdmi/outputs/vrml (kibot - kiplot.py:180)
```

But the close is for another file: `/builds/olliver/lapod/outputs/breakout_hdmi/vrml/breakout_hdmi-vrml.wrl`.

Looking at your job this is the consequence of using a symlink as target directory. Stop using it, just move the files if you want. I'll see if I can find a solution to support it.

set-soft, 2024-01-02:
Ok, the above patch solves this issue.

closed 2024-01-02 by set-soft

oliv3r, 2024-01-04:
> Looking at your job this is the consequence of using a symlink as target directory. Stop using it, just move the files if you want. I'll see if I can find a solution to support it.

There's a simple and good reason for using symlinks :)

I'm doing a multi-pcb project, and sadly, KiCad does not support this (multiple schematics, multiple PCB's in a single kicad_pro). So I have to do multiple directories, each with one PCB. I want to run  the pipeline for each board, which gives me basically two options. Generate a pipeline for each board at runtime (certainly possible, I have done this in the past, but its also quite complex, for something I'd like to keep simple), or itterate each pipeline step for each board. I've chosen the later, which means I have to cd into the board, and generate outputs for each. Since I want to have all outputs together at the root, I'd have to as you say copy or move them, but I don't know the file names. I could move all content for sure, but it just adds unnecessary fluff. So by putting things in a symlink, it simplifies things considerable!!

Also, we shouldn't artificially limit the creativity of users if we an avoid it :D your solution is obviously spot on, and should always be used in this way!!

P.S. I'm not sure on how python's realpath works, but I know shell implementations tend to differ, some always work, some only if the file is an actual symlink, so I have something `if [ -L "${file:-}" ]; then file="$(realpath -f "${file}")"; fi` or similar. Just something to keep in the back of your head if this causes problems in the future.

Thanks!!

commit e4614e76 (2024-01-02): Now we use the realpath instead of the user provided name

Helps to avoid unpleasant surprises like in the #544 case

Closes #544