INTI-CMNB/KiBot #546: [BUG] blender_export makes rotate_x mandatory

oliv3r, 2024-01-04:
When running the latest kibot 7 full DEV container, the render_x component is now mandatory, as kibot crashes otherwise. Instead, if kibot encounters the Nonetype, it should just use the '0' value for that component instead.

Defining the pov values 'fixes' this issue.

```yaml
     1	kibot:
     2	  version: 1
     3	
     4	global:
     5	  environment:
     6	    models_3d: '${KIPRJMOD}/packages3D'
     7	  filters:
     8	    - filter: 'KiCad 7 does not have environment.vars anymore. kibot#528'
     9	      number: 9
    10	      regex: 'KiCad config without environment.vars section'
    11	
    12	preflight:
    13	  run_drc: false
    14	  run_erc: false
    15	
    16	outputs:
    17	  - name: '3D_top_view'
    18	    comment: '3D render from the top.'
    19	    type: 'blender_export'
    20	    dir: 'outputs/pcb3d'
    21	    options: &blender_options
    22	      render_options:
    23	        transparent_background: true
    24	        no_denoiser: @BLENDER_NO_DENOISER@
    25	        samples: @BLENDER_SAMPLES@
    26	      outputs:
    27	        - type: 'blender'
    28	          dir: 'outputs/blender'
    29	        - type: 'render'
    30	          dir: 'outputs/render3d'
    31	  - name: '3D_front_30deg_view'
    32	    comment: '3D render from front 30 deg.'
    33	    type: 'blender_export'
    34	    dir: 'outputs/blender'
    35	    options:
    36	      <<: *blender_options
    37	      point_of_view:
    38	        rotate_x: 30
    39	        rotate_z: -20
    40	        view: 'front'
    41	  - name: '3D_bottom_view'
    42	    comment: '3D render from the bottom.'
    43	    type: 'blender_export'
    44	    dir: 'outputs/blender'
    45	    options:
    46	      <<: *blender_options
    47	      point_of_view:
    48	        rotate_y: 180
    49	        view: 'bottom'
```

Sadly I ran without debug, so I don't have as much info as I would like.

```
Using SCH file: breakout_hdmi.kicad_sch
- '3D render from the top.' (3D_top_view) [blender_export]
ERROR:Error from Blender run:
Traceback (most recent call last):
  File "/usr/local/lib/python3.11/dist-packages/kibot/blender_scripts/blender_export.py", line 438, in <module>
    main()
  File "/usr/local/lib/python3.11/dist-packages/kibot/blender_scripts/blender_export.py", line 416, in main
    c_views = apply_start_scene(args.scene)
  File "/usr/local/lib/python3.11/dist-packages/kibot/blender_scripts/blender_export.py", line 214, in apply_start_scene
    cur_rot = (cur_rot[0] + pov.get('rotate_x', 0),
TypeError: 'NoneType' object is not subscriptable
```

See https://gitlab.com/olliver/lapod/-/jobs/5862320699 for details.

closed 2024-01-05 by set-soft

set-soft, 2024-01-05:
You can see docker image releases here:

https://github.com/INTI-CMNB/kicad_auto_test/pkgs/container/kicad7_auto_full/versions

You pulled sha256:abf5c798a270be66319534955c1a111a8bddd950b2a9b8ea7c6ea3f98eb3a78d

This is dev_1.6.4-5bf6ae7_k7.0.9_d12.1_b3.5.1, looking at the *dev* commits (https://github.com/INTI-CMNB/KiBot/commits/dev/) this is from yesterday: https://github.com/INTI-CMNB/KiBot/commit/5bf6ae7c5da2d1fc3e2789dc89e7245931c6f153

So it doesn't include the relevant patch.

oliv3r, 2024-01-07:
Yeah, I switched to the dev version now; and set my pull to 'always'; it's going great :D

commit ab17559a (2024-01-05): [Blender export][Fixed] Problems when no point of view was defined

- The default view wasn't translated.
- Now translated
- Now the render script defaults to 0 rotation if the view is unknown
Fixes #546