INTI-CMNB/KiBot #548: [BUG] Docs do not render type properly.

oliv3r, 2024-01-05:
When looking at the latest readthedocs documentation, the 'types' field is left as an empty string:

Fill as much as you can, remove the rest

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Command line options passed to KiBot, configuration files (.kibot.yaml), PCB/Schematic example to reproduce it (Ex: a link to a repo where the files are)

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment (please complete the following information):**
Where are you running KiBot:
- In a CI/CD workflow? (Ex: `GitHub`/`GitLab`)
- In a GitHub action? Which tag are you using? (Ex `v1`, `v1_k6`, `v2`, `v2_d`, `v2_dk6`, `v2_k6`)
- Using a docker image? (Ex: `kicad_auto:ki6`, `kicad_auto:latest`, `kicad_auto:dev`, `kicad_auto:dev_k6`, a custom one? can you provide the `Dockerfile`)
- Using a desktop system? Which OS, distro and version?

**Additional context**
Add any other context about the problem here.

set-soft, 2024-01-05:
Hi @oliv3r !

The docs are correct, the default value for *type* is an empty string. And yes, the type is at the top. So there is no bug here. In fact the old docs just skipped the *type* because it was at the top and was quite obvious what it means.

That said: Showing how to fill it doesn't do any harm, so now the docs shows it in a way that resembles the YAML use

closed 2024-01-05 by set-soft

oliv3r, 2024-01-07:
Superb! @set-soft you are amazing :D