INTI-CMNB/KiBot #550: [BUG] Empty blender export pictures

laenzlinger, 2024-01-09:
Fill as much as you can, remove the rest

**Describe the bug**
First of all: Thank you very much for KiBot. Its such a great tool and a huge time saver! I am using the same configuration on several project and the blender output works on other projects. But on this very project, the output is just empty (only the gray background is visible) Is my pcb just too small? Its tiny 24x24mm

**To Reproduce**
* repo: https://github.com/pedalboard/pedalboard-led-ring
* github action: https://github.com/pedalboard/pedalboard-led-ring/blob/main/.github/workflows/ci.yml
* config: https://github.com/pedalboard/pedalboard-led-ring/blob/main/pedalboard-led-ring.kibot.yaml (more or less the default config)

**Screenshots**
see https://pedalboard.github.io/pedalboard-led-ring-site/latest/Browse/pedalboard-led-ring-navigate_PCB_3D.html

**Environment (please complete the following information):**
Where are you running KiBot:
- In a CI/CD workflow? Github Action
- Using a docker image? ghcr.io/inti-cmnb/kicad7_auto_full:latest

set-soft, 2024-01-09:
Hi @laenzlinger !

The size might be the problem, play with the `auto_camera_z_axis_factor` (enlarge it).

How tall is the board? Note that this is something KiBot can't currently compute

laenzlinger, 2024-01-09:
Thank you @set-soft  for your tip. 

I could not find  'auto_camera_z_axis_factor' in the config documentation. Where should I put this value in the config structure?

The board is small: 24x24mm

set-soft, 2024-01-10:
Hi @laenzlinger !

> I could not find 'auto_camera_z_axis_factor' in the config documentation. Where should I put this value in the config structure?

Ok, I see you are using *latest*, change the docker image to *dev*. I think the problem will go away without even changing the above parameter. You can find the docs for the *dev* version (1.6.4) [here](https://kibot.readthedocs.io/en/latest/)

> The board is small: 24x24mm

I saw this value, I was asking for the height of the tallest component. When trying to render a small board the camera is positioned close to the board and tall components could generate issues. But 1.6.4 changed various details to be smarter when dealing with small boards.

laenzlinger, 2024-01-10:
I have switched to dev, and things start to work.

Yes there is a hight component on the board. 

Thanks for your help!

closed 2024-01-11 by set-soft