INTI-CMNB/KiBot #551: Support different Visual Diff Colors

jtmullen, 2024-01-09:
**Is your feature request related to a problem? Please describe.**
We've started using the visual diff feature, unfortunately I am color blind and have a really hard time with the colors you've chosen. Specifically the red is indiscernible from the black as a regular zoom level. 

**Describe the solution you'd like**
Allowing custom colors fixes this for everyone, but could also just be changed to another color scheme (changing either the red or the black to a lighter shade of blue would be better than red for people with red-green color blindness. Of course there are other, more rare kinds of color blindness, that this would be worse for.)

set-soft, 2024-01-10:
Hi @jtmullen !

Is this using the "Diff" output? I'm asking because the *dev* branch introduced "KiRi", which is more "visual".

The problem here is not that simple, the colors are obtained mixing the RGB channels, so you can't control the resulting colors in a easy way. I'll investigate what can be done.

set-soft, 2024-01-10:
Hi @jtmullen !

The above patch adds a mode where you can control the colors. I tested it with a very simple config, like this:

```yaml
kibot:
  version: 1

outputs:
  - name: 'diff_pcb'
    comment: "PCB difference with reference file (KiCad 7)"
    type: diff
    layers: ['F.Cu', 'In1.Cu']
    options:
      old: tests/board_samples/kicad_7/light_control.kicad_pcb
      old_type: file
      cache_dir: .cache
      diff_mode: 2color
      color_added: "#3030FF"
      color_removed: "#FF3030"
      # zones: unfill
```

And got a red/blue diff:

[image]

Will be available in the *dev* branch.

closed 2024-01-10 by set-soft

jtmullen, 2024-01-10:
Hi @set-soft, yes this was with the basic `diff` output. Your solution looks good, thanks!

commit 2ccb54b0 (2024-01-10): [Diff][Added] New mode where we can control the added/removed colors

Also fixed the "zones" options help message, altered while creating
a base class
Closes #551