INTI-CMNB/KiBot #552: [BUG] Text search in PCB output is broken

martinlofgren, 2024-01-10:
**Describe the bug**
Text search in `pcb_print` used to work before, but doesn't anymore.

**To Reproduce**
Using the KiBot config file with the following output:
```yaml
  - name: 'Assembly_drawing_top'
    comment: "Assembly drawing top"
    type: pcb_print
    dir: 'output/Assembly_drawing'
    options:
      force_edge_cuts: true
      format: 'PDF'
      output: Assembly_top.pdf
      plot_sheet_reference: true
      pages:
        - layers: [ F.Paste, F.Silkscreen, F.Fab ]
          sheet: 'Assembly drawing top'
          monochrome: true
```

Running in Docker with the following command:

```sh
$ docker run -it \
    -v $(pwd):/data \
    -w /data \
    ghcr.io/inti-cmnb/kicad7_auto:latest \
    kibot -v -b project.kicad_pcb -c KiBotConfig/config.yaml Assembly_drawing_top
```

**Expected behavior**
Should render a pcb drawing with searchable text.

**Environment (please complete the following information):**
Have been running this in the context of a Gitlab CI job, but also in Docker in an Arch Linux system.

Previous, text search was functioning. I did try to run the same project using different versions, seems like the bug was introduced between [1.6.3-1_k7.0.5.1_d11.6](https://github.com/inti-cmnb/kicad_auto/pkgs/container/kicad7_auto/118459788?tag=1.6.3-1_k7.0.5.1_d11.6) and [1.6.3-1_k7.0.7_d12.1](https://github.com/inti-cmnb/kicad_auto/pkgs/container/kicad7_auto/135684285?tag=1.6.3-1_k7.0.7_d12.1).

martinlofgren, 2024-01-10:
Hi @set-soft. Sorry missed to write that, but I did test `dev` as well, but with same result (text search not working).

set-soft, 2024-01-10:
The problem seems to be in *rsvg-convert* tool. In the past it just dropped transparent text, so KiBot changed the text to be 99% transparent, but looks like the current version drops any text that isn't 0% transparent :-(

set-soft, 2024-01-10:
Ok, the above patch should fix it. Please report any side effect.

I'm changing the text to use the background color, this isn't perfect, in particular if you use a background image, but seems to be much better.

martinlofgren, 2024-01-11:
> The problem seems to be in _rsvg-convert_ tool. In the past it just dropped transparent text, so KiBot changed the text to be 99% transparent, but looks like the current version drops any text that isn't 0% transparent :-(

That's too bad. I took a brief look at the [librsvg repo](https://gitlab.gnome.org/GNOME/librsvg) but couldn't see anything about this in the changelog, nor the issue tracker. 

> The above patch tries to fix it, but fails Will investigate more

:pray: I can take a look as well if you want, please point me in the right direction if there is anything I can do.

set-soft, 2024-01-11:
> 🙏 I can take a look as well if you want, please point me in the right direction if there is anything I can do.

We compose the image using SVGs. If you know how to play with SVGs here is the problem: 

- KiCad uses its own fonts, so text is generated using graphics
- In order to make the real text available KiCad adds *text* tags with the real text using `opacity="0"`. So this text is invisible
- When converting the SVG to PDF using `rsvg-convert` the text is lost
- In previous versions (Debian 11) setting the text opacity to "0.01" preserved the text, now is lost
- If we set the opacity to 1 then the text is visible and interferes with the text drawn by KiCad

I think the solution could be something like moving all this text to a separated "layer", something behind the "paper". As PDFs can't be transparent we can draw a white (or any other color) rectangle before drawing the PCB, but after adding the text. So the text will be invisible, but be there. I must try this approach.

martinlofgren, 2024-01-11:
Works like a charm, for both text without transforms and rotated text.

commit 58641e2b (2024-01-10): [PDF_Print][Fixed] Text not searchable

- When using newer rsvg-convert tools i.e. 2.54.7
- Previous versions discarded 100% transparent text
- New versions discards text with any level of transparency

Fixes #552
commit 7bfd46e9 (2024-01-11): [PCB_Print] Changed searchable text strategy

Moved the text to the top of the SVG

Fixes #552