INTI-CMNB/KiBot #554: [FEATURE] Allow missing layers (pdf, fab)

oliv3r, 2024-01-14:
Currently, when I have defined multiple layers to make things easier to handle between different projects, not all layers may be existing in the pcb file.

Currently, we fail with

```
  File "/usr/local/lib/python3.11/dist-packages/kibot/out_pcb_print.py", line 230, in config
    self.layers = LayerOptions.solve(self.layers)
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/kibot/layer.py", line 224, in solve
    raise PlotError("Inner layer `{}` is not valid for this board".format(layer))
kibot.error.PlotError: Inner layer `In1.Cu (1 'Inner layer 1' In1_Cu)` is not valid for this board
```

However, it would be nice if kibot would just 'skip' that layer (with a unique ignoreable warning if needed).

I can solve this on the outside of course, by scanning the pcb file (using grep for example), and then generate the needed yaml entries. However that's far uglier and more complex, where instead we can just skip missing layers.

set-soft, 2024-01-15:
Hmmm ... I think you are looking for the following:

```yaml
# Example KiBot config file
kibot:
  version: 1

outputs:
  - name: 'print_front'
    comment: "Print Top and Bottom layers"
    type: pcb_print
    dir: Layers
    options:
      format: 'PDF'
      keep_temporal_files: true
      pages:
        - sheet: '%ln'
          repeat_for_layer: 'F.Cu'
          repeat_layers: copper
          layers:
            - layer: Edge.Cuts
            - layer: F.Cu
            - layer: F.Mask
              color: '#14332440'
            - layer: F.Paste
            - layer: F.SilkS
            - layer: F.Fab
              plot_footprint_refs: false
              plot_footprint_values: false
            - layer: Dwgs.User
```

This will generate 1 page for each copper layer, all of them using the same "template"

set-soft, 2024-01-15:
While looking at this example I also added this:

```yaml
# Example KiBot config file
kibot:
  version: 1

outputs:
  - name: 'print_front'
    comment: "Print Top and Bottom layers"
    type: pcb_print
    dir: Layers
    options:
      format: 'PDF'
      keep_temporal_files: true
      pages:
        - scaling: 2.0
          sheet: '%ln'
          repeat_for_layer: 'F.Cu'
          repeat_layers: outers
          layers:
            - layer: Edge.Cuts
            - layer: F.Cu
            - layer: F.Mask
              color: '#14332440'
            - layer: F.Paste
            - layer: F.SilkS
            - layer: F.Fab
              plot_footprint_refs: false
              plot_footprint_values: false
            - layer: Dwgs.User
        - scaling: 2.0
          sheet: '%ln'
          repeat_for_layer: 'F.Cu'
          repeat_layers: inners
          layers:
            - layer: Edge.Cuts
            - layer: F.Cu
            - layer: F.SilkS
            - layer: Dwgs.User
```

So you can separate inner/outer layers

oliv3r, 2024-01-15:
That's pretty cool, I'll look into that. Though this makes it hard to use individual colors per layer. ANd I want to use one template for all boards, so the content of the pcb shouldn't matter whether it has no inner layers. Granted, using the same color fixes that ..

The bigger problem was with the fab bit, as that uses a 'generated' defnition bit. I'll try your suggestion, and see if that helps. Ideally, we have an option t get the color from the PCB file, that would also help a lot :)

oliv3r, 2024-01-15:
After looking more closely, I noticed that at least JLCPCB has a sensible default for `_KIBOT_GERBER_LAYERS`. I'm not sure why the example brings this one up to define, but if all board house imports have proper defines, that work for 99% of the cases, then we'd be good! Simply removing this variable should make things work!

set-soft, 2024-01-16:
> we have an option t get the color from the PCB file, that would also help a lot :)

You can apply color themes. And note that the color schemes can be found in your repo (see `resources_dir` global option), no need to manually install the colors in the docker image

set-soft, 2024-01-16:
> After looking more closely, I noticed that at least JLCPCB has a sensible default for `_KIBOT_GERBER_LAYERS`. I'm not sure why the example brings this one up to define, but if all board house imports have proper defines, that work for 99% of the cases, then we'd be good! Simply removing this variable should make things work!
> 
> @set-soft Can you confirm my reasoning?

When you mention *the example* I guess you mean the *quick-start* generated config, right?

The generated file uses the detected layers, for the JLCPCB example we could just use *copper* to replace all the copper layers. But the detection goes farther, i.e. knows if you have a bottom silk screen.
 
> For now, it seems only P-Ban isn't setup correctly?
> 
> ```
> - 'Gerbers compatible with P-Ban' (_P-Ban_gerbers) [gerber]
> WARNING:(W049) Layer "Inner layer 1" (In1_Cu) isn't used (kibot - out_any_layer.py:186)
> ```

The problem with P-Ban is that I'm using the exact names generated by the *KiCad Gerber Zipper* tool. To get these names you can't use *copper*, the difference is subtle, but I can't be sure if this is really important.

set-soft, 2024-01-16:
Ok, I checked and the difference in names isn't there for KiCad 7, so I think the names should be ok using *copper*

closed 2024-01-16 by set-soft

oliv3r, 2024-01-16:
> > we have an option t get the color from the PCB file, that would also help a lot :)
> 
> You can apply color themes. And note that the color schemes can be found in your repo (see `resources_dir` global option), no need to manually install the colors in the docker image

Actually, I don't want to use any custom colors. By default, I want to just use whatever 'you' have defined. However, it would be nice, that those colors ideally are derived from the pcb file. In kicad 7, we can define the various colors for various layers. So if that is used as input (even if there's a `green` -> `nice color value` mapping somewhere decided by 'you', using the `resources_dir`, that's absolutely perfect.

set-soft, 2024-01-17:
Hi @oliv3r !

For the GUI (and plot) the colors aren't stored in the PCB (or project), you must create a color theme.

For the 3D render you can define physical colors (i.e. solder mask), but this is another history.

oliv3r, 2024-01-19:
> For the GUI (and plot) the colors aren't stored in the PCB (or project), you must create a color theme.

Are you sure? KiCad 7 can set the color of the mask in the stackup ...

```kicad_pcb
(stackup
  (layer "F.Mask" (type "Top Solder Mask") (color "#008400A8") (thickness 0.01))
)
```

I fully agree, that if the color is not set, the color from the kicad default colortheme is used, but that's also something we can extract?

Sadly, the 'layers' are a PCB color theme option, which are stored in the user-profile, not in the pcb file. imo that's a kicad bug? self-contained :p

set-soft, 2024-01-19:
> > For the GUI (and plot) the colors aren't stored in the PCB (or project), you must create a color theme.
> 
> Are you sure? KiCad 7 can set the color of the mask in the stackup ...
> 
> ```pcb-sexp
> (stackup
>   (layer "F.Mask" (type "Top Solder Mask") (color "#008400A8") (thickness 0.01))
> )
> ```

This is the *For the 3D render you can define physical colors (i.e. solder mask), but this is another history.*

This isn't the color for the GUI, which is the one used by KiCad for plot/print.

> I fully agree, that if the color is not set, the color from the kicad default colortheme is used, but that's also something we can extract?
> 
> Sadly, the 'layers' are a PCB color theme option, which are stored in the user-profile, not in the pcb file. imo that's a kicad bug? self-contained :p

KiCad format has tons of limitations, the worst is about the 3D models, but again: you can put a color theme in your repo and KiBot will be able to use it.

The stackup colors are used for 3D render.

oliv3r, 2024-01-19:
I agree that most limitations are probably from kicad then. though the pdf layer generation works well, the built-in theme 'is good enough'.

commit dedf2970 (2024-01-16): [Templates][Modified] P-Ban gerbers to use *copper*

- There is no difference for modern KiCad (just for KiCad 5)
- Avoids warnings about extra inner layers
- Enables the use of more than  8 layers

Closes #554