INTI-CMNB/KiBot #555: [BUG] Cannot checkout current branch

oliv3r, 2024-01-14:
While trying to debug by reproducing `ERROR:kicad-diff.py:Nothing to compare!` here https://gitlab.com/olliver/lapod/-/jobs/5926844760, I noticed that things may not work as expected when trying to compare using the branch that is the same as the currently checked out branch. This because `git worktree` refuses to checkout the same branch twice. Normally this is not a good idea, as one might be committing to the same branch from two different locations, but kibot only does read-only operations, so this is a non-issue, and kibot also removes the created branch afterwards. Because of that, we probably should use the `--force` parameter to `git worktree add`.

```
ERROR:Running ['/usr/bin/git', 'worktree', 'add', '/tmp/tmpfd90sbcp', 'dev'] returned 128 (kibot.gs - gs.py:791)
I have no name!@3c5acab7ecdb:/workspace/lapod/hello$ git worktree list
/workspace/lapod  815ec22 [dev]
I have no name!@3c5acab7ecdb:/workspace/lapod/hello$ git worktree add /tmp/asdf dev
Preparing worktree (checking out 'dev')
fatal: 'dev' is already checked out at '/workspace/lapod'
I have no name!@3c5acab7ecdb:/workspace/lapod/hello$ git worktree add -f /tmp/asdf dev
Preparing worktree (checking out 'dev')
HEAD is now at 815ec22 test
I have no name!@3c5acab7ecdb:/workspace/lapod/hello$ git worktree list
/workspace/lapod  815ec22 [dev]
/tmp/asdf         815ec22 [dev]
```

set-soft, 2024-01-15:
Hi @oliv3r !

During normal use you shouldn't need to checkout the same branch twice. But is true that you could get a collision if you don't pay attention.

The above patch adds a *--force*

closed 2024-01-15 by set-soft

oliv3r, 2024-01-15:
Thanks @set-soft!! 

I agree on the 'normally', but the patch is still wise I recon, especially since we are in a contained environment, doing our own cleanup.

Sadly, this didn't fix my bug :( So I'll investigate as to what causes 'Nothing to compare', and see if it is indeed a caching issue with gitlab, and just requires me to forcefully do a 'git fetch'.