INTI-CMNB/KiBot #559: [FEATURE] Support including yaml files in yaml files

oliv3r, 2024-01-16:
Officially, YAML does not allow inclusion of yaml files. Unofficially, there are yaml parsers that support the this by `(!)include`, or by concatenating files internally.

To follow best DRY principles, it would be very useful to be able to include yaml files, e.g. to have a `global.kibot.yaml(.inc)`, which defines a global section, that is always the same.

`import` seems to do things differently, and not the same as include.

set-soft, 2024-01-16:
Is true that import does things in a different way. But the question is: what's wrong with the import mechanism? You can import globals! And you have more control over imports, see the `imported_global_has_less_priority` option.

oliv3r, 2024-01-16:
Well the only thing I want to do, is have `global` and `preflight` in a re-usable file that i can include.

After inclusion, some files will need to enable certain features. e.g. in `erc.kibot.yaml`, I'll include my `common.kibot.yaml`, and then only has `preflight:run_drc: true` (the default in common would be false).

set-soft, 2024-01-17:
Ok, but this is something you can do using the *import* section.

If the import mechanism can do all you need then I think this FR should be closed.

oliv3r, 2024-01-17:
Let me try again; last time I tried, a few weeks ago, I was much newer to kibot :) If it works, I'll close this FR.

oliv3r, 2024-01-18:
> `imported_global_has_less_priority`

Hmm, searching in the dev-docs on readthedocs, it doesn't find that flag. (Its in the main docs however)

Regardless, it was working great actually, until I wanted to do the same trick you where doing with the fab templates.

```yaml
global:
  environment:
    models_3d: '${KIPRJMOD}/packages3D'
  restore_project: true

preflight:
  run_drc: false
  run_erc: false
  filters:
    @KICI_PREFLIGHT_FILTERS@
  set_text_variables:
    - name: 'VCS_PCB_DATE'
      command: 'date -d @$(git log -1 --format="%at" "${KIBOT_PCB_NAME}") -Idate'
    - name: 'VCS_PCB_TIMEDATE'
      command: 'date -d @$(git log -1 --format="%at" "${KIBOT_PCB_NAME}") -Iseconds'
    - name: 'VCS_PCB_HASH'
      command: 'git log -1 --format="%H" "${KIBOT_PCB_NAME}"'
      before: 'Git <'
      after: '>'
    - name: 'VCS_SCH_TIMEDATE'
      command: 'date -d @$(git log -1 --format="%at" "${KIBOT_SCH_NAME}") -Iseconds'
    - name: 'VCS_SCH_HASH'
      command: 'git log -1 --format="%H" "${KIBOT_SCH_NAME}"'
      before: 'Git <'
      after: '>'
    - name: 'VCS_TAG'
      command: 'git describe --always --dirty'
    - name: 'SHEETPATH' # Hack to avoid W082 Unknown sheetpath
      text: ''

...
definitions:
  KICI_PREFLIGHT_FILTERS: |
    - error: 'fake_dummy_error'
```

But here I get
```
ERROR:Error loading YAML expected '<document start>', but found '<block mapping start>'
  in "<file>", line 32, column 1 (kibot.gs - gs.py:788)
```
where line 32 is the `...` or `definitions:` block, depending if you start counting at 0 or 1.

Instead of the variable replacement, I can do another import, but of course, but I quite liked the `@SYNTAX@` variable substitution.

set-soft, 2024-01-18:
You are missing the `kibot` section, so you don't have a config document, will see if I can inform this error

closed 2024-01-18 by set-soft

oliv3r, 2024-01-18:
> You are missing the `kibot` section, so you don't have a config document, will see if I can inform this error

Interesting, so should the 'common' import also needs to be a valid (versioned) kibot. Good to know.

oliv3r, 2024-01-23:
@set-soft so one issue with import I found, is that it doesn't merge yaml segments.

For example, I have a `common.kibot.yaml` `drc.kibot.yaml` and a `erc.kibot.yaml`. Each *rc file includes the common one, where the common one sets the erc and drc to false. running the erc or drc sets the preflight section to 'true' where applicable. However having a `lint.kibot.yaml` which includes both *rc (and implicitly the common one) fails (This of course because the second re-import of common sets erc to false again).

This would of course be solved by ensuring imports are only imported once (in C you'd have if-guard blocks of course).

Another problem with import, is it doesn't merge yaml sections. So if I have a filter defined in my erc file, and a filter in my drc, one import overwrites the other import.

set-soft, 2024-01-23:
1. I don't understand why common.kibot.yaml sets run_erc and run_drc to false, this is the default. So I think your problem is in the eay you designed the files

2. You can select which stuff is imported from each file, so you can avoid overwrites and collisions when using filters with the same name (which you can avoid using unique names)

set-soft, 2024-01-24:
> I set them to false, to clearly indicate intent, that 'normally' these are not run.

Which isn't needed because this is the default, and doing it generates a problem to your workflow ...

> But generally, I'd like to use normally programming paradimes, e.g. 'import/include' 

But here you are dealing with data, not programs.

> that does its thing once, and then doesn't touch it anymore. Yaml objects that get 'merged' (which isn't a standard yaml feature, but some libaries supprot). All kinds of things that as a _programmer_ you'd expect, without making things too complex. e.g. creating the filter twice under different names etc. it's just kludgy, but I suppose that's just what yaml is.

This is because you are trying to force things to be like C/C++, instead of using the rich set of options.

> So in short, I made it work with imports, but could have been nicer if yaml where just a little easier to work with :)

The "force to false" example is a clear sign that you are forcing things instead of learning how to use them. I won't break compatibility to make the data look like a program, it will just favor some style in detriment of another, and break stuff, not an option.