INTI-CMNB/KiBot #560: [BUG]Kibot removes Mounting Hardware like SMD spacer

MarkusSBS, 2024-01-16:
May be related to https://github.com/INTI-CMNB/KiBot/issues/201

Kibot removes mounting Hardware like SMD spacer from the bom 

[image]
[image]
[image]
https://www.we-online.com/components/products/datasheet/9774030360R.pdf

The part is removed from all Kicad Output files 

Is this a bug or is there a flag to prefent kicad to remove this parts?

Best regards

Markus

Environment (please complete the following information):
Where are you running KiBot:

In a CI/CD workflow? (GitLab)
Using a docker image? ghcr.io/inti-cmnb/kicad7_auto_full:dev

set-soft, 2024-01-16:
Hallo @MarkusSBS !

Can you post a small example containing this component? Just a couple of component. The name of the footprints shouldn't be a problem.

MarkusSBS, 2024-01-16:
I send you an email with a full sample projekt

set-soft, 2024-01-16:
Hallo @MarkusSBS !

I got your project example, if you run KiBot with log information (i.e. adding `-L logfile`) you'll find the reason, it says:

```
DEBUG:Excluding 'H3': Field 'part' (MountingHole_Pad) matched 're.compile('^mount.*hole', re.IGNORECASE)' (kibot - fil_generic.py:170)
```

The problem here is that you are using a mounting hole symbol (Mechanical:MountingHole_Pad) for two things:

1. A real mounting hole, a pad with a hole (with footprint *Mounting_Wuerth:Mounting_Wuerth_WA-SMSI-M1.6_H3mm_9774030633*)
2. An SMD board separator 

I guess the simplest solution is to use a custom symbol, just copy *Mechanical:MountingHole_Pad* to a new symbol library called *Mounting_Wuerth* and name the component *Mounting_Wuerth* or something similar, you could even name it *Mounting_Wuerth_WA-SMSI-M1.6_H3mm_9774030633* and indicate the footprint is *Mounting_Wuerth:Mounting_Wuerth_WA-SMSI-M1.6_H3mm_9774030633*. In this way you'll have a symbol with the same name as the footprint. 

Doing this will avoid the confusion between a regular mounting hole and an SMD separator (which needs a mounting hole to be placed).

You can also split the symbol in two. Leave the current mounting hole and add a symbol for the SMD separator. Then mark this symbol as excluded from the PCB. This is more complex, but also more clear. In this way people knows you have a mounting hole and also an SMD separator. 

Of course you could define your own filter to replace the *_mechanical* filter, and make the regular expressions more narrow, but this is complex.

closed 2024-01-16 by set-soft

edlongman, 2024-03-06:
@set-soft  This has also recently caught me out when using a PCB mount cable clamp in an ordinary mounting hole. I'm not sure it's a good idea to silently perform this filtering, given that KiCad already has a field to exclude something from the BOM.

set-soft, 2024-03-06:
Hi @edlongman !

I think the solution is to narrow the filter. IMO changes in behavior are worst than this issue.

edlongman, 2024-03-13:
I think the documentation doesn't clearly state what is done either. It suggests it is to do with the exclude filter but I think the mounting holes is done separately.

``` exclude_filter : [string|list(string)=’_mechanical’] Name of the filter to exclude components from BoM processing. The default filter excludes test points, fiducial marks, mounting holes, etc. This option is for simple cases, consider using a full variant for complex cases. ```

set-soft, 2024-03-19:
Hi @edlongman !
The above patch adds more information to the docs.