INTI-CMNB/KiBot #561: [FEATURE] When mirroring a layer with text, allow the text to be mirrored 'back'

oliv3r, 2024-01-16:
When adding dimensions via the 'User.Drawings' layer (kicad default), we can put dimensions in documents. This looks great, until wil print a layer stack (pdf) mirrored.

```yaml
          layers:
          - layer: 'B.Cu'
          - layer: 'B.Mask'
            color: '#14332440'
          - layer: 'B.SilkS'
            color: '#f2eda1ff'
          - layer: 'User.Comments'
          mirror: true
```

However while its understandable that silk-screen gets mirrored (though in this case it gets unmirrored due to it being the back silkscreen), the dimensions now get mirrored numbers.

[image]

So what works great for the layers needed for fab, doesn't work for User.Comments/dimensions. What we should do however (with a attribute) re-mirror just the `gr_text` in the `Cmts.User` layer.
```kicad_pcb
    (gr_text "48,3 mm" (at 109.22 50.920001) (layer "Cmts.User") (tstamp c4fd811c-612b-4b7c-9986-befbd6bb585d)
      (effects (font (size 1 1) (thickness 0.15)) (justify mirror))
    )
```

Obviously we should also honor the `restore_project` flag, to ensure the repository is left as we found it.

It seems, that if no property is set, `justify` is intended, which is a bit weird, so may need a double check.

set-soft, 2024-01-18:
Ok, this is a little bit tricky, but I think the above patch does the work.

closed 2024-01-18 by set-soft

oliv3r, 2024-01-19:
@set-soft amazing, I didn't even know you fixed this, and was surprised that my dimensions all of a sudden where correct :)

set-soft, 2024-01-19:
> @set-soft amazing, I didn't even know you fixed this, and was surprised that my dimensions all of a sudden where correct :)

I´m scanning all the PCB elements and also all the elements inside footprints. The algorithm just looks for objects that inherits from the basic text class and implements *GetShownText*, I tried with all stuff that came to my mind and worked, but I can't be sure all objects are covered.

> I'm not sure I understand the options correctly, for me it was 'just working' with `mirror: true`. The PCB text is indeed mirrored, Regardless, the default work :D

The idea is that it should work *out-of-the-box*. As KiCad doesn't mirror text in the user layers when you explicitly ask for mirrored output the default is to just mirror the text. But you can disable it (in case you manually mirrored the text!). One option controls the text objects that are in the board and the other the objects inside footprints.

The code also adjusts the alignment (justification) of the text, so right justified text becomes left justified when mirrored.