INTI-CMNB/KiBot #564: [BUG] JLCPCB fab output incorrectly handles 3D files

oliv3r, 2024-01-19:
I hadn't run the jlcpcb output for a while, and hadn't spotted the warning before, but it seems like jlcpcb wants the 3D environment, but it's not using it from the global environment. My repo is self-contained, in that the 3D models are in the repo itself.

```yaml
global:
  environment:
    models_3d: '${KIPRJMOD}/packages3D'
```

Ironically, the render3d and blender jobs generate the 3d models just fine, which also require 3D models of course.

The only thing I found interesting, is that I see files such as `./kibot/resources/config_templates/JLCPCB_with_THT.kibot.yaml` and I also know that kibot generates THT parts on the fly for the blender renders ... but for the fab step, this sounds odd to me. After all, these steps should only be generating gerbers, drill files and assembly instructions?

In the blender run, i see these lines in my log (why are they for both 6 and 7?)
```
DEBUG:Using KICAD7_3DMODEL_DIR="/workspace/hello/packages3D" (from environment) (kibot - config.py:387)
DEBUG:Using KICAD6_3DMODEL_DIR="/workspace/hello/packages3D" (from environment) (kibot - config.py:387)
```
So if its working for blender, why not for jlcpcb?

The JLCPCB log bit below

```
- 'BoM for JLCPCB' (_JLCPCB_bom) [bom]
DEBUG:Output destination: /workspace/hello/outputs/fab/JLCPCB (kibot - kiplot.py:439)
DEBUG:Filters reset (kibot - fil_base.py:194)
DEBUG:Applying filter `_only_jlc_parts` to exclude (kibot - fil_base.py:187)
DEBUG:Looking for LCSC field name (kibot - optionable.py:530)
DEBUG:Using LCSC as LCSC field name (kibot - optionable.py:532)
DEBUG:Applying filter `_kibom_dnf_Config` to fitted (kibot - fil_base.py:205)
DEBUG:Applying filter `_kibom_dnc_Config` to fixed (kibot - fil_base.py:215)
DEBUG:Applying KiBoM style variants `default` (kibot - var_kibom.py:106)
DEBUG:Creating internal filter: {'type': 'expand_text_vars', 'comment': 'Internal default text variables expander', 'name': '_expand_text_vars'} (kibot - fil_base.py:320)
DEBUG:Applying transform filter `_expand_text_vars` (kibot - fil_base.py:173)
DEBUG:Reading KiCad config from `/tmp/.config/kicad/7.0/kicad_common.json` (kibot - config.py:441)
DEBUG:Using KICAD7_SYMBOL_DIR="/usr/share/kicad/symbols" (guessed) (kibot - config.py:416)
DEBUG:Using KICAD7_TEMPLATE_DIR="/usr/share/kicad/template" (guessed) (kibot - config.py:416)
DEBUG:Using KICAD_USER_TEMPLATE_DIR="/tmp/.local/share/kicad/7.0/template" (guessed) (kibot - config.py:416)
DEBUG:Using KICAD7_FOOTPRINT_DIR="/usr/share/kicad/footprints" (guessed) (kibot - config.py:416)
WARNING:(W010) Unable to find KiCad 3D models (kibot - config.py:433)
```

set-soft, 2024-01-22:
> I hadn't run the jlcpcb output for a while, and hadn't spotted the warning before, but it seems like jlcpcb wants the 3D environment, but it's not using it from the global environment. My repo is [...]

KiBot always looks for the KiCad environment variables, no matter which outputs are defined.

> In the blender run, i see these lines in my log (why are they for both 6 and 7?)
> 
> ```
> DEBUG:Using KICAD7_3DMODEL_DIR="/workspace/hello/packages3D" (from environment) (kibot - config.py:387)
> DEBUG:Using KICAD6_3DMODEL_DIR="/workspace/hello/packages3D" (from environment) (kibot - config.py:387)
> ```

Because you are using KiCad 7 and KiCad 7 libs needs both. All the official footprints uses *KICAD6_3DMODEL_DIR* (a bug IMHO), but all the environment variables are named *KICAD7_*. KiCad is "smart" enough to use any of them, but I don't trust in such a "smart" behavior and defining both is more robust. Here KiBot is informing you that it will apply the same "smart" behavior found in KiCad: use *KICAD7_3DMODEL_DIR* where *KICAD6_3DMODEL_DIR* is needed.

> So if its working for blender, why not for jlcpcb?

Without an example is hard to know. My guess is that you aren't defining *models_3d* when invoking the JLCPCB outputs, only when invoking the render outputs, after all you think this isn't needed. And in fact isn't needed and this is why you get a warning, not an error.

oliv3r, 2024-01-22:
I don't think that's true, as you can see from the pipeline (which has full log) and all my files, you'll see that I have 1 file that defines my globals, which _all_ jobs use, so it should work exactly the same for all jobs?

https://gitlab.com/olliver/lapod/-/pipelines/1145922251

set-soft, 2024-01-23:
If you think this is a bug, and not a bizarre use case, create a small example that shows it. 

Reduce the workflow to the minimal needed to expose the issue.

oliv3r, 2024-01-23:
I don't think it's a complex case at all; I've boiled down the repo to make it very simple https://gitlab.com/ci-includes/kici/-/pipelines/1148005113 just one job, just one fab.kibot.yaml.

oliv3r, 2024-01-23:
Not sure how complex the pipeline is; but here a simple pipeline with just one job, the fab output.

https://gitlab.com/ci-includes/kici/-/pipelines/1148306273

If I copy the contents of my imported file instead, it does work just fine; so it is related to the fact that things are being imported.

oliv3r, 2024-01-23:
I have 2 import statements, which should have been combined into one :)

I think the bug report stands, but I figured it out :) Why does JLC need the 3D stuff, and the others not.

Here's a job that will run just the fab stage, without the import, for pcbway and jlc, I think this is 'smallest' way to show the bug.

set-soft, 2024-01-23:
Hi @oliv3r !

This is really far from simple. I need something I can feed into KiBot and get the error, not tons of scripts that handles tons of particular details. I need a few KiBot configuration files.

oliv3r, 2024-01-23:
@set-soft not sure what you mean, there's 1 single kibot script in that pipeline,and both kicad projects (hello and world) fail on it.

So to trigger the bug locally, I do:

```sh
docker run --pull always --user "$(id -u):$(id -g)" --env HOME=/tmp --rm -it -v $(pwd):/workspace -w /workspace ghcr.io/inti-cmnb/kicad7_auto_full:dev /bin/bash
Status: Downloaded newer image for ghcr.io/inti-cmnb/kicad7_auto_full:dev
I have no name!@f6d0af5f0d76:/workspace$ cd hello
I have no name!@f6d0af5f0d76:/workspace/hello$ kibot -c .kibot/kici_fab.kibot.yaml
Using SCH file: hello.kicad_sch
- 'Gerbers compatible with PCBWay' (_PCBWay_gerbers) [gerber]
- 'Drill files compatible with PCBWay' (_PCBWay_drill) [excellon]
- 'Gerbers compatible with JLCPCB' (_JLCPCB_gerbers) [gerber]
- 'Drill files compatible with JLCPCB' (_JLCPCB_drill) [excellon]
- 'Pick and place file, JLCPCB style' (_JLCPCB_position) [position]
- 'BoM for JLCPCB' (_JLCPCB_bom) [bom]
WARNING:(W008) Unable to find KiCad configuration file (/tmp/.config/kicad/7.0/kicad_common.json) (kibot - config.py:199)
WARNING:(W010) Unable to find KiCad user templates (kibot - config.py:433)
WARNING:(W010) Unable to find KiCad 3D models (kibot - config.py:433)
- 'ZIP file for PCBWay' (_PCBWay_compress) [compress]
- 'ZIP file for JLCPCB' (_JLCPCB_compress) [compress]
Found 3 unique warning/s (3 total)
```
(or `kibot -c ../.kibot/kici_fab.kibot.yaml` does the same, so it's not related to the symlink or anything.)

E.g. To reprodcue only 4 files are needed, 3 kicad files and 1 fab.yaml file. nothing else, no scripts or anything strange :)

oliv3r, 2024-01-24:
> > @set-soft not sure what you mean, there's 1 single kibot script in that pipeline,and both kicad projects (hello and world) fail on it.
> 
> But take a look at your CI/CD workflow:
> 
> ```yaml
> include:
>   # Do not use $CI_COMMIT_SHA in any project, this is for testing here only!
>   - component: $CI_SERVER_HOST/$CI_PROJECT_PATH/gitlab-component@$CI_COMMIT_SHA
>     inputs:
>       stage_lint: 'lint'
>       stage_build: 'build'
>       attach_diff: 'true'
>       projects: 'hello world'
> ```
> 
> Where is the KiBot call here? I won't investigate it ...
Right here in the template; but it's not relevant, no need to investigate :)
> 
> > So to trigger the bug locally, I do:
> > ```shell
> > docker run --pull always --user "$(id -u):$(id -g)" --env HOME=/tmp --rm -it -v $(pwd):/workspace -w /workspace ghcr.io/inti-cmnb/kicad7_auto_full:dev /bin/bash
> > Status: Downloaded newer image for ghcr.io/inti-cmnb/kicad7_auto_full:dev
> > I have no name!@f6d0af5f0d76:/workspace$ cd hello
> > I have no name!@f6d0af5f0d76:/workspace/hello$ kibot -c .kibot/kici_fab.kibot.yaml
> > Using SCH file: hello.kicad_sch
> > - 'Gerbers compatible with PCBWay' (_PCBWay_gerbers) [gerber]
> > - 'Drill files compatible with PCBWay' (_PCBWay_drill) [excellon]
> > - 'Gerbers compatible with JLCPCB' (_JLCPCB_gerbers) [gerber]
> > - 'Drill files compatible with JLCPCB' (_JLCPCB_drill) [excellon]
> > - 'Pick and place file, JLCPCB style' (_JLCPCB_position) [position]
> > - 'BoM for JLCPCB' (_JLCPCB_bom) [bom]
> > WARNING:(W008) Unable to find KiCad configuration file (/tmp/.config/kicad/7.0/kicad_common.json) (kibot - config.py:199)
> > WARNING:(W010) Unable to find KiCad user templates (kibot - config.py:433)
> > WARNING:(W010) Unable to find KiCad 3D models (kibot - config.py:433)
> > - 'ZIP file for PCBWay' (_PCBWay_compress) [compress]
> > - 'ZIP file for JLCPCB' (_JLCPCB_compress) [compress]
> > Found 3 unique warning/s (3 total)
> > ```
> 
> Next time modify your workflow to just do this.
But the workflow IS just this, you can see here: https://gitlab.com/olliver/sandbox/bugs/kici/-/blob/simple_test/templates/gitlab-component.yml?ref_type=heads#L139
```
          kibot \
                ${DEBUG_TRACE_CI:+--verbose --verbose --verbose --verbose} \
                --defs-from-env \
                --no-warn 9,132 \
                --plot-config "${_job}" \
                --stop-on-warnings
          ${KIJOB_ATTACH_DIFF_PDF:+atta

```
the rest of the pipeline just creates some folders and exports some variables.
> 
> > (or `kibot -c ../.kibot/kici_fab.kibot.yaml` does the same, so it's not related to the symlink or anything.)
> > E.g. To reprodcue only 4 files are needed, 3 kicad files and 1 fab.yaml file. nothing else, no scripts or anything strange :)
> > (I forgot to do `kicad-cli --version` to remove W008, and I haven't seen the user-templates one, but that one doesn't show in the pipeline, so not too bothered there).
> 
> Running the example you show the only problem is that you forgot to import _kici_common.kibot.yaml_:
I removed it altogether, I made it as simple as possible, just to show, that PCBWAY does NOT require the 3d environment variable, but JLCPCB DOES require the 3d environment variable!
> 
> ```diff
> diff --git a/.kibot/kici_fab.kibot.yaml b/.kibot/kici_fab.kibot.yaml
> index 60a2849..ce1e88a 100644
> --- a/.kibot/kici_fab.kibot.yaml
> +++ b/.kibot/kici_fab.kibot.yaml
> @@ -2,6 +2,7 @@ kibot:
>    version: 1
>  
>  import:
> +  - file: kici_common.kibot.yaml
>    - file: 'PCBWay'
>      definitions:
>        _KIBOT_MANF_DIR: 'outputs/fab/PCBWay'
> ```
> 
> This removed the warning, and remember to also remove the unneeded _run_erc: false_ stuff

But the warning is still there?

set-soft, 2024-01-24:
> the rest of the pipeline just creates some folders and exports some variables.

Next time just put it as the only action, not included from some other place.

> Ofcourse this removes the warning, as that set the needed environment variable. However, JLCPCB needs it, PCBWAY does not. That was my point :)

KiBot needs it. If you don't like the warning just filter it or provide a definition.

closed 2024-01-24 by set-soft

oliv3r, 2024-01-25:
> KiBot needs it. If you don't like the warning just filter it or provide a definition.

I'm fine with kibot needing it, but why not with PCBWAY (and the other fabs), why ONLY with JLCPCB?

set-soft, 2024-01-25:
> I'm fine with kibot needing it, but why not with PCBWAY (and the other fabs), why ONLY with JLCPCB?

Different outputs, different data needs to be collected.

set-soft, 2024-01-25:
KiBot tries to avoid collecting everything and then do the work, instead collects data as needed. But some data comes in packs, so when you get the KiCad environment all the environment is solved, not just one variable.

oliv3r, 2024-01-26:
> > I'm fine with kibot needing it, but why not with PCBWAY (and the other fabs), why ONLY with JLCPCB?
> 
> Different outputs, different data needs to be collected.

And that is perfectly and fully understandable. I can reason with logic :) A job that requires the 3d models, will require the 3d paths to be setup :)

> KiBot tries to avoid collecting everything and then do the work, instead collects data as needed. But some data comes in packs, so when you get the KiCad environment all the environment is solved, not just one variable.

Sure, which is why I said initially, I figured it out, my include was wrong, I can work around the problem by just always defining the environment, no problem.

However, the question remains, what is so different about JLCPCB that requires 'more' of the variables? With that 'pack' coment though, I figure, because we also need POS files and other things for JLC's assembly service, that requires more environment variables, which 'accidentally' also needs the 3d environment stuff, even though JLCPCB doesn't really do anything with the 3D models.

Could have just said that :p but it's mostly clear now. So if you think it's not worth the effort to 'fix' this, I fully agree and understand.

set-soft, 2024-03-27:
Hi @oliv3r !

The above patch detects official docker images, GitLab and GitHub CI/CD workflows. In these cases W009 is omitted and two cases of W010 are also ignored.

We could add more warnings that are always found on CI/CD workflows.

Of course this behavior can be disabled, just using ` --warn-ci-cd`