INTI-CMNB/KiBot #568: [HOW] to overwrite revision and date in BOM

andersruneson, 2024-01-24:
**What do you want to achieve?**
I want to set (replace) the values of two fields in the BOM called revision and date. See screen:
[image]

I'm guessing those are set by the %r and %d, and the readme says "If you need to force the origin of the data you can use %bX for the PCB and %sX for the schematic, where X is the pattern to expand."

I've tried running KiBot with the following:

> -g r="testgR" -g %r="testgprcR" -g br="testgBR" -g %br="testgprcBR" -g %sr="testsprcBR" -g sr="testsBR"

but the revision is still "draft". 

I guess those are set some other way?

set-soft, 2024-01-24:
Hi @andersruneson !

Yes, they come from the PCB/Schematic. You can't change them. What you can actually do is use variables in your PCB and/or schematic. 

I usually use a KiCad variable for both. Then I define this variable inside KiCad (using the GUI) to indicate this is a work in progress revision. Finally I use *set_text_variables* in KiBot to give these variables a suitable value, i.e. the revision is the short git hash for the commit and the date is the date of the commit.

[image]

[image]

```yaml
preflight:
  set_text_variables:
    - name: REPO_DATE_SCH
      command: 'git log -1 --format="%as" -- "$KIBOT_SCH_NAME"'
    - name: REPO_HASH_SCH
      command: 'git log -1 --format="%h" -- "$KIBOT_SCH_NAME"'
    - name: REPO_DATE_PCB
      command: 'git log -1 --format="%as" -- "$KIBOT_PCB_NAME"'
    - name: REPO_HASH_PCB
      command: 'git log -1 --format="%h" -- "$KIBOT_PCB_NAME"'
```

closed 2024-01-24 by set-soft

andersruneson, 2024-01-25:
Thank you, that works excellent!