INTI-CMNB/KiBot #572: [FEATURE] Export KiCanvas with schematic like Interative BOM

andersruneson, 2024-01-26:
**Is your feature request related to a problem? Please describe.**
Well, when exporting gerbers, schematics etc with KiBot it would be nice to also get a stand-alone KiCanvas schematic (https://kicanvas.org/home/) just like the interactive BOM. I'm not sure KiCanvas can be packaged like that so I might need to suggest this to KiCanvas first.

**Describe the solution you'd like**
Add an output, kicanvas, where the schematic will be distributed with KiCanvas runtime, maybe the files need to be put in a folder, and then a .html-file that opens the KiCanvas in the local browser.

**Describe alternatives you've considered**
Ah, we can use the schematic pdf, but KiCanvas gives more features of course.

**Additional context**
[image]

set-soft, 2024-01-26:
Interesting! the script is self contained, it has some issues, but looks really promising

set-soft, 2024-01-26:
Hi @andersruneson !

Ok, I added some basic support. Some stuff isn't really working (quick-start example and Makefile use are broken). For testing I'm using:

```yaml
# Example KiBot config file
kibot:
  version: 1

outputs:
  - name: 'KiCanvas'
    comment: "Example of KiCanvas export"
    type: kicanvas
    dir: KiCanvas
    options:
      source: ['schematic', 'pcb', 'project']
      overlay: false
      # local_script: false
      # source: ['schematic', 'pcb', 'project', 'pp']
      # source: pp
```

andersruneson, 2024-01-26:
Wow fast, I'll have a try! Thanks.

bluespider42, 2024-01-29:
Thanks for this - seems to be working well. Only thing missing for me is that only the top level schematic is loaded.

andersruneson, 2024-01-30:
Hm, I get the following output:
```
- 'Example of KiCanvas export' (KiCanvas) [kicanvas]
ERROR:/builds/Parts/EA/EA00189/source/out/KiCanvas/EA00189-kicanvas.html (kibot - out_kicanvas.py:98)
ERROR:/builds/Parts/EA/EA00189/source/out/KiCanvas/kicanvas.js (kibot - out_kicanvas.py:98)
ERROR:/builds/Parts/EA/EA00189/source/out/KiCanvas/y.lib (kibot - out_kicanvas.py:98)
ERROR:/builds/Parts/EA/EA00189/source/out/KiCanvas/n.lib (kibot - out_kicanvas.py:98)
ERROR:/builds/Parts/EA/EA00189/source/out/KiCanvas/sym-lib-table (kibot - out_kicanvas.py:98)
ERROR:/builds/Parts/EA/EA00189/source/out/KiCanvas/EA00189.kicad_sch (kibot - out_kicanvas.py:98)
ERROR:/builds/Parts/EA/EA00189/source/out/KiCanvas/EA00189.kicad_pcb (kibot - out_kicanvas.py:98)
ERROR:/builds/Parts/EA/EA00189/source/EA00189.kicad_pro (kibot - out_kicanvas.py:98)
ERROR:/builds/Parts/EA/EA00189/source/EA00189.kicad_prl (kibot - out_kicanvas.py:98)
```

From this config:
```
# Example KiBot config file
kibot:
  version: 1

outputs:
  - name: 'KiCanvas'
    comment: "Example of KiCanvas export"
    type: kicanvas
    dir: KiCanvas
    options:
      source: ['schematic', 'pcb', 'project']
      overlay: false
      # local_script: false
      # source: ['schematic', 'pcb', 'project', 'pp']
      # source: pp
```

set-soft, 2024-01-30:
> Hm, I get the following output:
> 

This is just for debug, I was fixing the targets and wanted to list them in a very visible way. Now they are fixed.

> From this config:
>
> ```
> # Example KiBot config file
> kibot:
>   version: 1
>
> outputs:
>   - name: 'KiCanvas'
>     comment: "Example of KiCanvas export"
>     type: kicanvas
>     dir: KiCanvas [...]

Try excluding the project, it doesn't work as expected.

> The generated html-file renders completely empty in chome, but has the following content which looks decent:
> 
> ```
> <!DOCTYPE HTML>
> <html lang="en">
>   <body>
>     <script type="module" src="kicanvas.js"></script>
>     <kicanvas-embed controls="full" controlslist="nooverlay">
>       <kicanvas-source src="EA00189.kicad_sch"></kicanvas-source>
>       <kicanvas-source src="EA00189.kicad_pcb"></kicanvas-source>
>       <kicanvas-source src="EA00189.kicad_pro"></kicanvas-source>
>     </kicanvas-embed>
>   </body>
> </html>
> ```

Which browser are you using? For chrome you need something like:

```
google-chrome --allow-file-access-from-files HTML_FILE
```

Don't know about other browsers. This is because the Java Script is loading files from the local filesystem using a mechanism that is for the network. This cross reference is forbidden.

The last patch solves some problems with sub-sheets, but KiCanvas needs a lot of adjusts on this side. I added an issue to discuss it: theacodes/kicanvas#77

set-soft, 2024-01-31:
> Don't know about other browsers. This is because the Java Script is loading files from the local filesystem using a mechanism that is for the network. This cross reference is forbidden.

I added a Java Script check to inform this problem. And also suggested upstream to add such a check (theacodes/kicanvas#78)

I'm closing this issue, bugs in the KiCanvas output should go to new issues.

closed 2024-01-31 by set-soft

andersruneson, 2024-02-01:
It works, well done! Hope kicanvas fixes the issues you reported.

commit de1fc052 (2024-01-26): [KiCanvas][Added] Some very raw support

Closes #572