INTI-CMNB/KiBot #573: [FEATURE] PDF Scaling should honor grid origin

oliv3r, 2024-01-26:
Currently, when scaling, kibot scales from the 0,0 starting point, and everything becomes bigger towards the bottom right corner. Or maybe even, with 0,0 as our starting point, we 'bloom' outwards.

However, not only does the bottom right corner contain the PCB worksheet stuff, we also dissproportionally push stuff into the right corner, so that the result is off-center.

Instead, we should honor the grid-origin as our starting point, and bloom outwards from there. The result is that we scale nicely and proportionally, so that the image always remains in the center, and 'blooms' outwards from the center (center being where `grid_origin` is defined. This also fills the image more equally and is slightly less likely to overrun the worksheet corner.

Also this helps tremendously with the 'mirroring' feature, because mirroring is done after scaling, a mirrored page is also of center by the opposite amount. Also here, by scaling based on the `grid_center` we completely avoid/ignore this issue, resulting in far nicer scaled images.

Obviously this does require the user to define the `grid_origin` in the pcb file, but that should be a good idea anyway.

(If `grid_origin is deemed inappropriate, we can also use `aux_axis_origin` as well).

oliv3r, 2024-02-01:
@set-soft Absolutely! and sorry for the delay :)

[image]

Here you can see, that with a scaling of 1.5, the image is scaled from the top left, towards the bottom left. You can see the center is no longer the center.

```yaml
kibot:
  version: 1

import:
  - file: 'kici_common.kibot.yaml'

outputs:
  - name: 'pdf_sch_print'
    comment: 'Exports the schematic to the most common exchange format. Suitable for printing.'
    type: 'pdf_sch_print'
    dir: 'outputs/pdf'
    options:
      all_pages: true
      background_color: false
      frame: true
      monochrome: false
  - name: 'pdf_pcb_print'
    comment: 'Exports the PCB to the most common exchange format. Suitable for printing.'
    type: 'pcb_print'
    dir: 'outputs/pdf'
    options:
      force_edge_cuts: true
      format: 'PDF'
      individual_page_scaling: false
      keep_temporal_files: false
      scaling: @KICI_PDF_SCALING@
      pages:
        - sheet: 'PCB top'
          layers:
          - layer: 'F.Cu'
          - layer: 'F.Mask'
            color: '@KICI_PDF_COLOR_PCB_MASK@'
          - layer: 'F.SilkS'
            color: '@KICI_PDF_COLOR_PCB_SILK@'
          - layer: 'User.Comments'
        - sheet: 'Fabrication top'
          layers:
          - layer: 'F.Fab'
          - layer: 'F.Paste'
          - layer: 'F.Adhes'
          - layer: 'User.Comments'
        - sheet: 'Silkscreen top'
          layers:
          - layer: 'F.SilkS'
        - sheet: 'F.Cu'
          layers:
          - layer: 'F.Cu'
        - sheet: '%ln layer'
          repeat_for_layer: 'F.Cu'
          repeat_layers: 'inners'
          layers:
          - layer: 'F.Cu'
        - sheet: 'B.Cu'
          layers:
          - layer: 'B.Cu'
        - sheet: 'Silkscreen bottom'
          layers:
          - layer: 'B.SilkS'
          mirror: true
        - sheet: 'Fabrication bottom'
          layers:
          - layer: 'B.Fab'
          - layer: 'B.Paste'
          - layer: 'B.Adhes'
          - layer: 'User.Comments'
          mirror: true
        - sheet: 'PCB bottom'
          layers:
          - layer: 'B.Cu'
          - layer: 'B.Mask'
            color: '@KICI_PDF_COLOR_PCB_MASK@'
          - layer: 'B.SilkS'
            color: '@KICI_PDF_COLOR_PCB_SILK@'
          - layer: 'User.Comments'
          mirror: true
      plot_sheet_reference: true

...
definitions:
  KICI_PDF_SCALING: 1.5
  KICI_PDF_COLOR_PCB_MASK: '#14332440'
  KICI_PDF_COLOR_PCB_SILK: '#f2eda1ff'
```

(Luckily) the dimensions are unrelated, as a pure silkscreen page also has this same alignment issue. I'll attach the full pdf for reference.
[breakout_hdmi-assembly.pdf](https://github.com/INTI-CMNB/KiBot/files/14122786/breakout_hdmi-assembly.pdf)

closed 2024-02-01 by set-soft

oliv3r, 2024-02-01:
@set-soft your amazing!!

> Ok, now I understand what do you mean with "scaled from the top left, towards the bottom left.", but this isn't the case. > The image is scaled and then centered.
Ahh, so it IS the dimensions that is responsible here. What confused me I suppose, was the fact that the layers without dimensions where also incorrectly centered?

If you would have centered first, and then scaled, it would have also worked? But I like your trick, it's good!

After pasting in those changes, I was a bit confused about the invert/force options, they read a little unnatural at first, but
[image]

perfectly centered!

I'm so excited and happy now, I got all main functions of kibot working.

I'll figure out the re-use files bit out in the next few weeks, so that I can do present and navigate; and probably add kicanvas to the mix at that point too :)

Btw, when do you expect to release 1.6.4? When is the feature freeze? I see you are working on kicanvas now.

set-soft, 2024-02-01:
> > Ok, now I understand what do you mean with "scaled from the top left, towards the bottom left.", but this isn't the case. 
> > The image is scaled and then centered.

>  Ahh, so it IS the dimensions that is responsible here. What confused me I suppose, was the fact that the layers without dimensions where also incorrectly centered?

But they ARE correctly centered!! The other pages affects the centering!! (remember you can ask for individual centering)

> If you would have centered first, and then scaled, it would have also worked? 

Not really, KiCad just computes the center, doesn't really matter the order. And is KiCad the one that does it.

> But I like your trick, it's good!
>
> After pasting in those changes, I was a bit confused about the invert/force options, they read a little unnatural at first, but [...]

I want to do a release ASAP.

oliv3r, 2024-02-01:
> But they ARE correctly centered!! 
Depending on the eye of the beholder :) but yes yes, they where :)

> I want to do a release ASAP.

For what it's worth, all the bugs/features I have found/was after are now in order, so I'd say it's "stable" enough.

Hopefully soon is in the next few days :p

oliv3r, 2024-02-03:
@set-soft thanks for the tag :D Makes me super happy :)

Btw, I think my original feature request is still somewhat valid. You currently scale and then center, but the center is based on the sheet. Instead, the centering should happen based on `grid_origin` instead, again still due to the worksheet border.

If you look at the following screenshot, you see I tried to center my PCB slightly in the upper half, because that's where it LOOKS more centered (vertically).
[image]

However, in the pdf, while perfectly according to instructions of course, it looks 'off' and not 'scaled but in the same location' anymore.

[image]

Logically, If `grid_origin` is unset in the pdf, what you do now is perfect. However, if it IS set, move the center of the _image_ to the center of the `grid_origin` instead.

This way, the center is what the user has designed the center to be (or rather, the `grid_origin`. I think you could also look for the dimensional center, `(dimension (type center) (layer "Dwgs.User")` in my case, but I think you can have multiple of those, so that'll become messy quickly, even though the intend is far more clear :)

set-soft, 2024-02-05:
Hi @oliv3r !

You can do this with 1.6.4! 

Just choose any empty layer you want to indicate the center, lets say *User.Eco1*. 

Now draw anything you want to be used as center, lets say a *center dimension* and save your PCB. Then use something like:

```yaml
kibot:
  version: 1

outputs:
  - name: 'print_front'
    comment: "Experiment"
    type: pcb_print
    options:
      plot_sheet_reference: true
      format: 'PDF'
      scaling: 2.0
      pages:
        - layers:
            - layer: User.Eco1
              color: "#FFFFFF00"
              use_for_center: true
            - layer: F.Cu
              use_for_center: false
            - layer: Edge.Cuts
              color: "#FF4040"
              use_for_center: false
```

Do you get the idea? Explanation:

1. The first layer we plot is *User.Eco1* and we say this layer is the only one used for centering purposes. We also use white color to make it the same as the default background.
2. We plot the other layers on top of the *User.Eco1*, so the center dimension can't be seen

That's all, the center is defined by the drawings in the auxiliary layer. You have full control over what is used to center the PCB.

oliv3r, 2024-02-09:
Ohh That is very cool. So this is mutually exclusive to
```
      force_edge_cuts: true
      forced_edge_cuts_use_for_center: false
      invert_use_for_center: true
```
however, right? I mean, Apparently the edge-cuts aren't being calculated properly for centering purposes (or just make it hard to properly calculate or whatever). So instead, I should use the `use_for_center`.

A little trickier then I first thought, as it needs to be on each layer. 'force_center_layer: User.Eco1' would also be cool ;) which works like force_edge_cuts, but you can also specify which layer to use to find the center.

Though tbf, if we use edge-cuts for center calculation, we'd also be there already.

But I already put the center-dimension on a specific layer :)

set-soft, 2024-02-09:
> Ohh That is very cool. So this is mutually exclusive to
> 
> ```
>       force_edge_cuts: true
>       forced_edge_cuts_use_for_center: false
>       invert_use_for_center: true
> ```

Not to *force_edge_cuts: true*. And you could actually use *invert_use_for_center: true* to make it shorter.

> 
> however, right? I mean, Apparently the edge-cuts aren't being calculated properly for centering purposes (or just make it hard to properly calculate or whatever). So instead, I should use the `use_for_center`.

Nope, they are perfectly calculated, is just that you want to use a different center. Not the center of the board.

set-soft, 2024-02-13:
> Then I'm not sure how to do it correctly. I need to add it to every single `layers`, a `layer` attribute with `invert_use_for_center: true` correct? 

Nope, `invert_use_for_center: true` is not a layer attribute. This is an attribute for all pages.

> But the invert property is only applicable in the top directive?

Yes.

> Very concretly, if I have the following config, where do I put the `User.drawings` layer to use for centering?
> 
> ```yaml
> outputs:
>   - name: 'pdf_sch_print'
>     comment: 'Exports the schematic to the most common exchange format. Suitable for printing.'
>     type: 'pdf_sch_print'
>     dir: 'outputs/pdf'
>     options:
>       all_pages: true
>       background_color: false
>       frame: true
>       monochrome: false
>   - name: 'pdf_pcb_print'
>     comment: 'Exports the PCB to the most common exchange format. Suitable for printing.'
>     type: 'pcb_print'
>     dir: 'outputs/pdf'
>     options:
>       force_edge_cuts: true
>       forced_edge_cuts_use_for_center: false

Remove `forced_edge_cuts_use_for_center`

>       invert_use_for_center: true
>       format: 'PDF'
>       individual_page_scaling: false
>       keep_temporal_files: false
>       scaling: @KICI_PDF_SCALING@
>       pages:
>         - sheet: 'PCB top'
>           layers:
>           - layer: 'F.Cu'
>           - layer: 'F.Mask'
>             color: '@KICI_PDF_COLOR_PCB_MASK@'
>           - layer: 'F.SilkS'
>             color: '@KICI_PDF_COLOR_PCB_SILK@'
>           - layer: 'User.Comments'
>           - layer: 'User.Drawings'
>             color: '#ffffff00'
>             use_for_center: true

Change it to *false*, you are inverting it!
Same for the other pages.

>         - sheet: 'Fabrication top'
>           layers:
>           - layer: 'F.Fab'
>           - layer: 'User.Comments'
>           - layer: 'User.Drawings'
>             color: '#ffffff00'
>             use_for_center: true
>         - sheet: 'Silkscreen top'
>           layers:
>           - layer: 'F.SilkS'
>           - layer: 'User.Drawings'
>             color: '#ffffff00'
>             use_for_center: true
>         - sheet: 'F.Cu'
>           layers:
>           - layer: 'F.Cu'
>           - layer: 'User.Drawings'
>             color: '#ffffff00'
>             use_for_center: true
>         - sheet: '%ln layer'
>           repeat_for_layer: 'F.Cu'
>           repeat_layers: 'inners'
>           layers:
>           - layer: 'F.Cu'
>           - layer: 'User.Drawings'
>             color: '#ffffff00'
>             use_for_center: true
>         - sheet: 'B.Cu'
>           layers:
>           - layer: 'B.Cu'
>           - layer: 'User.Drawings'
>             color: '#ffffff00'
>             use_for_center: true
>         - sheet: 'Silkscreen bottom'
>           layers:
>           - layer: 'B.SilkS'
>           - layer: 'User.Drawings'
>             color: '#ffffff00'
>             use_for_center: true
>           mirror: true
>         - sheet: 'Fabrication bottom'
>           layers:
>           - layer: 'B.Fab'
>           - layer: 'User.Comments'
>           - layer: 'User.Drawings'
>             color: '#ffffff00'
>             use_for_center: true
>           mirror: true
>         - sheet: 'PCB bottom'
>           layers:
>           - layer: 'B.Cu'
>           - layer: 'B.Mask'
>             color: '@KICI_PDF_COLOR_PCB_MASK@'
>           - layer: 'B.SilkS'
>             color: '@KICI_PDF_COLOR_PCB_SILK@'
>           - layer: 'User.Comments'
>           - layer: 'User.Drawings'
>             color: '#ffffff00'
>             use_for_center: true
>           mirror: true
>       plot_sheet_reference: true
> ```
> 
> because what I did here SEEMS to work, but isn't 'short'?

This isn't short because you have a lot of pages, use YAML anchors to make it simpler.

> But then, I also think it's about interpretation? I mean 'Use the center position to center the result around', but it could also mean 'use the center position to center' Sounds close, but different result. Hard to explain too. Kind of like 'center first, then resize, vs resize, then center.

Not sure what you mean, KiCad computes the center of the object and puts it in the center of the page.

oliv3r, 2024-02-13:
> Not sure what you mean, KiCad computes the center of the object and puts it in the center of the page.

I mean, What I expected with the 'User.Eco1' layer center marker, is being used to position the resulting image. E.g. if I put the grid-center marker on 0,0, I would have expected that the resulting image, would be outside of the worksheet, with its center position on 0,0. E.g. a image of 10x10 size, would end up being from -5 to +5, thus with only a corner within the worksheet in this case.

commit b88e09c4 (2024-02-01): [PCB Print][Added] Options to fine tune the centering

- You can select which layers are used for centering purposes

Closes #573