INTI-CMNB/KiBot #574: [HOW] Is it possible to enable / disable footprints in variants?

Kampi, 2024-01-27:
**What do you want to achieve?**

I have some text and a logo in my PCB design on a specific layer and I want to enable / disable it in the Gerber files for specific variants.

[image]

**Do you have some PCB/Schematic to use as example?**

Please use [this](https://github.com/Kampi/ZSWatch-Dock) one.

**Do you have some configuration file (.kibot.yaml) that you are using?**

```yaml
kibot:
  version: 1

variants:
  - name: "Full"
    comment: "Dock with integrated J-Link and second USB"
    type: kibom
    variant: Full
  - name: "Reduced"
    comment: "Dock without integrated J-Link and with second USB"
    type: kibom
    variant: Reduced
  - name: "Minimum"
    comment: "Dock without integrated J-Link and without second USB"
    type: kibom
    variant: Minimum

global:
  out_dir: "../production/Export-%D"
  date_format: "%Y-%m-%d"

outputs:
  - name: "Generate IBOM"
    comment: "Interactive Bill of Materials"
    type: ibom
    dir: docs

  - name: "Print Schematics"
    comment: "Print schematic (PDF)"
    type: pdf_sch_print
    dir: docs
    options:
      output: Schematic.%x

  - name: "Board Top"
    comment: "Top layer view"
    type: pcbdraw
    dir: docs/images
    options:
      format: png
      bottom: false
      output: "PCB_Top.%x"

  - name: "Board Bottom"
    comment: "Bottom layer view"
    type: pcbdraw
    dir: docs/images
    options:
      format: png
      bottom: true
      output: "PCB_Bottom.%x"

  - name: "Generate Drill"
    comment: "Generate drill files"
    type: excellon
    dir: gerber
    options:
      metric_units: true
      mirror_y_axis: false
      pth_and_npth_single_file: false
      left_digits: 0
      minimal_header: false
      use_aux_axis_as_origin: true
      route_mode_for_oval_holes: true
      zeros_format: "DECIMAL_FORMAT"

  - name: "Generate Gerbers"
    comment: "Generate Gerbers"
    type: gerber
    dir: gerber
    options:
      exclude_edge_layer: false
      exclude_pads_from_silkscreen: false
      plot_sheet_reference: false
      plot_footprint_refs: true
      plot_footprint_values: true
      force_plot_invisible_refs_vals: false
      tent_vias: true
      line_width: 0.15

      use_aux_axis_as_origin: false
      subtract_mask_from_silk: true
      use_protel_extensions: false
      gerber_precision: 4.5
      create_gerber_job_file: true
      use_gerber_x2_attributes: true
      use_gerber_net_attributes: false

    layers:
      - layer: F.Cu
        suffix: F_Cu
        description: Front copper
      - layer: B.Cu
        suffix: B_Cu
        description: Bottom copper
      - layer: B.Paste
        suffix: B_Paste
        description: Bottom solder paste
      - layer: F.Paste
        suffix: F_Paste
        description: Front solder paste
      - layer: B.Silkscreen
        suffix: B_Silkscreen
        description: Bottom silkscreen (artwork)
      - layer: F.Silkscreen
        suffix: F_Silkscreen
        description: Front silkscreen (artwork)
      - layer: B.Mask
        suffix: B_Mask
        description: Bottom soldermask (negative)
      - layer: F.Mask
        suffix: F_Mask
        description: Front soldermask (negative)
      - layer: Edge.Cuts
        suffix: Edge_Cuts
        description: Board shape
      - layer: B.Courtyard
        suffix: B_Courtyard
        description: Bottom courtyard area
      - layer: F.Courtyard
        suffix: F_Courtyard
        description: Front courtyard area
      - layer: B.Fab
        suffix: B_Fab
        description: Bottom documentation
      - layer: F.Fab
        suffix: F_Fab
        description: Front documentation
      - layer: User.1
        suffix: User_1
        description: User layer 1

  - name: "Pick & Place"
    comment: "Pick & Place"
    type: position
    dir: centroid
    options:
      format: ASCII
      only_smd: false

  - name: "PCB Specifications"
    comment: "PCB Specifications"
    type: pcb_print
    dir: docs
    options:
      output: "PCB.pdf"
      format: PDF
      pages:
      - layers:
        - layer: Edge.Cuts
        - layer: User.1
        sheet: "PCB Specifications"
        monochrome: true

  - name: "Ref Top"
    comment: "Ref Top"
    type: pcb_print
    dir: docs
    options:
      output: "Ref_Top.%x"
      format: PDF
      pages:
      - layers:
        - layer: Edge.Cuts
        - layer: F.Fab
        sheet: "Ref-Top"
        monochrome: true

  - name: "Ref Bottom"
    comment: "Ref Bottom"
    type: pcb_print
    dir: docs
    options:
      output: "Ref_Bot.%x"
      format: PDF
      pages:
      - layers:
        - layer: Edge.Cuts
        - layer: B.Fab
        sheet: "Ref-Bottom"
        monochrome: true
        mirror: true

  - name: "BOM for PCBWay"
    comment: "BOM for PCBWay"
    type: bom
    options:
      format: XLSX
      count_smd_tht: true
      output: "BOM.%x"
      ref_separator: ','
      columns:
        - field: Row
          name: Item
        - field: References
          name: Ref Des
        - field: Build Quantity
          name: Quantity
        - field: Mfr.
          name: Manufacturer
        - field: Mfr. No.
          name: Mfg Part#
        - field: Value
          name: Description / Value
        - field: Footprint
          name: Package
        - field: Footprint Type
          name: Type
        - field: Footprint Side
          name: Side
      xlsx:
        title: "%f BOM"
        hide_stats_info: false
        quote_all: true

  - name: "ZIP Gerber"
    comment: "ZIP Gerber"
    dir: .
    type: compress
    options:
      format: ZIP
      files:
        - source: gerber/**
      output: "Gerber_%D.%x"

  - name: "ZIP Centroid"
    comment: "ZIP Centroid"
    dir: .
    type: compress
    options:
      format: ZIP
      files:
        - source: centroid/**
      output: "Centroid_%D.%x"

  - name: "Step file"
    comment: "STEP file generation"
    type: step
    dir: cad

  - name: "3D rendering"
    comment: "3D rendering"
    type: render_3d
    dir: docs/images
    options:
      ray_tracing: true
      rotate_x: 3
      rotate_z: -2
      output: "PCB_Rendering.%x"
```

**Environment (please complete the following information):**

Where are you running KiBot:
- GitHub CI/CD workflow

Kampi, 2024-01-29:
Hi @set-soft,

thank you. Is it possible to include specific layers from outside in the Gerber files? I placed the variable content on a different layer so I can leave this layer out.

set-soft, 2024-01-29:
I'm  not sure if I understand your question. You can generate gerbers for any layer, even for documentation layers. The variable to replace should be in the layer you want to generate. 

In your project the logo and message are in the front silk screen, so you could change the text from "Important"... to @SOME_NAME@ and then use *set_text_variables* to change it, i.e. using the content of a text file.

As you can't remove the rectangle containing the text you could use a different text when you want to "disable" it, just any other stuff that fits in the rectangle

closed 2024-01-30 by Kampi

set-soft, 2024-04-09:
Hi @Kampi !

If you want to experiment: the above patch adds a mechanism to change footprints using the variants mechanism.

An example of config can be found in [var_rename_footprint.kibot.yaml](https://github.com/INTI-CMNB/KiBot/blob/dev/tests/yaml_samples/var_rename_footprint.kibot.yaml) (also [var_rename_kicost_footprint.kibot.yaml](https://github.com/INTI-CMNB/KiBot/blob/dev/tests/yaml_samples/var_rename_kicost_footprint.kibot.yaml))

The test cases are: [var_rename_footprint.kicad_pcb](https://github.com/INTI-CMNB/KiBot/blob/dev/tests/board_samples/kicad_7/var_rename_footprint.kicad_pcb) and [var_rename_footprint.kicad_sch](https://github.com/INTI-CMNB/KiBot/blob/dev/tests/board_samples/kicad_7/var_rename_footprint.kicad_sch) (also [var_rename_kicost_footprint.kicad_pcb](https://github.com/INTI-CMNB/KiBot/blob/dev/tests/board_samples/kicad_7/var_rename_kicost_footprint.kicad_pcb) and [var_rename_kicost_footprint.kicad_sch](https://github.com/INTI-CMNB/KiBot/blob/dev/tests/board_samples/kicad_7/var_rename_kicost_footprint.kicad_sch))