INTI-CMNB/KiBot #576: [FEATURE] KiCad v8 support (DRC/ERC preliminarly)

oliv3r, 2024-02-01:
KiCad 8 is just around the corner (RC2 was release just last week), though the end of january deadline has lapsed.

With v8, one exciting new feature is ERC/DRC can be now called from the CLI, and produces JSON output.

Would be great if KiBot would migrate to v8 support, so we don't have to use the KiCad GUI, and video capture, to get ERC and DRC results!

oliv3r, 2024-02-09:
Well it is just an RC for now :( but that's sad. Was hoping we could get away with the record desktop functionality :(

So lets see if they can fix things before 8.0 gets released!

set-soft, 2024-02-11:
16965 was fixed, but even the GUI is full of bugs, exclusions not always work, stray messages about locks, really poor performance if pcbnew open eeschema to do the schematic parity check, schematic parity check always performed in the CLI, etc. 

Needs much more work, is not a "release candidate"

set-soft, 2024-02-17:
Now 16964 is fixed too. So now KiAuto supports E/DRC using *kicad-cli*

I found another bug, related to the exclusions: [17004](https://gitlab.com/kicad/code/kicad/-/issues/17004)

set-soft, 2024-03-21:
Ok, KiCad 8 should be supported by the current *dev* images.

It uses the CLI for ERC and DRC. 

Needs testing

closed 2024-03-21 by set-soft

set-soft, 2024-04-05:
And now we have new preflights named `erc` and `drc`.
They are for KiCad 8 and has more functionality and less bizarre stuff.
You can even get a report in HTML format.