INTI-CMNB/KiBot #577: Unable to access the documentation

LeanderGlanda, 2024-02-03:
A few days ago the documentation was in the Readme file. Now it's gone there and you can find a link to ReadTheDocs, but that link leads to nowhere and i get a 404 error. I guess this is not how it should be?

For now I'll view the docs in the commit history. If anyone else misses the docs right now, here is a link to the old Readme:
https://github.com/INTI-CMNB/KiBot/blob/caad9ed862cdd9b6c650d6d4669652e15ced087b/README.md

Here a screenshot of the 404 error:
[image]

Nudelsalad, 2024-02-03:
Use this in meantime [Link](https://kibot.readthedocs.io/en/latest/index.html)

set-soft, 2024-02-04:
Thanks @LeanderGlanda !

I had to change *stable* to *master*, the former didn't work, and forgot to change the link.

The link provided by @Nudelsalad is right, thanks. But will be updated to the dev stuff, so better use the https://kibot.readthedocs.io/en/master/ link

commit 8770087d (2024-02-04): [DOCs][Fixed] README link to docs

I had to move stable to master in RTD

Fixes #577
closed 2024-02-04 by set-soft