INTI-CMNB/KiBot #578: [BUG] ibom+dnf_filter causes AttributeError: 'SchematicComponentV6' object has no attribute 'id'

vermut, 2024-02-11:
**Describe the bug**
Enabling dnf_filter on ibom output causes AttributeError. Commenting it out works file.

**To Reproduce**
```
# docker run --rm -ti -v .:/app -w /app/Hardware/OpenMowerMainboard ghcr.io/inti-cmnb/kicad7_auto:dev_1.6.4-280648f_k7.0.10_d12.1 kibot -s all ibom

Using config file: OpenMowerMainboard.kibot.yaml
WARNING:(W044) More than one SCH file found in `.`.
  Using ./OpenMowerMainboard.kicad_sch if you want to use another use -e option. (kibot - kiplot.py:828)
- 'Generates an interactive web page useful to identify the position of the components in the PCB.' (ibom) [ibom]
Traceback (most recent call last):
  File "/usr/local/bin/kibot", line 8, in <module>
    sys.exit(main())
             ^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/kibot/__main__.py", line 534, in main
    generate_outputs(outputs, args.target, args.invert_sel, args.skip_pre, args.cli_order, args.no_priority,
  File "/usr/local/lib/python3.11/dist-packages/kibot/kiplot.py", line 594, in generate_outputs
    _generate_outputs(outputs, targets, invert, skip_pre, cli_order, no_priority, dont_stop)
  File "/usr/local/lib/python3.11/dist-packages/kibot/kiplot.py", line 584, in _generate_outputs
    run_output(out, dont_stop)
  File "/usr/local/lib/python3.11/dist-packages/kibot/kiplot.py", line 488, in run_output
    out.run(get_output_dir(out.dir, out))
  File "/usr/local/lib/python3.11/dist-packages/kibot/out_base.py", line 213, in run
    self.options.run(target)
  File "/usr/local/lib/python3.11/dist-packages/kibot/out_ibom.py", line 192, in run
    GS.sch.save_netlist(f, self._comps)
  File "/usr/local/lib/python3.11/dist-packages/kibot/kicad/v5_sch.py", line 2066, in save_netlist
    self.save_netlist_components(root, comps, excluded, fitted, no_field)
  File "/usr/local/lib/python3.11/dist-packages/kibot/kicad/v5_sch.py", line 1937, in save_netlist_components
    tstamps[c.ref] = c.id
                     ^^^^
AttributeError: 'SchematicComponentV6' object has no attribute 'id'
```

Source code available here: https://github.com/rfvermut/OpenMower/tree/extra-caps-and-silkscreen/Hardware/OpenMowerMainboard

**Environment (please complete the following information):**
- Using a docker image: `kicad7_auto:dev_1.6.4-280648f_k7.0.10_d12.1`
- Using a desktop system? Which OS, distro and version? 'Windows/WSL/Podman`, but I doubt it matters

**Additional context**
This is probably something related to my own schematics being upgraded from v6 to v7, some tiny field missing the value, but I cannot figure out which one.

set-soft, 2024-02-11:
Hi @vermut !

Thanks for reporting (and nice project)!

The problem was related to components found in the PCB, but not in the schematic, they needed the UUID. The above patch adds it. Workaround: add them to the schematic, tedious, but more robust.

closed 2024-02-11 by set-soft

set-soft, 2024-02-12:
> Thanks @set-soft!
> 
> Just to clarify - anytime I check this box I'd break KiBot?

Not really. Is "only in PCB" the problem. And it affected the internal netlist generation.

> BTW, also `kiri` was unhappy about missing UUIDs when I gave it enough commits to process. 

If you still have problems with it just open an issue

commit 0db1ded2 (2024-02-11): [Fixed][Netlist] Problems for comps in PCB but not SCH

Missing UUID

Fixes #578