INTI-CMNB/KiBot #579: [HOW] Could kibot DNP a hierarchical sheet

N0cynym, 2024-02-12:
**What do you want to achieve?**

I love working with kibot but would like to know if kibot is able to DNP a hierarchical sheet.

**Do you have some PCB/Schematic to use as example?**

In the attached [kicad-ci-tests.zip](https://github.com/INTI-CMNB/KiBot/files/14236111/kicad-ci-tests.zip) is a project with two LED schematics.

D1 is the base schematic and D2 is just the copy of D1.

Now I want to make a variant with D1 and one with D2 (including or excluding the pre resistor).

Is there a way of doing this with kibot or do I need to create two schematics without any bindings?

set-soft, 2024-02-12:
LEDs.kicad_sch is missing in the .zip, so I don't know if I have the whole picture.

Why do you want to have one with D1 and another with D2? Why not just change the LED and R values? Can you draw the circuits for each variant?

In general variants are used to have 1 physical PCB for multiple products.

N0cynym, 2024-02-13:
I'm sorry I forgot to add the LED schematic.

Here is a cleaned archive of my template:
 
- [kicad-ci-tests.zip](https://github.com/INTI-CMNB/KiBot/files/14256578/kicad-ci-tests.zip)

What I want to do is to draw just one origin schematic which I copie.

My problem is that I can't change parameters without affecting the copie or origin.

Means I can't say in my example `D1` and `R3` = `variant_1` and `D2` and `R4` = `variant_2`.

Or much better schematic LEDs is placed on `variant_1` and schematic LEDs1 on `variant_2`.

Thanks for looking into this 🙏

set-soft, 2024-02-14:
I don't fully understand what you want.

What's the difference between variant_1 and variant_2? The values of R3/R4? From your original description it looks like you want to "remove" one of the sub-sheets ... it looks bizarre. 

Or perhaps you want to mark all components in one of them to DNP? If this is the case just mark them using the "reference", i.e. create a filter to make DNP R3 and D1, and another to DNP R4 and D2. You can filter by reference.

N0cynym, 2024-02-14:
Thanks for your advices 👍 

I hope the following GIF helps to explain what I mean:

D1 and R3 are in the origin schematic LEDs.

D2 and D4 are in the copied schematic of LEDs (LEDs1).

So any changes in this componets effect each other now mean:

You change the config of D2 -> same for D1...

[image]

I mean in KiCad I would wish to say DNP LEDs or LEDs1 so that I just draw one schematic and copy as much as I need of it to config later. But it seems that kibot once again solved my problem!

Thanks for your advice that I just need to write a filter which I can use in my variant in kibot 👍 

Your project is awesome 💪

closed 2024-02-14 by N0cynym

set-soft, 2024-02-16:
Yes, KiCad can't handle different fields on recycled schematic sheets, but KiBot can do it. 

You can play with the *sch_variant* output, if you force a change in a field of a component that is in a recycled sheet you'll see the saved schematic will create 2 separated sheets, no more recycle. This is how KiBot internally solves the KiCad limitation.

So you can work using recycled sheets, and force things that KiCad doesn't support, but you must use filters. You can also use the KiCost mechanism to change fields, in this way you can force different "Values" for a recycled component, even different 3D model.